Koenkk/zigbee2mqtt #17642: [Wrong picture]: TS0601 (_TZE200_ztc6ggyl & _TZE204_ztc6ggyl) without device picture

SAOPP, 2023-05-11:
### Link

https://www.zigbee2mqtt.io/devices/TS0601_smart_human_presence_sensor.html

### Model

TS0601_smart_human_presence_sensor

### Description

Smart Human presence sensor

### Vendor

TuYa

### Picture (link)

https://ae01.alicdn.com/kf/S03606ad5345441988b4a187ff2dbdbabH.jpg

### Database entry

-

### Notes

Hi!

In current dev builds of Z2M the Tuya presence sensor is not have device picture:

[image]
[image]
[image]

In previous builds, the picture was present.

SAOPP, 2023-05-11:
Build: 1.30.4-dev commit: [307e0dc](https://github.com/Koenkk/zigbee2mqtt/commit/307e0dc)

Koenkk, 2023-05-11:
Fixed (check again in 1 hour from now)

SAOPP, 2023-05-11:
@Koenkk I just updated to 1.30.4-dev commit: [3085608](https://github.com/Koenkk/zigbee2mqtt/commit/3085608), not sure if it's correct build commits right now, 'coz I didn't found in commits any affilated to my questions strings (maybe I had wrong searching), but here is the consol of the dashboard:

[image]

Here is broken links for me:
```
https://slsys.github.io/Gateway/devices/png/TS0601.png
https://www.zigbee2mqtt.io/images/devices/TS0601_smart_human_presence_sensor_1.jpg
```

Perhaps I was a little hasty?

Koenkk, 2023-05-12:
I used the wrong file name, should be fixed now.

SAOPP, 2023-05-12:
Thanks homie! I will check it tomorrow and give feedback. :+1:

SAOPP, 2023-05-13:
@Koenkk nope :| I don't know what to say, maybe it's my local issue, but? Some of my homies had check it, and do not have the problem with that pictures, but these links above, is not working for they too.

Koenkk, 2023-05-14:
@SAOPP maybe some browser cache?

SAOPP, 2023-05-15:
@Koenkk of course I tried another browser and hard reset z2m page, dunno what it is :-|

SAOPP, 2023-05-15:
Ooops, just check it. :-)

[image]

SAOPP, 2023-05-15:
But, need to this:

https://ae01.alicdn.com/kf/S03606ad5345441988b4a187ff2dbdbabH.jpg

Koenkk, 2023-05-15:
Updated the picture.

SAOPP, 2023-05-16:
Thanks homie! Issue has been resolved. :fire:

SAOPP, 2023-05-16:
[image]

closed 2023-05-16 by SAOPP