Koenkk/zigbee2mqtt #17643: Aqara FP1 almost unusable (RTCZCGQ11LM)

Caros2017, 2023-05-11:
### What happened?

Like many unsolved closed issues (#16628, #16724 etc) , I am having troubles using the Aqara FP1.
The `Presence `and `presence_events` are not reliable.

When I add the device (tried repairing a lot) I always get:
`zigbee-herdsman-converters:xiaomi:aqara_fp1: Unknown key 0x010c = 3`

I also see a lot of unhandled keys in the log:
```
Received Zigbee message from 'Aqara FP1', type 'attributeReport', cluster 'aqaraOpple', data '{"323":0}' from endpoint 1 with groupID 0
zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x0143 = 0
Received Zigbee message from 'Aqara FP1', type 'attributeReport', cluster 'aqaraOpple', data '{"322":1}' from endpoint 1 with groupID 0
zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x0142 = 1
zigbee-herdsman-converters:xiaomi:aqara_fp1: Unhandled key 0x00f7 = 3:28:16:5:21:1:0:8:21:36:1:0:21:0:21:0:20:14:10:20:1:12:20:0:65:20:1:66:20:3:67:20:0:68:20:0:69:20:0:0:20:1:0:20:3
```

I have checked the basic requirements:
I am running the latest Zigbee2MQTT docker
I am NOT running on Zigbee channel 21-24
I am running on the CC2652P (Sonoff dongle) with firmware 20211217, so newer than required 20211114

### What did you expect to happen?

I expect presence to update always, about one minute after, presence_events state goes to leaving or away.

Both states are not reliable without constant usage of `reset_nopresence_status`. Sometimes it's working, but most of the times it isn't.

### How to reproduce it (minimal and precise)

Just add the device and it will happen.

### Zigbee2MQTT version

1.30.4 commit: b2dd21e

### Adapter firmware version

20211217

### Adapter

Sonoff dongle

### Debug log

See text

markjwiggins, 2023-05-11:
Porting my experience over from the closed ticket...

My FP1 (about a month or two old) was working fine until today, when I now see the same error as the post in Z2M, and the presence does not flip to false when away...

Also presence_event does not change from approach after leaving the room, so can't reliably use that as an automation

If I do a reset_no_presence call, event does update to leave and presence False, and does work correctly when I enter the room, but then when I exit it doesn't reset to unoccupied without me manually resetting

I have a separate motion sensor which I can temporarily use to trigger the reset via an automation, which works, but far from ideal, especially as it seemed to be working until today.

Also I haven't done any updates to HA, Z2M or anything for a couple of weeks so nothing new causing the issue.

closed 2023-06-18 by github-actions[bot]

JaroslawSurowiec, 2023-06-18:
I have the same issue, already upgraded to the latest version, but still not working 
Zigbee2MQTT version 1.31.2

sorryusernameisalreadytaken, 2023-10-20:
Here the same problem. I test 6 devices and all had the same issue.

How can we help to fix that?

AArt36, 2024-02-14:
i have this problem too

sdierick, 2024-02-27:
Hello. I have the same problem with elelabs popp dongle on a raspberry pi running zigbee2mqtt

kamberi, 2024-03-17:
I have a problem too!
for me, recently my fp1 sensor started to flip into false presence even though Im present. It was working fine for a year. I have updated now Z2M to the latest but still same issue (using home assistant, sonoff zigbee coordinator dongle with z2m

TomasWeij, 2024-04-04:
I have the same problem.