Koenkk/zigbee2mqtt #17644: [Feature request]: External Converter Review for Tuya Human Presence Sensor (_TZE200_xpq2rzhq)

seintuit, 2023-05-11:
### Is your feature request related to a problem? Please describe

I left an inquiry a few days ago because of a warning message from '_TZE200_xpq2rzhq'

[image]

### Describe the solution you'd like

I've written a script. (Additional context)
There have been no warnings or error messages since I applied this external converter.
Please let me know if the script lacks or needs to be modified.

### Describe alternatives you've considered

In addition, we request to modify the product photo for the same device such as '_TZE200_xpq2rzhq' and '_TZE204_ztc6ggyl'.

[image]

### Additional context

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'minimum_range', 'maximum_range', 'detection_delay', 'fading_time', 'indent_level', 'trigger_level']
}

const definition = {

    fingerprint: [
        { modelID: 'TS0601', manufacturerName: '_TZE200_xpq2rzhq' }
    ],
    model: 'TS0601_smart_human_presence_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
        exposes.numeric('indent_level', ea.STATE_SET).withValueMin(0).withValueMax(3).withValueStep(1)
            .withDescription('Indent Level'),
        exposes.numeric('trigger_level', ea.STATE_SET).withValueMin(0).withValueMax(3).withValueStep(1)
            .withDescription('Trigger Level'),
        exposes.enum('self_test', ea.STATE, Object.values(tuya.tuyaHPSCheckingResult))
            .withDescription('Self test status'),
    ],
    meta: {
        tuyaDatapoints: [
            [6, 'self_test', tuya.valueConverterBasic.lookup({'checking': 0, 'check_success': 1, 'check_failure': 2, 'others': 3, 'comm_fault': 4, 'radar_fault': 5})],
            [104, 'illuminance_lux', tuya.valueConverter.raw],
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [3, 'minimum_range', tuya.valueConverter.divideBy100],
            [4, 'maximum_range', tuya.valueConverter.divideBy100],
            [9, 'target_distance', tuya.valueConverter.divideBy100],
            [101, 'detection_delay', tuya.valueConverter.divideBy10],
            [102, 'fading_time', tuya.valueConverter.divideBy10],
            [107, 'indent_level', tuya.valueConverter.raw],
            [108, 'trigger_level', tuya.valueConverter.raw],

        ],
    }
};

module.exports = definition;

Koenkk, 2023-05-11:
Please make a pull request to update https://github.com/Koenkk/zigbee-herdsman-converters/blob/15735e3387cd07cd9ed3219a52889e6102315081/devices/tuya.js#L4361 (I will review there and merge it if OK)

closed 2023-06-18 by github-actions[bot]

Vaarinvaliliha, 2025-01-08:
Hi, I have Heiman HS8OS-TY

[tuya-dfb24ca64c0a67909d183bd9bdf8a734-Heiman HS8OS-TY-907249753cf6b5f978f4a940f36a52c3.json](https://github.com/user-attachments/files/18341783/tuya-dfb24ca64c0a67909d183bd9bdf8a734-Heiman.HS8OS-TY-907249753cf6b5f978f4a940f36a52c3.json)

It has no support (z2m/zha/Tuya-integration)

BR Joonas

**{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4619,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0b05",
        "0xef00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019",
        "0xef00"
      ]
    }
  },
  "manufacturer": "_TZ6210_duv6fhwt",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`**