Koenkk/zigbee2mqtt #17647: Should not send color_mode when there is no cached color for lights

corporategoth, 2023-05-11:
### What happened?

Details from https://github.com/home-assistant/core/issues/92946

HA does not handle it well if color_mode is sent in an update (for a light), without an actual color associated with it.

This can happen if the light has not been switched on since Z2M initialized, so could not cache the previous color of the light (and the light doesn't send it's color if it's off).

### What did you expect to happen?

Z2M should not send a color_mode without a color in a state update.

### How to reproduce it (minimal and precise)

Pair an RGB light to Z2M (never seen before).
Do NOT turn it on.
Profit?

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ezsp (Sonoff Dongle-E)

Koenkk, 2023-05-12:
Based on https://github.com/home-assistant/core/issues/92946 I'm wondering how you got `color_mode: xy` without having a `color`. Could you provide me the debug logging of:

- stop z2m
- clear the device state in `data/state.json`
- start z2m
- reproduce the issue

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

corporategoth, 2023-05-12:
@Koenkk - If you're talking about non-herdsman logging - I can't start my instance with debug logging enabled due to https://github.com/Koenkk/zigbee2mqtt/issues/17318 - though I can start if with debug herdsman logging (in fact, some times I have to to work around https://github.com/Koenkk/zigbee2mqtt/issues/17532).  Did you want the herdsman logs only?

Note, if I DO log with herdsman, it will still be a lot of logging - I have more than 150 devices on my zigbee network.

Koenkk, 2023-05-13:
I need the debug logging (not the herdsman one), what you can do is start z2m with info level and once started change the log level through: https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

closed 2023-06-20 by github-actions[bot]

skrobul, 2024-02-01:
I had the same problem and stumbled upon this issue. Eventually, I found out that it happened also when the light was turned off/on even from the z2m dashboard, but the problem stopped as soon as I used `light.turn_on` with explicitly provided color for the first time.

In other words, the color was not cached because it was never actually set - I have only used `light.turn_on` and `light.turn_off` since the device was added to z2m.

ps. this was on the Hue Go light with zzh! adapter.

Debug logs:
[z2m_issue_17647.log.gz](https://github.com/Koenkk/zigbee2mqtt/files/14118760/z2m_issue_17647.log.gz)

Last entry in HA:
```
2024-02-01 00:13:20.997 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value '{'brightness': 229, 'color_mode': 'xy', 'linkquality': 102, 'power_on_behavior': None, 'state': 'ON', 'update': {'installed_version': 1124101125, 'latest_version': 1124101125, 'state': 'idle'}, 'update_available': None}' received for entity light.bedroomgo
```

The request to set color was sent at `2024-02-01 00:15:07`

Hope this helps :-)