Koenkk/zigbee2mqtt #17650: IKEA E1743 seems to lose connection to Z2M but bound devices work well

KaHeMu, 2023-05-12:
### What happened?

The IKEA E1743 switches repeatedly lose connection with the coordinator after a short period of 2-3 days. They seem to have disappeared. No debug logs or anything the switch is pressed. However, bound devices, that we bound with the E1743 via the Z2M Frontend, even non-IKEA devices, still receive their on/off signal from the E1743 and work properly but no reporting from the E1743 to Z2M. I own multiple of the E1743 and all behave the same. Latest device FW v2.3.079 installed via OTP.  

### What did you expect to happen?

I expect that the device does not lose connection and keeps sending messages to Z2M

### How to reproduce it (minimal and precise)

Join the device.
Bind the device with a target (e.g. a simple on/off-switch like KS-SM001).
Device works fine and reports any action to the coordinator.
After 1-3 days it stops reporting. The Frontend shows "N/A" under Devices -> Exposes -> action
Still the device works with the directly bound target. 

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652 stick

### Debug log

No debug log exists

programmieramt, 2023-05-13:
I have the same problems with this ikea device. Same issue I have for ikea E1812.

I use the Home Assistant Yellow

cometsfernbroccoli, 2023-05-14:
I also have the same issue. button won't show anything in the logs, light (ikea) are still working via HA but not via button anymore.
The button itself is about 3 meters away from the coordinator.
I have this issue since the latest firmware update of the ikea button which also made it impossible to bind the button to a group.

The number of lights bound to this button is 3. could it be an issue that Z2M sends 3 commands directly to the lights which might interferes ?

Soepgroenten, 2023-05-17:
I got exaclty the same issue with the same device as also the E1812 devices from Ikea and the Silvercrest HG08164 buttons from Lidl. in the end I think those are both a tuya rebranded device. So I've decided to say goodbye to all Tuya (and rebrands).

When re-adding the device it wil work for a day or so, at some point Z2M starts popping up messages in green balloons like: "Accepting joining not in blocklist device '<Device ID>'" when the button is pressed.

The log says:
[image]

Got this already for over a year and from time to time I spended hours investigating and finding a solution but it seems unsolvable.

However there were signs that Binding the device to the coordinator manually helped some of us (of course it didnt workout for me)

Also [some say](https://community.home-assistant.io/t/zigbee2mqtt-device-re-pairing/439449/21) that downgrading the firmware helped..

Would be nice if this issue got resolved permanently but actually I think it's Tuya who tries to prevent customers from adding the device to non-tuya solutions.

I can help in some way I would be glad

metalblue, 2023-05-23:
This is happening to me too on a regular basis. I've got 10 of these devices and I'm currently only going 1-2 days before at least one of them drops off the network.

Removal of the battery from the E1743 and then reinsertion appears to temporarily fix the issue as the device soon reconnects. Initially I thought I'd got a bad batch of batteries, but this doesn't appear to be the case.

None of my other IKEA devices, E1525/E1745 (motion) or (E1812) shortcut button have issues.

I'm using the Texas Instruments LAUNCHXL-CC1352P-2 as my coordinator and this issue has got much worse since I upgraded the firmware on it to the latest about a month ago.

Soepgroenten, 2023-05-23:
> This is happening to me too on a regular basis. I've got 10 of these devices and I'm currently only going 1-2 days before at least one of them drops off the network.
>
> Removal of the battery from the [...]

Hi I finally seem to solve this issue by directly bind it to the coordinator.
I was pointed the right direction when reading [this post ](https://github.com/Koenkk/zigbee2mqtt/issues/9439#issuecomment-956456214:~:text=edited-,Just%20in%20case%20anyone%20is%20having%20the%20same%20issue%2C%20a%20workaround%20to%20make%20the%20E1766%20to%20send%20messages%20again%20to%20z2m%20is%20to%20bind%20it%20to%20the%20coordinator%20after%20pairing%20by%20sending%20the%20mqtt%20message%20zigbee2mqtt/bridge/request/device/bind%20with%20payload%20%7B%22from%22%3A%20%22%3Cdevice%20friendly%20name%3E%22%2C%20%22to%22%3A%20%22coordinator%22%7D,-%F0%9F%91%8D)

it took me some time to figure it out but in the end I discovered you have to follow these steps:

In HA: 
1.  Go to Settings --> Devices & Services --> Search for the "Mosquitto broker
MQTT" Integration
2. Click Configure at the "Mosquitto brokerMQTT" tile.
3. Under "Publish a packet" at "Topic" place the following: zigbee2mqtt/bridge/request/device/bind (Make sure the "Retain" switch is turned on)
4. At "Payload" place the following: {"from": "Device Friendly Name", "to": "coordinator"}
5. Hit the "Publish" button

Below my last command:
[image]

My E1743 and E1812 devices are running very well for 5 days now, usually they stopped working after 1 or 2 days so I'm pretty conviced this solved my issue and I'm happy te be able to share this with you :-)

Let me know if you have any questions

metalblue, 2023-06-05:
This has not fixed the issue for me. My E1743 and E1812 still become unavailable after a few days.

KaHeMu, 2023-06-07:
After re-pairing and putting my E1743 directly near the coordinator (where it is of no use to me 👎) the issue is gone. It stayed connected for >3 weeks without any issue whereas previously, when connected via a TuYa or IKEA device as a router, it lost connection approx. every other day. Any idea what this behavior might hint to? Is it a router issue or E1743 issue, or even Z2M? Can anyone try if you observe the same when putting the E1743 directly at the coordinator (~20 cm)?

closed 2023-07-15 by github-actions[bot]

Meisteryoda1994, 2023-07-16:
Me 2 still having the same problem unfortunately