Koenkk/zigbee2mqtt #17651: [New device support]: honyer HY0157

65698888, 2023-05-12:
### Link

http://en.hongyan.com.cn/

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x000d6f00175d78f2","nwkAddr":61068,"manufId":4451,"manufName":"","powerSource":"Mains (single phase)","modelId":"HY0157","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"65024":{"attributes":{"0":4294967295,"1":{"type":"Buffer","data":[48,51,51,51,51,51,51,51,51,51,51,51,51,51,51,51]},"2":{"type":"Buffer","data":[52,49,52,51,57,55,54,55,97,53,98,53,52,56,98,97]},"3":300}},"genBasic":{"attributes":{"32768":0,"modelId":"HY0157","manufacturerName":"","powerSource":1,"zclVersion":0,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20200717","swBuildId":"20200717 14:46"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":2250,"rmsCurrent":0,"activePower":0,"acPowerMultiplier":1,"acPowerDivisor":10,"acVoltageMultiplier":1,"acVoltageDivisor":10,"acCurrentMultiplier":1,"acCurrentDivisor":1000}},"seMetering":{"attributes":{"currentSummDelivered":[0,329],"multiplier":1,"divisor":1000}},"genOnOff":{"attributes":{"32768":0,"32771":0,"32773":1,"32774":0,"tuyaOperationMode":0,"onOff":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812c4d","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812c4d","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812c4d","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812c4d","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"manufacturerName":"","zclVersion":0,"appVersion":0,"stackVersion":0,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022812c4d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20200717","swBuildId":"20200717 14:46","zclVersion":0,"interviewCompleted":true,"meta":{"configured":1435641180},"lastSeen":1683870489013,"defaultSendRequestWhen":"immediate"}

### Comments

Honyer  U86Z223A10-ZJU01(GD)

Request merge
[image]

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['HY0157'],
    model: 'HY0157',
    vendor: 'Honyar',
    description: 'power point with USB',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.energy(), e.current(), e.voltage()],
	endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true, multiEndpointSkip: ['energy', 'current', 'voltage', 'power']},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
		const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
		await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
        await endpoint1.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
        await reporting.activePower(endpoint1, {change: 10});
        await reporting.rmsCurrent(endpoint1, {change: 50});
        await reporting.rmsVoltage(endpoint1, {change: 5});
		await reporting.onOff(endpoint1);
		await reporting.onOff(endpoint2);
		await reporting.currentSummDelivered(endpoint1);
        await reporting.readMeteringMultiplierDivisor(endpoint1);
        endpoint1.saveClusterAttributeKeyValue('seMetering', {divisor: 1000, multiplier: 1});
        endpoint1.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 10, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
            acPowerDivisor: 10,
        });
        device.save();
    },
};
module.exports = definition;
```

fantasyjqc, 2023-05-12:
@65698888 电压/电流和功率似乎都需要除以10才是正确的结果。建议state改为plug和usb

65698888, 2023-05-12:
> @65698888 电压/电流和功率似乎都需要除以10才是正确的结果。建议state改为plug和usb

抱歉，复制的时候选错了文件。
USB单独控制没有什么用，修改成一个switch。

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['HY0157'],
    model: 'HY0157',
    vendor: 'Honyar',
    description: 'power point with USB',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.energy(), e.current(), e.voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
        await reporting.onOff(endpoint);
        await reporting.activePower(endpoint, {change: 10});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.rmsVoltage(endpoint, {change: 5});
		await reporting.currentSummDelivered(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 1000, multiplier: 1});
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
            acVoltageMultiplier: 1, acVoltageDivisor: 10, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
            acPowerDivisor: 10,
        });
        device.save();
    },
};

module.exports = definition;

closed 2023-06-19 by github-actions[bot]