Koenkk/zigbee2mqtt #17659: TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.

jzielke84, 2023-05-12:
### What happened?

Getting this warning in the HA plugin protocol:
TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.

### How to reproduce it (minimal and precise)

Install add-on, start it in HA.

### Zigbee2MQTT version

1.30.4-dev commit: 3085608

### Adapter firmware version

20221102

### Adapter

Sonoff USB 3.0 P-Version

jzielke84, 2023-06-22:
@Koenkk This problem also spams the syslog every 30 seconds.

Koenkk, 2023-06-22:
Please provide debug logs

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jzielke84, 2023-06-22:
@Koenkk 

Addon config:
```yaml
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: /dev/ttyUSB0
zigbee_herdsman_debug: true
```

Log:
```shell
# cat ~/z2m_log.txt 
[20:35:29] INFO: Preparing to start...
[20:35:29] INFO: Socat not enabled
[20:35:30] INFO: Zigbee Herdsman debug logging enabled
[20:35:30] INFO: Starting Zigbee2MQTT...
2023-06-22T18:35:32.700Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-06-22T18:35:32.702Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[208,55,45,145,170,147,250,135,97,214,133,137,6,87,30,107],"panID":6754,"extendedPanID":[221,130,23,66,2,74,39,8],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-06-22T18:35:32.702Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-06-22T18:35:32.708Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2023-06-22T18:35:32.710Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-06-22T18:35:32.711Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-06-22T18:35:32.715Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2023-06-22T18:35:32.715Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2023-06-22T18:35:32.716Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2023-06-22T18:35:32.717Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2023-06-22T18:35:32.718Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-06-22T18:35:32.718Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-06-22T18:35:32.718Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-06-22T18:35:32.723Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2023-06-22T18:35:32.724Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2023-06-22T18:35:32.724Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2023-06-22T18:35:32.724Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2023-06-22T18:35:32.724Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-06-22T18:35:32.724Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2023-06-22T18:35:32.725Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2023-06-22T18:35:32.731Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,174,140,52,1,0,121]
2023-06-22T18:35:32.731Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,174,140,52,1,0,121]
2023-06-22T18:35:32.731Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,174,140,52,1,0] - 121
2023-06-22T18:35:32.732Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20221102}
2023-06-22T18:35:32.732Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-06-22T18:35:32.732Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2023-06-22T18:35:32.732Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20221102})
2023-06-22T18:35:32.733Z zigbee-herdsman:adapter:zStack:startup beginning znp startup
2023-06-22T18:35:32.734Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":130}
2023-06-22T18:35:32.734Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,130,0,178]
2023-06-22T18:35:32.737Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,24,0,104]
2023-06-22T18:35:32.737Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,24,0,104]
2023-06-22T18:35:32.737Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [24,0] - 104
2023-06-22T18:35:32.738Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":24}
2023-06-22T18:35:32.738Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-06-22T18:35:32.738Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":130,"offset":0}
2023-06-22T18:35:32.738Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,130,0,0,0,187]
2023-06-22T18:35:32.744Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,97,28,0,24,0,208,55,45,145,170,147,250,135,97,214,133,137,6,87,30,107,1,0,32,25,241,144,0,166]
2023-06-22T18:35:32.744Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,97,28,0,24,0,208,55,45,145,170,147,250,135,97,214,133,137,6,87,30,107,1,0,32,25,241,144,0,166]
2023-06-22T18:35:32.744Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 3 - 1 - 28 - [0,24,0,208,55,45,145,170,147,250,135,97,214,133,137,6,87,30,107,1,0,32,25,241,144,0] - 166
[... cut ...]
2023-06-22T18:36:04.575Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,26,220]
2023-06-22T18:36:04.575Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,26,220]
2023-06-22T18:36:04.575Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,26] - 220
2023-06-22T18:36:04.575Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":26}
2023-06-22T18:36:04.575Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2023-06-22:
Did the issue always occur or since a recent version?

jzielke84, 2023-06-22:
Well as far as I can look back at my logs the error already on May 22nd until today.

Koenkk, 2023-06-22:
What was the last known working z2m version, and if you go back to that, is the issue fixed?

jzielke84, 2023-06-22:
I can't say for sure. Just stumbled across this one because I took a look at the syslog. It never affected general Z2M operations so I didn't notice it before.

jzielke84, 2023-07-23:
Don’t you dare, GitHub bot!

jzielke84, 2023-08-23:
Nope still an issue

jzielke84, 2023-09-24:
Still there:
`(node:6) TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1. `

Koenkk, 2023-09-24:
Can you take a look at https://github.com/Koenkk/zigbee2mqtt/issues/17659#issuecomment-1603154994 ? Without this info it's hard for me to solve this.

jzielke84, 2023-09-24:
I would like to, but how shall I do that? I can only download the latest version in HA.

Koenkk, 2023-09-24:
Looking at the log, it seems we can also get the warning with the `--trace-warnings` option. To get the log here:

- Update to the edge addon in 2 hours from now
- Enable the `herdsman_debug_log` option

jzielke84, 2023-09-24:
OK, I'll report back to you.

Koenkk, 2023-09-28:
it looks like you've set a very high availability timeout for a certain device, lowering it should fix this.

jzielke84, 2023-09-28:
It's set to `43200` for routers and battery devices. I've experienced that some devices, when not used regularly, are marked as offline which isn't the case.

Koenkk, 2023-09-28:
Try half of that value

jzielke84, 2023-09-28:
With half (`21600`) the log still says:
```
(node:7) TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(node:7) TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(node:7) TimeoutOverflowWarning: 2592001000 does not fit into a 32-bit signed integer.
```

jzielke84, 2023-09-28:
I'll close the issues since the main problem has been solved.