Koenkk/zigbee2mqtt #17662: [New device support]: 45857GE Energy Monitoring Missing

imcdona, 2023-05-12:
### Link

https://easyzigbee.com/mwdownloads/download/link/id/2029/

### Database entry

{"id":60,"type":"Router","ieeeAddr":"0x0022a300001f4182","nwkAddr":45642,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"45857","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,1794],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"45857","manufacturerName":"Jasco Products","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ef71d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":260,"inClusterList":[0,3,2821],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":18,"stackVersion":80,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":91512617},"lastSeen":1683925757163,"defaultSendRequestWhen":"immediate"}

### Comments

Same as issue #15014 except for model 45857GE (Dimmer switch) instead of 45856GE (Switch). Everything works fine except for the lack of energy monitoring.

Koenkk, 2023-05-13:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['45857'],
    model: '45857GE',
    vendor: 'GE',
    description: 'ZigBee in-wall smart dimmer',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.energy(), e.power()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 10000, multiplier: 1});
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

imcdona, 2023-05-13:
_power_ (Instantaneous measured power) and _energy_ (Sum of consumed energy) are reported now.  However I've lost the ability to dim the lights. The 45857GE is the dimmer switch variant. FWIW I was able to dim the lights prior to installing the external converter.

Thanks for your work on this.

Koenkk, 2023-05-14:
Added that back and integrated it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-14 by Koenkk