Koenkk/zigbee2mqtt #17663: OTA for Inovelli VZM31-SN fails with error 404

bubalouie, 2023-05-13:
### What happened?

OTA page indicates an update is available for Inovelli Blue 2-1 VZM31-SN:
[image]
Clicking on the 'Update device firmware' button results in the error: "Update of 'Switch - ST - 01' failed (Request failed with status code 404)"

To be certain, I went to the Dev console and checked the genBasic attribute of swbuildid and received this:
[image]

### What did you expect to happen?

Expected firmware to be updated to the newest version. I am currently on 2.08 (20221110) and the newest is 2.14.

### How to reproduce it (minimal and precise)

Click update button in OTA screen for a VZM31-SN with 2.08 installed

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

6.10.3.0 build 297

### Adapter

TubesZB EFR32 MGM21 over Ethernet

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11468412/log.txt)

kiler129, 2023-05-13:
It's not a one-off - all my Inovelli switches fail with this error as well.

eregev, 2023-05-15:
Likewise.  Failing here as of a few days ago.

bgorczyk, 2023-05-17:
2.14 was pulled.  I believe Z2M reads https://files.inovelli.com/firmware/firmware.json to obtain version info.  The .ota was pullled by the JSON was not updated, causing the 404.  The JSON has been updated.

closed 2023-06-24 by github-actions[bot]