Koenkk/zigbee2mqtt #17668: Hue Hue white and color ambiance LightStrip blocked

d1nd141, 2023-05-13:
### What happened?

Hi,

I joined a second Hue Hue white and color ambiance LightStrip to z2m.
The first one works fine.
The second is interviewed, and fully detected.
But controlling does not work.
I can control 1* to off state (powered on when power adapter plugged in).

In stable 1.30.3, the switch to turn on/off does not change.
In dev latest, same, but the light does not turn off. Debug logs attached.

I also tried without the first Strip (removed power adapter).
Both Lights are at the same place. So no distance issue.
Both lights where connected to a Hue Hub before. -> No problem
Did a reset with a Dimmer switch(12seconds.....)

### What did you expect to happen?

Light turn on/off -> controllable

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

0x26780700

### Adapter

Deconz

### Debug log

Zigbee2MQTT:debug 2023-05-13 09:44:04: Received MQTT message on 'zigbee2mqtt/Kueche_Boden_Eingang/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2023-05-13 09:44:04: Publishing 'set' 'state' to 'Kueche_Boden_Eingang'
Zigbee2MQTT:debug 2023-05-13 09:44:10: Received MQTT message on 'zigbee2mqtt/Kueche_Boden_Eingang/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-05-13 09:44:10: Publishing 'set' 'state' to 'Kueche_Boden_Eingang'
Zigbee2MQTT:debug 2023-05-13 09:44:15: Received MQTT message on 'zigbee2mqtt/Kueche_Boden_Eingang/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2023-05-13 09:44:15: Publishing 'set' 'state' to 'Kueche_Boden_Eingang'
Zigbee2MQTT:error 2023-05-13 09:44:15: Publish 'set' 'state' to 'Kueche_Boden_Eingang' failed: 'Error: Command 0x0017880100ce37d7/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (3))'
Zigbee2MQTT:debug 2023-05-13 09:44:15: Error: Command 0x0017880100ce37d7/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (3))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:656:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:835:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:294:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:961:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:36)
Zigbee2MQTT:error 2023-05-13 09:44:21: Publish 'set' 'state' to 'Kueche_Boden_Eingang' failed: 'Error: Command 0x0017880100ce37d7/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (4))'
Zigbee2MQTT:debug 2023-05-13 09:44:21: Error: Command 0x0017880100ce37d7/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (4))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:656:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:835:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:294:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:961:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:36)
Zigbee2MQTT:error 2023-05-13 09:44:25: Publish 'set' 'state' to 'Kueche_Boden_Eingang' failed: 'Error: Command 0x0017880100ce37d7/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (5))'
Zigbee2MQTT:debug 2023-05-13 09:44:25: Error: Command 0x0017880100ce37d7/11 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (5))
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:656:23)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:81:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:835:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:294:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:961:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:247:36)

5olu7ion, 2023-05-13:
I maybe have the same problem with hue lamps. All works well, and suddenly random devices get unavailable. I'm using z2m since two days, and this is the second time, that this happened. Never occured with the hue bridge. The only thing that helps, is resetting the lamps with the hue dimmer. I'm using an unflashed Sonoff Zigbee 3.0 USB Dongle Plus (E)  with a raspberry Pi 3B+.

Error 2023-05-13 20:49:24Publish 'set' 'brightness' to 'sofalampe' failed: 'Error: Command 0x0017880110425419/11 genLevelCtrl.moveToLevelWithOnOff({"level":177,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45612 - 11 - 184 - 8 - 11 after 10000ms)'
Error 2023-05-13 20:49:44Publish 'set' 'color' to 'sofalampe' failed: 'Error: Command 0x0017880110425419/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":41870,"colory":22354}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45612 - 11 - 188 - 768 - 11 after 10000ms)'
...
Warning 2023-05-13 20:53:04Failed to ping 'sofalampe' (attempt 1/2, Read 0x0017880110425419/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45612 - 11 - 193 - 0 - 1 after 10000ms))
Warning 2023-05-13 20:53:27Failed to ping 'sofalampe' (attempt 2/2, Read 0x0017880110425419/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45612 - 11 - 194 - 0 - 1 after 10000ms))

d1nd141, 2023-05-13:
MAYBE i found the guilty.
My Deconz USB device was reset several times around the time of the problem. No idea why.
`[Sat May 13 09:48:00 2023] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[Sat May 13 09:48:02 2023] usb 1-2: reset full-speed USB device number 18 using xhci_hcd
[Sat May 13 09:48:04 2023] usb 1-2: USB disconnect, device number 18
[Sat May 13 09:48:04 2023] usb 1-2: new full-speed USB device number 19 using xhci_hcd
[Sat May 13 09:48:04 2023] usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[Sat May 13 09:48:04 2023] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat May 13 09:48:04 2023] usb 1-2: Product: ConBee II
[Sat May 13 09:48:04 2023] usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
`
Meanwhile i ordered a Sonoff one to test. But the Hue device seems fine on the map.
[image]

5olu7ion, 2023-05-13:
You are right, it looks healthy, but it isn't. Have you recognized, that the "broken" devices are missing the typical zigbee router "/" values? (e.g. 86/112). It's the same on my end, all router devices have two values separated by a "/" (LQI values for attached devices), except the unavailable ones, they only have on LQI Value. Are you using a raspberry pi? If yes, you can try changing the usb driver.
/boot/config.txt: `dtoverlay=dwc2`

d1nd141, 2023-05-13:
No, i'm using an intel Nuc -> Proxmox -> RHEL -> z2m in docker (deconz connected with small Usb cable of +-30cm)
I will check tomorrow again on this.

d1nd141, 2023-05-14:
Small additional info regarding slash between numbers:
https://github-wiki-see.page/m/dresden-elektronik/deconz-rest-plugin/wiki/LQI-explained

d1nd141, 2023-05-15:
I got the Sonoff USB 3 coordinator. Repaired the devices.
This time, it's the other Lightstrip not working. The one i had previously the issue works fine.
I can't even add the device to a group.

```
debug 2023-05-15 17:33:15Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x0017880100ce37d4/11","group":"Kueche_Boden","transaction":"qjeqs-26"}'
info 2023-05-15 17:33:15Adding 'Kueche_Boden_Terrasse' to 'Kueche_Boden'
error 2023-05-15 17:33:41Failed to add from group (Command 0x0017880100ce37d4/11 genGroups.add({"groupid":1,"groupname":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))

```
Maybe related:
https://github.com/Koenkk/zigbee2mqtt/issues/12774

d1nd141, 2023-05-15:
My issue seems solved atm.
Swapped the Deconz to a Sonoff USB 3 -> OK

5olu7ion, 2023-05-15:
I'm glad its working for you, do you also have the 6.10.3.0 build 297 firmware on you Dongle E? (-> /settings/about)
I'm thinking about flashing a new 7.2.x Firmware, but cant find a changelog online: https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.2.3.0_115200.gbl

d1nd141, 2023-05-15:
No, i flashed
zStack3x0
20221226

5olu7ion, 2023-05-15:
Ah, thats interesting, I think I will try the same. It should be this File for Dongle E: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip

And I think re paring should not be needed in this scenario: https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices

5olu7ion, 2023-05-15:
I have some new findings, I tried to flash the new firmware and therefore restarted my mqtt and zigbee2mqtt docker compose stack. And before flashing the new firmware to my dongle e, the light started to become available again. I will wait until it happens again, to see if restarting zigbee2mqtt or mqqt did the trick.

d1nd141, 2023-05-16:
Issue solved for now.

closed 2023-05-16 by d1nd141