Koenkk/zigbee2mqtt #17670: Channel change, device list empty

darkner258, 2023-05-13:
### What happened?

 I’m running Zigbee2MQTT alongside Homebridge and due to some reliability issues and possible interferience with WiFi, I changed the channel from 11 to 15 hoping it’d help.
I got prompted to restart Zigbee2MQTT, I did so, but now all of my devices are gone from the dashboard.

They are still in the configuration.yaml (I see theim in settings-raw, but not in the Devices list in the frontend. Please, how do I fix this? I do have a backup ready, I’m just lost as to how to restore from it.

Many thanks

### What did you expect to happen?

I hoped all devices would simply migrate to the new channel similar to WiFi devices

### Zigbee2MQTT version

1.30.1 commit: eb878d3d

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee Dongle

VincentSC, 2023-05-13:
Channel change results in all devices needing to be repaired. Somehow the network-key depends on the channel.

darkner258, 2023-05-13:
Understood, I repaired them. Wasn’t as big of a hassle than expected :) Thanks!

closed 2023-05-13 by darkner258