Koenkk/zigbee2mqtt #17674: Profalux cover pairing issues - cover not functional through zigbee2mqtt despite pairing semi-successfully

sven337, 2023-05-14:
### What happened?

Profalux covers are notoriously hard to pair successfully.  I have a MOT-C1Z06C ("NSAV061") for which the remote regularly loses ability to control the cover, so I have to re-pair the cover regularly.

Every so often when I do so, I observe that I  cannot control the cover from Zigbee2Mqtt. In fact, it's supposed to expose control clusters on endpoint 1, but there is no endpoint 1:
   ```
 0x20918a000017cddc
        Endpoint 2
            Output clusters
                genTime
                genOta
                genPollCtrl
                64544
            Input clusters
                genBasic
                genPowerCfg
                genDeviceTempCfg
                genIdentify
                genGroups
                genScenes
                genAlarms
                genTime
                closuresWindowCovering
                haDiagnostic
                manuSpecificSamsungAccelerometer
                64533
                64544
                64545
        Endpoint 242
            Output clusters
                greenPower
            Input clusters
                greenPower
```

I don't know why this is happening, but I wonder if something is going wrong with the pairing process. I dumped debug info from zigbee-herdsman:

[profalux_dump.txt](https://github.com/Koenkk/zigbee2mqtt/files/11471534/profalux_dump.txt)

I have a couple of questions:
```2023-05-13T20:17:49.024Z zigbee-herdsman:controller:device:log Interview - start device '0x20918a000017cddc'
2023-05-13T20:17:49.025Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0x20918a000017cddc', already got it```

Is this concerning? Should I do something so the node descriptor request is done again?

`2023-05-13T20:17:49.094Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 4 | 5`

The log is full of these "unexpect packet sequence" cases. What does it mean? Does it imply that maybe we're missing/not handling some of the packets?

I'm not great at Zigbee stuff, is there anything in my log that helps you understand why the control clusters (6 and 8) are not exposed on endpoint 1?

Thanks

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

EZSPv8    6.10.3.0 build 297 

### Adapter

Sonoff ZBDongle-E

sven337, 2023-05-14:
What works to control the cover once in this state is endpoint 2, cluster 0x102. Command 0 : open, command 1: close, command 5: level control (but the percentage is non-linear, 33% corresponds to 50% in reality).
I don't see cluster 0x8 exposed at all by the cover in this state -- it seems like it will pick one mode or the other? :/

Iroamstaff, 2023-05-18:
Salut j'ai 4 volets profalux 4 fonctionne parfaitement depuis 6 mois. Ils sont un peut difficile a intégrer mais c'est faisable
Tu as qu'un seul volet profalux ou plusieurs ?

sven337, 2023-05-18:
I think what I have is the new model (bought in March 2023). It works, but with the above caveat. I had to change the control clusters after re-pairing it once. I don't know why, but that's how it is. After a couple of weeks, the remote stops controlling the covers, and my solution was that was a full reset (blue wire on live, wait for three series of movements, cut power, ... then 5*R on the remote and stop button).

For now there's only one cover, I've got 4 more waiting to be installed.

Last time I re-paired ,the cover ended up exposing the closuresWindowCovering cluster instead of genLevelCtrl/genOnOff.  closuresWindowCovering is conceptually more correct.
Cover firmware version is reported as 20000101

sven337, 2023-05-18:
Merci, mais ça ne change pas mon rapport de bug : le volet peut s'appairer (ça m'est arrivé plusieurs fois) dans le mode que je décris où il expose un différent système de contrôle que celui utilisé par Z2M. Dans ce cas la télécommande fonctionne mais pas Z2M. D'où le bug report et le patch !

Note que ta procédure ne change a priori rien - tu rajoutes un volet au réseau mais si en sortie de reset le moteur trouve un réseau déjà ouvert il va le rejoindre directement. 

J'en suis à une dizaine de ré-appairages sur ce volet, parce que régulièrement la télécommande "saute". La prochaine fois que ça arrive je snifferai le réseau pour comprendre ce qu'il se passe au lieu de faire un reset bêtement.

Quoiqu'il en soit, lorsque le volet expose closuresWindowCovering, il faut le contrôler par ce biais et non par genLevelCtrl/genOnOff. Pourquoi est-ce que le volet se met à exposer cela je n'en sais rien.

Iroamstaff, 2023-05-18:
Par contre a chaque fois qu'un nouveau volet entre dans le réseau il fonctionne avec ça télécommande et dans z2m les commandes ne fonctionnait pas directement, ya juste le retour d'état qui fonctionne mais pas les commandes open stop close.
Donc pour qu'elle fonctionne dans z2m la seule solution que j'ai trouvé c'était de redémarrer tout mon système

sven337, 2023-05-18:
Ah, donc tu as peut-être rencontré le même problème que moi ! Jette un oeil aux clusters exposés par ton volet quand le problème se présente, je parie que tu verras closuresWindowCovering.

Iroamstaff, 2023-05-18:
J'ai eu ce problème a chaque fois quand j'ajoute un volet au réseau mais le fait de redémarrer mon rspi règle le problème définitivement et ne reviens pas

sven337, 2023-05-18:
@Koenkk is there a way for Z2M to adapt to different clusters being exposed by the same device?
Sometimes genLevelCtrl, sometimes closuresWindowCovering, but never both at the same time. I don't know how to hook this up

Iroamstaff, 2023-05-18:
As-tu bien la télécommande qui est dans le réseau avec le volet ?

Iroamstaff, 2023-05-18:
Pour moi ça vient pas de z2m mais de la procédure appairage

1 Fais un reste moteur Méthode avec le fil noir et les 3 séries de va et vient couper puis remettre l'alimentation le volet fait des mouvements de va-et-vient (il cherche a joindre la télécommande il ne faut pas activer l'appairage sur z2m) 

2 Prendre la télécommande et bouton stop.

3 Attendre la fin des voyants.

4 Regarder si la télécommande fonctionne bien avec le volet

5 Si tout est ok prendre la télécommande et faire 1 appuis sur F (Ne pas appuyer sur R) le voyant s'allume soit violet ou soit rouge 

-Si le voyant s'allume violet utilise la méthode que je t'ai décrit ci-dessus 4 appuis puis 1  appuis de 15 secondes sur R ( les 5 appuis doit être fait rapidement )

-Si par contre le voyant s'allume rouge il utiliser ta méthode avec les trois mouvements de va-et-vient, appairage et bouton stop

Iroamstaff, 2023-05-18:
La procédure ne marche pas à tous les coups donc recommencer plusieurs fois si nécessair
télécommande et le volet doivent s'annoncer si 1 seul recommencer car il faut les 2 sur le même réseau pour que le volet fonctionne

Iroamstaff, 2023-05-18:
Dernière chose  le moteur est-il bien configuré ? Il y a une manipulation a faire décrite dans la notice du moteur profalux pour les fin de course pour qu'il enregistre l'ouverture maximum et la fermeture maximum. Et il faut aussi vérifier la configuration du moteur ( brise soleil, store, ou volet )

sven337, 2023-05-23:
I received 4 new Profalux models and they expose *both* genOnOff/genLevelCtrl on endpoint 1, and closuresWindowCovering on endpoint 2.  The remotes seem to use closuresWindowCovering so this is probably what Z2M should do also.
But some of these models don't expose closuresWindowCovering. I don't know how to write a converter that uses one or the other based on the available clusters.

closed 2023-07-02 by github-actions[bot]