Koenkk/zigbee2mqtt #17683: Sonoff ZBDongle-E Failed to interview

supagu, 2023-05-15:
### What happened?

I have the ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E" (I tried to get the P model but ended up with an E)

When trying to pair my devices, most are failing (here is an example device that is failing: https://www.zigbee2mqtt.io/devices/SNZB-04.html#sonoff-snzb-04)

Interview completed = False

Is the problem the software? I'm trying to work out if I need to get the P model if the problem is the E model.

Anyone got the E model working with their devices? 

### What did you expect to happen?

Pair would be successful

### Zigbee2MQTT version

1.30.4-dev commit: 1b65ff5

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11475790/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11475791/log.txt)

5olu7ion, 2023-05-15:
I Paired 13 Hue devices (motion sensors, bulps, a light strip and a dimmer), 1 Ikea Bulp and 1 Aqara Magic Cube with the dongle E. I had this issue with 3 different devices. It helped me, to bring the devices as close to the coordinator as possible (about 5 cm), and when this also didn't helped, I started to move the device a little bit around.

supagu, 2023-05-15:
Okay trying today everything just started working! I tried paring closer to the sonoff which may have done the trick.
And removed the device a few times from zigbee2mqtt each time I tried to re-pair.

closed 2023-05-15 by supagu

kikohome, 2023-11-16:
what to do -  my snzb-04 not working.    message : the device is not supported