Koenkk/zigbee2mqtt #17685: [Wrong device]: TS0207, manufacturer _TZ3000_k4ej3ww2

kzajac83, 2023-05-15:
### Link

https://a.aliexpress.com/_mMcgKl2

### Model

IH-K665

### Description

Water leak detector

### Vendor

Aubess

### Picture (link)

https://ae01.alicdn.com/kf/S043d952aecf0491aa1758a79178f28b30.jpg_640x640Q90.jpg_.webp

### Database entry

{"id":28,"type":"EndDevice","ieeeAddr":"0xa4c138e2bb7b006d","nwkAddr":27623,"manufId":4417,"manufName":"_TZ3000_k4ej3ww2","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3000_k4ej3ww2","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00259087e4","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087e4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1684145432704,"defaultSendRequestWhen":"immediate"}

### Notes

In instruction

Koenkk, 2023-05-15:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-15 by Koenkk

kzajac83, 2023-05-17:
Thank you, is fine in Edge.