Koenkk/zigbee2mqtt #17687: Failed to validate path: 'Error: spawn udevadm ENOENT' - what does this mean?

cloudbr34k84, 2023-05-15:
### What happened?

[23:02:52] INFO: Preparing to start...
[23:02:53] INFO: Socat not enabled
[23:02:53] INFO: Zigbee Herdsman debug logging enabled
[23:02:53] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 15-05-2023 23:02:55: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  15-05-2023 23:02:55: Logging to console and directory: '/config/zigbee2mqtt/data/log/2023-05-15.23-02-55' filename: log.txt
Zigbee2MQTT:debug 15-05-2023 23:02:55: Removing old log directory '/config/zigbee2mqtt/data/log/2023-05-15.22-11-34'
Zigbee2MQTT:info  15-05-2023 23:02:55: Starting Zigbee2MQTT version 1.30.4 (commit #unknown)
Zigbee2MQTT:info  15-05-2023 23:02:55: Starting zigbee-herdsman (0.14.111)
Zigbee2MQTT:debug 15-05-2023 23:02:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[26],"extendedPanID":[1,2,3,4,5,6,7,8],"networkKey":"HIDDEN","panID":48739},"serialPort":{"path":"tcp://192.168.68.103:6638"}}'
2023-05-15T13:02:55.658Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-05-15T13:02:55.659Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[172,175,49,1,59,130,80,163,137,113,236,81,47,31,97,240],"panID":48739,"extendedPanID":[1,2,3,4,5,6,7,8],"channelList":[26]},"serialPort":{"path":"tcp://192.168.68.103:6638"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-05-15T13:02:55.659Z zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.68.103:6638
2023-05-15T13:02:55.666Z zigbee-herdsman:adapter:zStack:znp:log Socket connected
2023-05-15T13:02:55.666Z zigbee-herdsman:adapter:zStack:znp:log Socket ready
2023-05-15T13:02:55.667Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-05-15T13:02:55.667Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]

### Zigbee2MQTT version

Current version: 1.30.4-1 

### Adapter firmware version

zStack3x0

### Adapter

Zigbee ESP32GW (Zigstar)

### Debug log

[23:02:52] INFO: Preparing to start...
[23:02:53] INFO: Socat not enabled
[23:02:53] INFO: Zigbee Herdsman debug logging enabled
[23:02:53] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 15-05-2023 23:02:55: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  15-05-2023 23:02:55: Logging to console and directory: '/config/zigbee2mqtt/data/log/2023-05-15.23-02-55' filename: log.txt
Zigbee2MQTT:debug 15-05-2023 23:02:55: Removing old log directory '/config/zigbee2mqtt/data/log/2023-05-15.22-11-34'
Zigbee2MQTT:info  15-05-2023 23:02:55: Starting Zigbee2MQTT version 1.30.4 (commit #unknown)
Zigbee2MQTT:info  15-05-2023 23:02:55: Starting zigbee-herdsman (0.14.111)
Zigbee2MQTT:debug 15-05-2023 23:02:55: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[26],"extendedPanID":[1,2,3,4,5,6,7,8],"networkKey":"HIDDEN","panID":48739},"serialPort":{"path":"tcp://192.168.68.103:6638"}}'
2023-05-15T13:02:55.658Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-05-15T13:02:55.659Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[172,175,49,1,59,130,80,163,137,113,236,81,47,31,97,240],"panID":48739,"extendedPanID":[1,2,3,4,5,6,7,8],"channelList":[26]},"serialPort":{"path":"tcp://192.168.68.103:6638"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-05-15T13:02:55.659Z zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.68.103:6638
2023-05-15T13:02:55.666Z zigbee-herdsman:adapter:zStack:znp:log Socket connected
2023-05-15T13:02:55.666Z zigbee-herdsman:adapter:zStack:znp:log Socket ready
2023-05-15T13:02:55.667Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-05-15T13:02:55.667Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]

cloudbr34k84, 2023-05-15:
@Koenkk i did some testing and this error dissappears when I disable zigbee_herdsman_debug:

Koenkk, 2023-05-16:
@cloudbr34k84 that makes sense because this is logged to the herdsman debug logging. What it means that it cannot autodetect the adapter, but this is not a problem since you already specified the path manually. You can ignore this error.

closed 2023-05-16 by Koenkk

cloudbr34k84, 2023-05-16:
Thanks again!

paccoe, 2024-02-07:
Hi, I have also the same problem. This is the trace, 

[15:10:17] INFO: Preparing to start...
[15:10:17] INFO: Socat not enabled
[15:10:19] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-02-07 15:10:26: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-07.15-10-26' filename: log.txt
Zigbee2MQTT:info  2024-02-07 15:10:26: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-07 15:10:26: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-02-07 15:10:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-07 15:10:27: Failed to start zigbee
Zigbee2MQTT:error 2024-02-07 15:10:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-07 15:10:27: Exiting...
Zigbee2MQTT:error 2024-02-07 15:10:27: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:284:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

Any guess?. Thanks mates

paccoe, 2024-02-07:
hi @cloudbr34k84,

How you turn to disable zigbee_herdsman_debug?
Thanks.

mingil, 2024-05-27:
how did you fix it? i have same trouble........