Koenkk/zigbee2mqtt #17690: TuYa TS011F_plug_1, Zigbee2MQTT, the power gets stuck

sved-SI, 2023-05-15:
### What happened?

Tuya smart power, the power gets stuck. I try delete plug and repair.
After first run with heater, plug show correct power and current, after turn off heater, power and current still show same value ( 1150W )
I try smart socket with tuya zigbee hub and work all normal.

[image]

### What did you expect to happen?

show correct power and current

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

20210708

### Adapter

Sonoff dongle-P

EvgeneF, 2023-05-15:
I have exactly the same. Make tests 4 times, problem in last version z2m.

jabadabadu, 2023-05-17:
I am getting also the same problem with the _TZ3000_gvn91tmx (bought as Aubess 20 Amps). However _TZ3000_hdopuwv6 I have also is not affected with this problem. Using now Z2M version: 1.30.4-1, previously everything was correct.
@sved-SI as you can see under #16709 comment https://github.com/Koenkk/zigbee2mqtt/issues/16709#issuecomment-1512369226 the discussion is in progress.

c0mplex1, 2023-05-17:
I also have exactly the same problem with 2 of my TS011_plug_1 smart plugs running with Zigbee2mqtt version 1.30.4-1
( _TZ3000_gznh2xla and _TZ3000_gvn91tmx )
The issue started after another fix...
https://github.com/Koenkk/zigbee2mqtt/issues/16709#issuecomment-1512369226

EvgeneF, 2023-05-17:
> 

Will try on weekend

unnaDmed, 2023-05-17:
> Looks like a duplicate of #16709, please try with the latest dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Tried it a few minutes ago and can confirm that it's working like a charm.

Thank you very much!

Alexofhiswalnut, 2023-05-19:
Hi,
When will there be a stable update available to fix this issue?
@Koenkk

Koenkk, 2023-05-20:
@Alexofhiswalnut issue will be fixed in the next release on 1 June.

closed 2023-05-20 by Koenkk

rablatbenjamin, 2024-01-11:
Hi. Power, current and energy meters doesn't work anymore