Koenkk/zigbee2mqtt #17695: Setup is not creating a database.db file

cloudbr34k84, 2023-05-15:
### What happened?

Im starting again as i had numerous issues that nothing was solving. However, I have noticed that my zigbee addon is not generating a database.db file. 

### What did you expect to happen?

database.db file to be created

### Zigbee2MQTT version

Zigbee2MQTT Edge

### Adapter firmware version

zStack3x0

### Adapter

Zigbee ESP32GW (Zigstar)

lux73, 2023-05-16:
check user permission folder - your user that is running z2m must have write access

cloudbr34k84, 2023-05-16:
> check user permission folder - your user that is running z2m must have write access

How do I check that?

cloudbr34k84, 2023-05-18:
> basic Linux knowledge
> 
> https://linuxize.com/post/understanding-linux-file-permissions/

finally found, VSS had hidden the file somehow

closed 2023-05-18 by cloudbr34k84