Koenkk/zigbee2mqtt #17696: Aqara Motion Sensor P1 - cannot connect

przemekgithub, 2023-05-16:
### What happened?

I have two P1 sensors, and with one of them I started to have issues - it didnt recognized movement anymore. So I resetted both of them and forced to remove from zigbee2mqtt. Now I cant completely connect them. Nothing shows in logs. Upgraded firmware on zigbee stick to newest, zigbee2mqtt is newest.
Device: https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html
Zigbee2MQTT version [1.30.4]
Coordinator type zStack3x0
Coordinator revision 20221226

### What did you expect to happen?

It would be possible to connect it again

### How to reproduce it (minimal and precise)

Reset device and force removal

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

zStack3x0

lux73, 2023-05-19:
try to restart z2m AFTER force remove the Device

check the Battery - just Aqara Devices need healthy Battery to connect the Network (anything under 3V are bad)

try to update your Coordinator - latest [Dev-Firmware 20230507](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) works really really good!

closed 2023-06-26 by github-actions[bot]

Programmierus, 2023-11-29:
For those who will still search it on Google. I don't know if it was my system specific but it helped me to pair multiple sensors:

1. Disable z2m network join
2. Wait a few minutes
3. VERY IMPORTANT - Put P1 to pairing mode by holding it's button for more than 5 seconds (z2m joining disabled!)
4. Enable z2m network join

djkemp, 2023-12-05:
I recently got 3 of these after watching all the rave reviews and tried so many times to pair them. 

The closest I got was a failed pairing which required me to force remove and then neither of the 2 I opened were recognised again after that. I must have tried without exaggeration about 100 times to get them to pair, restarting Z2M, putting them in pairing mode before enabling network join etc.

Updating HA, Z2M and the dongle firmware to no avail so had to return them in the end.
Was annoyed because I have some Aqara light switches, wireless buttons and a temp sensor that are all great.

It could have been the battery but these were brand new and that battery compartment is ridiculously difficult to open. 
Now on the search for alternative.

jjanderson, 2023-12-06:
after about 5 years of working flawlessly in deconz, I recently started to migrate al my gear over to Z2M. These particular sensors are however quite problematic with Z2M., not sure what the issue is. However, I recently started replacing all the aquara stuff with [these](https://s.click.aliexpress.com/e/_Dlvlrpj) - they come up as Tuya in Z2M and I find they work really well, they are also really really really cheap (like £3 a pop cheap)

TeHashX, 2024-01-08:
I successsfully paired a new Aqara Motion P1 and updated firmware from v0.0.0_0006 to v0.0.0_0010, then I removed device from Z2M. After trying to rejoin network it didn't work, tried more than 20 times, restarted Z2M addon, restarted Home Assistant, even rebooted the server, still no success and nothing in the debug log.
As a last resort I enabled "Permit Join" (Philips Hue Bulb) only and the device was recognised instantly. 

If someone else facing the same issue, try to select another router device, not (All) or (Coordinator)!

ankitmehtame, 2024-02-29:
> For those who will still search it on Google. I don't know if it was my system specific but it helped me to pair multiple sensors:
>
> 1. Disable z2m network join
> 2. Wait a few minutes
> 3. VERY [...]

I tried everything, but no avail. This really worked. Need to wait some time even after disabling join from Z2M. Thanks a bunch!

john-nosky, 2024-04-13:
I also had a lot of trouble getting mine to connect after hours of attempts. I have a bunch of 3rd Reality electrical outlets and it might have been them causing the issue. Once I only allowed my coordinator to permit connecting did the Aqara finally join the network.

irlstu, 2024-05-17:
+1 here.  Made dozens of attempts to pair but nothing happened.  It worked when I put the P1 into pairing mode first then put z2m into 'permit join' mode.  Joined instantly. Thanks for the tips :)

CreatureFear, 2024-10-15:
Oct 2024. +1. Same as above 

> For those who will still search it on Google. I don't know if it was my system specific but it helped me to pair multiple sensors:
> 
> Disable z2m network join
> Wait a few minutes
> VERY IMPORTANT - Put P1 to pairing mode by holding it's button for more than 5 seconds (z2m joining disabled!)
> Enable z2m network join

This **works**! However, you might need to do the operation several times.

When I tried this the first time, my sensor got paired, but then the states of the entites quickly became "unknown" and the the schema tab of Z2m addon I could see that my sensor wasn't linked to the coordinator..
I had to force delete the Aqara motion sensor in zigbee2MQTT then re add it 2 or 3 times before HA could retrieve the statuses properly.

 
_PS : This drove me mad. I was searching Logs, unplugging/plugging the Zigbee dongle, restarting MQTT, restarting, HA,  and didn't suspect at any time that following a specific order/timing solve it.
In the end I have no explanation for why my sensor stopped working. The cell still was at 3.05v and after pairing my sensor with a new one, I tried again with the original cell back in.. and it worked.
Maybe an issue was caused by a hard reboot? an auto upgrade? I don't know, I wasn't home when issues started and I lost the traces to investigate when I deleted to device in mqtt to pair it again._

PulsarFX, 2024-10-18:
> For those who will still search it on Google. I don't know if it was my system specific but it helped me to pair multiple sensors:
>
>     1. Disable z2m network join
>
>     2. Wait a few minutes
>
>     3. [...]

THANK YOU!! 
This was the only successful way after lot of tries according to the z2m readme.

LiFeTecX, 2025-06-24:
> I successsfully paired a new Aqara Motion P1 and updated firmware from v0.0.0_0006 to v0.0.0_0010, then I removed device from Z2M. After trying to rejoin network it didn't work, tried more than 20 [...]

Thank you tis was the solution for me. permit all dit not work selectecting a phillips hue light worked for me.