Koenkk/zigbee2mqtt #17699: [Feature request]: Retry connection to adapter before quitting

ict-one-nl, 2023-05-16:
### Is your feature request related to a problem? Please describe

Sometimes there seems to be a momentary loss of connectivity to my network connected zigbee adapter. Can't hone in on that, but zigbee2mqtt seems to detect that and quit immediately. This leads to all entities going to unavailable in HA.

### Describe the solution you'd like

A retry machanism that retries with incrementally longer intervals would make it more reliable. It needs to be clear that there is a problem, so logging is essential.

### Describe alternatives you've considered

-

### Additional context

-

sjorge, 2023-05-16:
You can probably handle this with systemd by restarting on a failure:
https://www.freedesktop.org/software/systemd/man/systemd.service.html#Restart=

Systemd should also be logging those restarts

ict-one-nl, 2023-05-16:
Running on Docker and that does basically that. But that's not really robust, I understand it would eventually crash, but something more graceful would be nice.

ict-one-nl, 2023-06-16:
Not stale.

ict-one-nl, 2023-07-17:
still valid.

closed 2023-08-25 by github-actions[bot]

ict-one-nl, 2023-08-25:
Not.planned?