Koenkk/zigbee2mqtt #17705: OTA updates fail and get stuck at 0% even after a reboot

benflux, 2023-05-17:
### What happened?

I updated firmware on Hue bulbs and it got stuck at 0% and now I can't update the firmware:

[image]

### What did you expect to happen?

The firmware process would progress from 0% to 100% and complete

### How to reproduce it (minimal and precise)

Update several  devices at once

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20220219

### Adapter

Electrolama zzh!

benflux, 2023-05-17:
this could have occurred whilst there was wifi interference with the Zigbee channel, I've subsequently fixed that but still can't update these devices stuck in this status

lux73, 2023-05-19:
in my opinion you should not update more than one Device at time...

as i remember correctly there will be Files to manually delete if the update gets stuck - don't know which Thread i have seen this.

pls search around the forums

benflux, 2023-05-19:
I was able to get around this issue by starting the install from the Home Assistant notification telling me that an update was available but I think it is still an issue to be solved so that other users don’t get themselves stuck like I was

closed 2023-06-26 by github-actions[bot]