Koenkk/zigbee2mqtt #17706: [New device support]: Tuya LED Controler - pairing issues/how to perform factory reset

pdrobek, 2023-05-17:
### Link

https://pl.aliexpress.com/item/1005005138965957.html

### Database entry

No pair mode available?

### Comments

Hi,
Recently i bought two LED zigbee controllers, my zigbee2mqtt infrastructure cannot see it. I guess i need to factory reset it. Can you descibe possible ways to perform factory reset. On PCB nor enclosure box there are no reset switches. Can you identify IC which used as brain of this device?

[image]

pdrobek, 2023-05-19:
Hue Remote is some kind of device? should i buy it?
i am thinking about solder some pins on IC on the board like i did on esp8266 IC.

lux73, 2023-05-19:
this one: https://www.amazon.com/Philips-Hue-Wireless-Installation-Free-Exclusive/dp/B08PKMT2DV?ac_md=2-1-VW5kZXIgJDUw-ac_d_bv_bv_bv&sbo=RZvfv%2F%2FHxDF%2BO5021pAnSA%3D%3D

but be aware, i can't guarantee this will be working with your Device ^^

if you can solder some Pins & this will work to setup your Device in pairing Mode i would prefer this cheaper Way ;-)

pdrobek, 2023-05-19:
yes it is cheaper but do you know anything about possible pinouts of this ICs?  I did not found anything on the Internet.

lux73, 2023-05-20:
no sorry, not familiar with this ^^

closed 2023-06-27 by github-actions[bot]