Koenkk/zigbee2mqtt #17709: Legrand 067775/741811 update invalid image failure

yashijoe, 2023-05-17:
### What happened?

Interface tells me there is an update
I try to upgrade and get the message “invalid image”

### What did you expect to happen?

Upgrade successfully 

### How to reproduce it (minimal and precise)

Press upgrade firmware 
Get the message:
“ Error 2023-05-17 09:18:40Update of 'refrigerator_plug' failed (Update failed with reason: 'invalid image')”

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

004a

### Adapter

Legrand 067775/741811

essegi65, 2023-05-18:
same problem.
hassioos Z2M v. 1.30.4.1.

novisys, 2023-05-25:
Same problem with my Legrand product (Cover/Power socket)

vezyr, 2023-05-30:
+1 
I've the same problem ("Invalid image"). Same Adapter, firmware version and Zigbee2MQTT version.

rossowl, 2023-05-31:
Same problem with 46 outlets. I tried setting up a custom index.json where was only a single OTA file and it reports the same error. The FW file is not designed for this device. Maybe it will work fine in the next version of Z2M, as mentioned in some earlier issue.

yashijoe, 2023-06-02:
22 outlets here with this issue

SilentT-FR, 2023-06-09:
Same problem 

ive found the issue

https://github.com/Koenkk/zigbee-OTA/commit/3b61e152fbdfefa00c2df4351a454b5b82734208

the NLP firmware was replace by a 412171 its a NLIS Firmware 

@benjR

SilentT-FR, 2023-06-09:
need to revert this lines

https://github.com/Koenkk/zigbee-OTA/commit/3b61e152fbdfefa00c2df4351a454b5b82734208#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975L1539-L1541

benjR, 2023-06-09:
> Same problem
> 
> ive found the issue
> 
> [Koenkk/zigbee-OTA@3b61e15](https://github.com/Koenkk/zigbee-OTA/commit/3b61e152fbdfefa00c2df4351a454b5b82734208)
> 
> the NLP firmware was replace by a 412171 its a 4 121 71 Contacteur connecté DRIVIA 20AX 1 module
> 
> @benjR

Yes it's for this device Legrand [412171](https://www.zigbee2mqtt.io/devices/412171.html)

SilentT-FR, 2023-06-09:
Yes but NLP the previous, was for 0 677 25 	NLP 	Prise de courant 16A 
the powersoket and the contactor is not on same firmware

SilentT-FR, 2023-06-10:
i have create PR : https://github.com/Koenkk/zigbee-OTA/pull/322

closed 2023-06-12 by Koenkk

yashijoe, 2023-06-12:
Hello Koenkk,
With Edge version I get this error:

Error 2023-06-12 20:57:20Update of 'lcd' failed (ENOENT: no such file or directory, open '/config/zigbee2mqtt/https:/github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-004e4203-NLP-78.zigbee

Thanks!

SilentT-FR, 2023-06-12:
@Koenkk I think you have dont check my review : https://github.com/Koenkk/zigbee-OTA/pull/322/files#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975R2751

SilentT-FR, 2023-06-12:
I think we can reopen the issue @Koenkk 

Ive i test to update via "Legrand Gateway" a power socket 'NLP' 
the gateway update my powersocket in v78

i have test to MITM the gateway but, its not provide the link and make "HTTP Partial Content"

But after that Legrand open the library firmware

i have created PR with all new firmware found on this website

BUT 

This message from my v78 Power Socket rejoining my own zigbee Network

debug 2023-06-12 23:24:33Got OTA request '{"fieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":5129215}'

as you can see
https://github.com/Koenkk/zigbee-OTA/blob/master/index.json#L1535

node script detect :  "fileVersion": 5128707,

https://static.developer.legrand.com/files/2023/05/1021-0011-004e4203-NLP.zip
from the legrand server is the v78
[image]

and from my non-update powersocket

debug 2023-06-12 23:26:47Got OTA request '{"fieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":4867071}'

The version is older than the detect version and the version of zigbee message
but its continue with invalid image

debug 2023-06-12 23:40:57Error: Update failed with reason: 'invalid image' at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.js:406:24

So Legrand lieing me ?

The firmware on website is not the same ?

SilentT-FR, 2023-06-12:
Its have a forum, i thinks leaving a post on them shortly

SilentT-FR, 2023-06-12:
If other user have other product if we can test all of them firmware @benjR

yashijoe, 2023-06-13:
Hello Silent, I no longer have the gateway as it should be useless :-)
I've updated the index.json as by Koenkk link ( https://github.com/Koenkk/zigbee-OTA/pull/322/files#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975R2751)
Anyway I still get: 
Update of 'lcd' failed (ENOENT: no such file or directory, open '/config/zigbee2mqtt/https:/github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-004e4203-NLP-78.zigbee')

Just to double check, this is the related code in the index.json:

`    {
        "fileVersion": 5128707,
        "fileSize": 250871,
        "manufacturerCode": 4129,
        "imageType": 17,
        "sha512": "1fd2b8bb2db0970f00fc87a92b855227001e5456fec5df3e6360c133ba7053bf55c93dbd30a31b6fb672307e57a3649e7b8c1ca92f005767f7ff7704ddde32df",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-004e4203-NLP-78.zigbee",
        "path": "images/Legrand/1021-0011-004e4203-NLP-78.zigbee"
    }`

SilentT-FR, 2023-06-13:
> Hello Silent, I no longer have the gateway as it should be useless :-) I've updated the index.json as by Koenkk link ( [...]

No same problem for me

```
debug 2023-06-12 23:26:48ZigbeeOTA: downloaded main index
debug 2023-06-12 23:26:48ZigbeeOTA: Loading override index https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
debug 2023-06-12 23:26:48getNewImage for '0x00047400000b5377', meta {"fileVersion":5128707,"fileSize":250871,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-004e4203-NLP-78.zigbee","sha512":"1fd2b8bb2db0970f00fc87a92b855227001e5456fec5df3e6360c133ba7053bf55c93dbd30a31b6fb672307e57a3649e7b8c1ca92f005767f7ff7704ddde32df"}
debug 2023-06-12 23:26:48ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-004e4203-NLP-78.zigbee
debug 2023-06-12 23:26:48OTA update checksum validation succeeded for '0x00047400000b5377'
```
z2m fetching good the firmware

SilentT-FR, 2023-06-14:
NLC v31 Tested and OK
Before 

`cluster 'genOta', data '{"fieldControl":0,"fileVersion":1524227,"imageType":21,"manufacturerCode":4129}' from endpoint 1 with groupID 0`

```
    "update": {
        "installed_version": 1524227,
        "latest_version": 2048515,
        "progress": 9.78,
        "remaining": 1491,
        "state": "updating"
    },
```

`debug 2023-06-14 10:03:17Received Zigbee message from '0x000474000016f15c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":2049023,"imageType":21,"manufacturerCode":4129}' from endpoint 1 with groupID 0`

```
    "update": {
        "installed_version": 2049023,
        "latest_version": 2048515,
        "state": "idle"
    },
```

yashijoe, 2023-06-15:
Hello Silent. It’s not clear to me how did you upgrade: with gateway or through zigbee2mqtt 
Thanks

SilentT-FR, 2023-06-15:
With NLC two method for compare the file version

With NLC two method also, but Z2M didnt works and by the legrand gateway works

yashijoe, 2023-06-16:
I’ve modified the index.json with this simple code:
[
    {
        "url": "1021-0011-004e4203-NLP-78.zigbee",
        "force": true
    }
]

I’ve uploaded the firmware from Koenkk repo. Result is timeout invalid image
I’ve uploaded the firmware from Legrand repo. Result is timeout invalid image

SilentT-FR, 2023-06-16:
Yes i have the same probleme the image from Legrand is invalid...

essegi65, 2023-06-17:
I still have problems updating the BTICINO K4531C device (invalid_image) which on Z2M is seen as LEGRAND 067775/741811.
There is a solution?
thanks

yashijoe, 2023-06-17:
Not yet. Also dev version doesn’t work

SilentT-FR, 2023-06-18:
@essegi65 No solution at the moment,
i have upload the firmware come from Legrand Website, and it doesn't works...

i have open a forum post https://developer.legrand.com/forums/topic/connected-outlet-nlp-78-firmware/

yashijoe, 2023-06-19:
Hello Silent. Do you think that we should open somehow the thread again?
I wonder if this closed one is no more read by admin. Thanks

yashijoe, 2023-06-19:
Silent, do you think we can intercept the firmware that the original gateway download?
Maybe via adguard

benjR, 2023-06-19:
I wonder if it's not an issue with how zigbee2mqtt recognizes the firmware and choose to apply it to a device rather than an issue on the fw file ?

SilentT-FR, 2023-06-19:
> Silent, do you think we can intercept the firmware that the original gateway download?

i have tested to sniffing the gateway but contrary to this how to 
https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Corner_Retreiving-Legrand-Firmware.md

i see the url but not the same and i cant use it, if you want i have the wireshark, i think legrand change this method to download because i see multiple time the same url but with partial content response

or is not the same method with a Legrand gateway like this https://www.legrand.fr/pro/catalogue/pack-de-demarrage-drivia-with-netatmo-pour-installation-connectee-1-module-control compared with https://www.legrand.fr/catalogue/interrupteur/kit-de-demarrage-pour-connecter-une-premiere-piece-avec-4-produits-celiane-with-netatmo-avec-plaque-blanc

SilentT-FR, 2023-06-19:
> Hello Silent. Do you think that we should open somehow the thread again? I wonder if this closed one is no more read by admin. Thanks

Yes but in https://github.com/Koenkk/zigbee-OTA project

yashijoe, 2023-06-19:
Ok. Done. Feel free to add your experience and comments

thanks

Loic691, 2023-08-29:
Hi all
I have the same problem for 41217 Legrand device
Update failed with invalid image

For 067774, the message is device didn't responding to OTA request

marcbernis, 2023-09-02:
Hello I've same issue on 412171 Zigbee2Mqtt 1.33.0
OTA update fail.
Temporary fix:
       - push pair button on device >5s  till it left network (flash red) 
       - Go to Z2M & activate pairing
       - push pair button < 1s
       - Immediately after joining go to OTA update and click on  the update 
       - It start updating

Loic691, 2023-09-05:
Update fails always with 1.33.0 for the 067774 module legrand / drivia

Koenkk, 2023-09-05:
The `invalid image` comes from the device so either:
- The OTA is invalid
- There is an issue with the device

Did you try the workaround from @marcbernis

marcbernis, 2023-09-19:
Hello denis,
You'r right the temporary fix I proposed do not work anymore (while I updated a previous version with it)
Here are my observations:
     1/ Tried update from "normal operations" device in operation for a while : update immediately failed
     2/  Powered down device and immediately update (previous fix ok too) : Update start and go 100%... and did not update
 Below are the 2 partial logs (full log attached)

1/ Immediately failed if not powered down :
Debug 2023-09-19 12:48:52Received Zigbee message from 'Chauffe Eau Legrand', type 'readResponse', cluster 'genBasic', data '{"dateCode":" \u000020220706\u0000\u0000\u0000\u0000\u0000","swBuildId":"0025"}' from endpoint 1 with groupID 0
Debug 2023-09-19 12:49:52Update of 'Chauffe Eau Legrand' failed (Error: Device didn't respond to OTA request)

2/ After power down update 100% but did not apply update (or version bad reported):
Info 2023-09-19 13:20:09MQTT publish: topic 'zigbee2mqtt/Chauffe Eau Legrand', payload '{"auto_mode":"off","device_mode":"switch","linkquality":98,"power":0,"state":"OFF","update":{"installed_version":2441731,"latest_version":5325315,"progress":100,"remaining":18,"state":"updating"},"update_available":null}'

Debug 2023-09-19 13:21:10Device 'Chauffe Eau Legrand' announced itself
Info 2023-09-19 13:21:10MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Chauffe Eau Legrand","ieee_address":"0x0004740000854ff5"},"type":"device_announce"}'
Info 2023-09-19 13:21:10Finished update of 'Chauffe Eau Legrand'
Info 2023-09-19 13:21:10Configuring 'Chauffe Eau Legrand'
==================================================================================
Info 2023-09-19 13:21:10MQTT publish: topic 'zigbee2mqtt/Chauffe Eau Legrand', payload '{"auto_mode":"off","device_mode":"switch","linkquality":98,"power":0,"state":"OFF","update":{"installed_version":5325315,"latest_version":5325315,"state":"idle"},"update_available":null}'
==================================================================================
Debug 2023-09-19 13:21:11Received Zigbee message from 'Chauffe Eau Legrand', type 'readResponse', cluster 'genBasic', data '{"dateCode":" \u000020220706\u0000\u0000\u0000\u0000\u0000","swBuildId":"0025"}' from endpoint 1 with groupID 0
Info 2023-09-19 13:21:11Device 'Chauffe Eau Legrand' was updated from '{"dateCode":" \u000020220706\u0000\u0000\u0000\u0000\u0000","softwareBuildID":"0025"}' to '{"dateCode":" \u000020220706\u0000\u0000\u0000\u0000\u0000","softwareBuildID":"0025"}'
Info 2023-09-19 13:21:11MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":" \u000020220706\u0000\u0000\u0000\u0000\u0000","software_build_id":"0025"},"id":"Chauffe Eau Legrand","to":{"date_code":" \u000020220706\u0000\u0000\u0000\u0000\u0000","software_build_id":"0025"}},"status":"ok","transaction":"qxcoh-2"}'

So, I suppose there are 2 issues 1/ updating device failed when not initialized or power down before 2/ After power down or reset, the device receive update but it is not applied or bad reported.

@Koenk could you have a look to logs ?

Many thanks for your support
Marc

[LegrandDriviaUpdateIssueLogs09192023.txt](https://github.com/Koenkk/zigbee2mqtt/files/12659383/LegrandDriviaUpdateIssueLogs09192023.txt)

War3333, 2023-09-28:
> Hello I've same issue on 412171 Zigbee2Mqtt 1.33.0 OTA update fail. Temporary fix: - push pair button on device >5s till it left network (flash red) - Go to Z2M & activate pairing - push pair button < 1s - Immediately after joining go to OTA update and click on the update - It start updating

I can confirm that this method works on 
Legrand (mine is Bticino) | 412170 | 20220519 | 0025

I had to try several times to make it work.

r0m1CH, 2024-03-11:
I'm encounter same issue. I can update my Legrand device. Any tips on how to do it ?
I have some 412171, 064882 and 412170 devices.
I tried almost all methods without any success.
Any way to handle this ?

Thanks

Loic691, 2024-03-12:
For me, I had updated my 412171 few month ago with Z2M.
I don't remember the Z2M version and my dongle firmware version.
It was complex. 3 ou 4 times try to upload...
My actual problem is to update legrand batteryless command...

r0m1CH, 2024-03-12:
@Loic691  Which version are you now ?
Are you up to date ?

Loic691, 2024-03-12:
Now I am on last 1.36.0
Dongle E with 7.4.1

But I updated my devices with legrand gateway...
For now when i'm starting Z2M, i have several messages which says than my
firmware devices all more recent than online firmware...
I think Z2m haven't last one.

I'm testing my legrand devices with 1.36.because with batyeryless legrand
device I have issues with long delai between action on switch and having
the event in HA. it's very randomly