Koenkk/zigbee2mqtt #17716: Ota update show only philips devices

platini76, 2023-05-18:
### What happened?

[image]

### What did you expect to happen?

[image]

I think images speaks alone... why I see only philips devices on my ota updates check?

### Zigbee2MQTT version

1.30.4-dev

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

lux73, 2023-05-18:
obviously only your Philips Devices supports/have available OTA Files ^^

Koenkk, 2023-05-20:
Not all devices support OTA, see https://www.zigbee2mqtt.io/guide/usage/ota_updates.html

closed 2023-05-20 by Koenkk