Koenkk/zigbee2mqtt #17718: High CPU Usage after running over a day. Restart solves Problem

weltmeyer, 2023-05-18:
### What happened?

After some time of running zigbee2mqtt it starts taking 100% cpu usage on one core.

### What did you expect to happen?

Dont take 100% cpu usage :)

### How to reproduce it (minimal and precise)

For me, its starting zb2mqtt und let it run over some time.

I just wish i could see what in zigbee2mqtt takes that 100% Loop.
I hav it running on a raspi3b. nothing else is running there.

Installed as a service.

### Zigbee2MQTT version

1.30.3

### Adapter firmware version

20221226

### Adapter

Slaesh's

### Debug log

 
Nothing suspicious in the log file.

2023-05-18 11:17:17Succesfully changed options
Info 2023-05-18 11:17:17MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"3lzq6-3"}'
Debug 2023-05-18 11:17:20Received Zigbee message from 'zb2mq_RepeaterITKeller', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:21MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKeller', payload '{"battery":100,"humidity":25.86,"illuminance":13839,"illuminance_lux":24,"last_seen":1684401440762,"linkquality":109,"occupancy":true,"pressure":1002,"state":"ON","temperature":20.63}'
Debug 2023-05-18 11:17:21Received Zigbee message from 'zb2mq_RepeaterBüro', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:22MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterBüro', payload '{"battery":100,"humidity":33.2,"illuminance":5887,"illuminance_lux":4,"last_seen":1684401441720,"linkquality":69,"occupancy":true,"pressure":1001,"state":"ON","temperature":21.58}'
Debug 2023-05-18 11:17:22Received Zigbee message from 'zb2mq_BEM_Treppenhaus_OG', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":25806}' from endpoint 2 with groupID 0
Info 2023-05-18 11:17:22MQTT publish: topic 'zigbee2mqtt/zb2mq_BEM_Treppenhaus_OG', payload '{"battery":100,"illuminance":25806,"illuminance_lux":381,"last_seen":1684401442221,"led_indication":false,"linkquality":43,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":18.54,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":false}'
Debug 2023-05-18 11:17:23Received Zigbee message from 'zb2mq_BEM_Treppenhaus_OG', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1741}' from endpoint 2 with groupID 0
Info 2023-05-18 11:17:23MQTT publish: topic 'zigbee2mqtt/zb2mq_BEM_Treppenhaus_OG', payload '{"battery":100,"illuminance":25806,"illuminance_lux":381,"last_seen":1684401443157,"led_indication":false,"linkquality":43,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":17.41,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":false}'
Debug 2023-05-18 11:17:28Received Zigbee message from 'zb2mq_purifier_01', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"particulateMatter25Measurement":46}' from endpoint 1 with groupID 0
Info 2023-05-18 11:17:28MQTT publish: topic 'zigbee2mqtt/zb2mq_purifier_01', payload '{"air_quality":"poor","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":2,"fan_state":"ON","last_seen":1684401448476,"led_enable":true,"linkquality":116,"pm25":46,"replace_filter":false,"update":{"installed_version":69633,"latest_version":69633,"state":"idle"},"update_available":false}'
Debug 2023-05-18 11:17:29Received Zigbee message from 'zb2mq_RepeaterBüro', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:29MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterBüro', payload '{"battery":100,"humidity":33.2,"illuminance":5887,"illuminance_lux":4,"last_seen":1684401449488,"linkquality":65,"occupancy":true,"pressure":1001,"state":"ON","temperature":21.58}'
Debug 2023-05-18 11:17:30Received Zigbee message from 'zb2mq_RepeaterITKammer_OG', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Debug 2023-05-18 11:17:30Received Zigbee message from 'zb2mq_RepeaterITKammer_OG', type 'attributeReport', cluster 'genBinaryOutput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:31MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKammer_OG', payload '{"battery":100,"humidity":24.72,"illuminance":15827,"illuminance_lux":38,"last_seen":1684401450925,"linkquality":247,"occupancy":true,"pressure":1001,"state":"ON","temperature":25.38}'
Info 2023-05-18 11:17:31MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKammer_OG', payload '{"battery":100,"humidity":24.72,"illuminance":15827,"illuminance_lux":38,"last_seen":1684401450925,"linkquality":247,"occupancy":true,"pressure":1001,"state":"ON","temperature":25.38}'
Debug 2023-05-18 11:17:34Received Zigbee message from 'zb2mq_S01_bar_links', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-05-18 11:17:34MQTT publish: topic 'zigbee2mqtt/zb2mq_S01_bar_links', payload '{"last_seen":1684401454642,"linkquality":65,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":587765297,"latest_version":587765297,"state":"idle"},"update_available":false}'
Debug 2023-05-18 11:17:35Received Zigbee message from 'zb2mq_RepeaterBüro', type 'attributeReport', cluster 'genBinaryOutput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:36MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterBüro', payload '{"battery":100,"humidity":33.2,"illuminance":5887,"illuminance_lux":4,"last_seen":1684401455720,"linkquality":65,"occupancy":true,"pressure":1001,"state":"ON","temperature":21.58}'
Debug 2023-05-18 11:17:43Received Zigbee message from 'zb2mq_RepeaterITKeller', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:44MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKeller', payload '{"battery":100,"humidity":25.86,"illuminance":13839,"illuminance_lux":24,"last_seen":1684401463774,"linkquality":109,"occupancy":true,"pressure":1002,"state":"ON","temperature":20.63}'
Debug 2023-05-18 11:17:46Received Zigbee message from 'zb2mq_RepeaterITKeller', type 'attributeReport', cluster 'genBinaryOutput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:47MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKeller', payload '{"battery":100,"humidity":25.86,"illuminance":13839,"illuminance_lux":24,"last_seen":1684401466792,"linkquality":109,"occupancy":true,"pressure":1002,"state":"ON","temperature":20.63}'
Debug 2023-05-18 11:17:47Received Zigbee message from 'zb2mq_purifier_02', type 'attributeReport', cluster 'manuSpecificIkeaAirPurifier', data '{"fanSpeed":20}' from endpoint 1 with groupID 0
Info 2023-05-18 11:17:48MQTT publish: topic 'zigbee2mqtt/zb2mq_purifier_02', payload '{"air_quality":"unhealthy","child_lock":"UNLOCK","fan_mode":"auto","fan_speed":3,"fan_state":"ON","last_seen":1684401467754,"led_enable":true,"linkquality":120,"pm25":73,"replace_filter":false,"update":{"installed_version":69633,"latest_version":69633,"state":"idle"},"update_available":false}'
Debug 2023-05-18 11:17:50Received Zigbee message from 'zb2mq_RepeaterITKeller', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:17:51MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterITKeller', payload '{"battery":100,"humidity":25.86,"illuminance":13839,"illuminance_lux":24,"last_seen":1684401470807,"linkquality":109,"occupancy":true,"pressure":1002,"state":"ON","temperature":20.63}'
Debug 2023-05-18 11:17:52Received Zigbee message from 'zb2mq_Buero_Deckenlicht_3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":553804336,"imageType":20496,"manufacturerCode":4635}' from endpoint 1 with groupID 0
Info 2023-05-18 11:17:52MQTT publish: topic 'zigbee2mqtt/zb2mq_Buero_Deckenlicht_3', payload '{"brightness":254,"color":{"h":40,"hue":40,"s":49,"saturation":49,"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"last_seen":1684401472724,"linkquality":109,"p [... cut ...] dle"},"update_available":false}'
Debug 2023-05-18 11:18:05Received Zigbee message from 'zb2mq_RepeaterBüro', type 'attributeReport', cluster 'genBinaryOutput', data '{"presentValue":1,"statusFlags":8}' from endpoint 8 with groupID 0
Info 2023-05-18 11:18:06MQTT publish: topic 'zigbee2mqtt/zb2mq_RepeaterBüro', payload '{"battery":100,"humidity":33.2,"illuminance":5887,"illuminance_lux":4,"last_seen":1684401485769,"linkquality":61,"occupancy":true,"pressure":1001,"state":"ON","temperature":21.58}'

lux73, 2023-05-18:
which nodeJS Version do you use?

`node -v`

weltmeyer, 2023-05-18:
root@zigbee2mqtt:~# node -v
v14.21.1

lux73, 2023-05-18:
this is quite an old version ^^

pls try updating to v18.x - it's running flawless here

lux73, 2023-05-18:
it could be that you have to

`npm rebuild`

your z2m Installation after updating NodeJS

weltmeyer, 2023-05-18:
Okay. i am now on 18.x.
root@zigbee2mqtt:/tmp# node -v
v18.16.0

I did npm rebuild and started this as a service again. I will let it run for some time and come back with an update :)
Thanks for your help!

lux73, 2023-05-18:
np you're welcome

weltmeyer, 2023-05-20:
Today i noticed slow response times(1-2sec) when turning lights on/off, looking at zigbee2mqtt cpu usage it is goind up an down between 99% and 4%
[image]

Restarting zigbee2mqtt makes the responsive back to normal (nearly instand)
so it seems the nodejs version was not the problem.

lux73, 2023-05-23:
you could try to update `npm`

what's your npm Version?

`npm -v`

update with

 `npm install -g npm@9.6.7`

9.6.7 is the latest AFAIK & works for me

don't forget a `npm rebuild` your z2m installation

but i doub't this is really the Problem

lux73, 2023-05-23:
generally i have not the best experience with RasPi3b for my HomeAutomation Software ^^

using z2m with OpenHAB 3 Years ago - but really soon i realized that i have to switch the Hardware. There are several Problems that i could not sort out (drop outs, OoM Problems and more)

After buying an APU2C4 all Problems are gone

weltmeyer, 2023-05-23:
Yeah.. Everything else is running on a VM on a real server in a rack. But as i need the zigbee stick somewhat close to my zigbee devices this raspberry is my low-power zigbee station :) the apu seems to be overkill for this purpose. i will watch this thing for some time now and if it fails again or we cant find a solution using the herdsman logs i may look for something else with a low power footprint :)

lux73, 2023-05-23:
okay, make sense...  you could try another Linux Distribution on the RasPi

 /me is using **Debian** - rock solid

weltmeyer, 2023-05-25:
Today i woke up to ca crashed zigbee2mqtt... last loglines:
 ```
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,40,86,0,252,254,27,68,129,0,0,1,0,40,86,8,1,0,247,0,139,213,64,0,0,7,8,96,10,33,0,32,200,40,86,29,142] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [40,86,0] - 252 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":22056,"relaycount":0,"relaylist":[]} +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,0,40,86,8,1,0,247,0,139,213,64,0,0,7,8,96,10,33,0,32,200,40,86,29,142] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,40,86,8,1,0,247,0,139,213,64,0,0,7,8,96,10,33,0,32,200,40,86,29] - 142 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":22056,"srcendpoint":8,"dstendpoint":1,"wasbroadcast":0,"linkquality":247,"secu
rityuse":0,"timestamp":4248971,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,96,10,33,0,32,200]}} +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"r
eservedBits":0},"transactionSequenceNumber":96,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":200}],"Command":{"ID":10,"name":"rep
ort","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":22056,"endpoint":8,"linkquality":247,"groupID":0,"wasBroad
cast":false,"destinationEndpoint":1}' +2ms
  zigbee-herdsman:controller:endpoint DefaultResponse 0x00124b00225769e4/8 1(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDef
aultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +1s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b00225769e4:22056/8 (0,0,39742) +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +737ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +1s
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +1m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,203,207,1,191,177,143,254,5,69,196,203,207,1,191,177,143] +752ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,203,207,1,191,177,143,254,5,69,196,203,207,1,191,177,143] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [203,207,1,191,177] - 143 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":53195,"relaycount":1,"relaylist":[45503]} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,203,207,1,191,177,143] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [203,207,1,191,177] - 143 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":53195,"relaycount":1,"relaylist":[45503]} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +739ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +738ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +739ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +736ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +738ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +737ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +739ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +737ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +737ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +4ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +4ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +3ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +4ms
  zigbee-herdsman:controller:group Command 10 genOnOff.off({}) +4ms
  zigbee-herdsman:controller:endpoint Read 0x000d6ffffe8e68ba/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"dis
ableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +7s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":42032,"destendpoint":8,"srcendpoint":1,"clusterid":16,"transid":30,"options":0,"radius":30,"len":5,"data
":{"type":"Buffer","data":[24,224,11,10,0]}} +7s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,48,164,8,1,16,0,30,0,30,5,24,224,11,10,0,91] +7s
Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
zigbee2mqtt@zigbee2mqtt:/opt/zigbee2mqtt$

```

I have no idea what this is :)
after starting zigbee2mqtt again, everything works fine. -.-

lux73, 2023-05-25:
you sure your RasPi ist 100% stable - bad power supply? bad sd-card?

z2m is normally really very very stable what i can say ^^

any chance you could Test your Setup with different Hardware few Days?

weltmeyer, 2023-05-25:
I am a fews seconds away from buying some mini-pc and migrating to it.

https://www.amazon.de/BMAX-Windows-Computer-Ethernet-Bluetooth/dp/B0BFQFFHPM this looks okay-ish.
What do you think? should be <20w and has enough power for z2m

lux73, 2023-05-25:
completely overkill if only for z2m :-)

and i'm not sure if Linux will ship all Drivers - while specially declared for Win11

i'm using such an Board -> https://pcengines.ch/apu2.htm

weltmeyer, 2023-05-25:
Yeah. But if it is not working well with z2m i can use it as a new thin client for other things. 
There are some reports that "linux" is running well.. so i'll give it a try with ubuntu or debian.

lux73, 2023-05-25:
ah okay, then sounds good - keep us informed

weltmeyer, 2023-05-26:
I now migrated to the new mini-pc. now running with :
```
root@z2mserver:/opt# uname -ar
Linux z2mserver 5.15.0-72-generic #79-Ubuntu SMP Wed Apr 19 08:22:18 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
root@z2mserver:/opt# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
root@z2mserver:/opt# npm -v
9.5.1
root@z2mserver:/opt# node -v
v18.16.0
root@z2mserver:/opt#

```

We will see how it works.
Power usage is around 5 Watts more than before(i am only measuring the room. but there is only some network devices....)

weltmeyer, 2023-05-26:
[image]
CpuTemp is currently 44 seems okay :)

lux73, 2023-05-28:
@weltmeyer

any news? new System stable? :-)

weltmeyer, 2023-05-28:
Up to now it seems stable. what i also changed: i removed an ikea E1812 that generated >100mqtt messages per minute.. i removed it and paired it again. dunno if this has also to do with the hiccups on the pi.

weltmeyer, 2023-06-02:
Seems like this is running stable. But as i have 100+ devices over 3 Floors on it, i am now switching over to 2 POE based zigbee adapters(ESP32+CC2652P). I did not know they existed :)

Having one ZB-Network per floor seems the better way for multiple floors. Plus i dont have to power a little computer sitting away from my rack. i can now use a docker container per instance. wonderful :)

closed 2023-06-02 by weltmeyer

lux73, 2023-06-02:
exactly my experience that z2m on RasPi is not running well

glad you sort this out 👍