Koenkk/zigbee2mqtt #17720: Device with IEEE address 0x0000000000000000 pops up in device list

popy2k14, 2023-05-18:
### What happened?

An device with the IEEE address 0x0000000000000000 is shown in the device list.
When i delete it, it will get created again in a few days.
It also joins when the network is NOT open for parinig.

[image]

### What did you expect to happen?

I have no device with address nor have i opened my network at this time!

### How to reproduce it (minimal and precise)

It seems the issue occurs since mid 2022 according to @Venom84, which is using Z2M longer than me.

### Zigbee2MQTT version

1.30.3, 1.30.4

### Adapter firmware version

20230507

### Adapter

zStack3x0 on an sonoff zigbee dongle plus CC2652P

### Debug log

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/11507427/log.zip)

Venom84, 2023-05-18:
Thanks for tagging me @popy2k14
Yeah mine is back too. @Koenkk please let me know if you need additional herdsman logs and ill get some logs going for you.

popy2k14, 2023-05-19:
@Venom84 you'r welcome.

I have looked a my log and you can clearly see the join at around:

```
2023-05-18T08:33:48.356Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":52748,"extaddr":"0x0000000000000000","parentaddr":388}
2023-05-18T08:33:48.356Z zigbee-herdsman:controller:log Device '0x0000000000000000' joined
```

But dont know which device (or coordinator) is causing the join?

Koenkk, 2023-05-20:
This is very interesting indeed, the adapter is emitting a join event. Do you maybe have sniffer hardware to see what happens during this time?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

popy2k14, 2023-05-20:
@Koenkk sorry i have no sniffer hardware.

Just an idea:
Should'nt be 0x0000... and 0xffff... IEEE addr be blocked by design. I think in the wild there is no device with such an address.
Also my heart as an hw dev speaks: Sure, the issue should be solved and not be worked around :-)

popy2k14, 2023-06-20:
not solved, i have now blocked the device with 0x0000.... IEEE addr.

mhoogenbosch, 2023-10-06:
Last week all of a sudden, half my network went down. No apparent reason. After repairing all bulbs and sockets and the day after all offline reported sensors, I noticed exactly the same.

[image]

[image]

mhoogenbosch, 2023-10-06:
And on top of that, it keeps on losing connection to devices, constantly devices drop off.

popy2k14, 2023-10-06:
Ok which channel do you running your network?
Channel 25 is recommended.

MattWestb, 2023-10-07:
The IEEE address `0x0000000000000000` is used then paring devices before them is allowed in the network (its depends of paring method) but if one device is having it then its **in** the network (have getting the network key) its somthing very bad. 
So one device is reporting its self with bad IEEE **or** is some other router in the system doing it (like OSRAM plugs) and then its happening the network is more or less collapsing then its out of the spec and cant routing frames OK.

mhoogenbosch, 2023-10-08:
> Ok which channel do you running your network? Channel 25 is recommended.

I wasn't aware of this, but it is on channel 11. Why would 25 be recommended? I haven't got any overlapping wifi channels. Non the less, i'll change it to 25 and hopefully the warning message is wrong.

mhoogenbosch, 2023-10-08:
> The IEEE address `0x0000000000000000` is used then paring devices before them is allowed in the network (its depends of paring method) but if one device is having it then its **in** the network [...]

So, if I understand correctly, it is either a bad plug or lightbulb. And how do I check this? All devices seem okay. Everything works like I expect them to do. Since last Friday, the longest last seen value is of a button, which is to expected.

closed 2024-05-06 by github-actions[bot]

silkyclouds, 2024-08-14:
same issue here. not closed.

Chrisni2, 2024-11-22:
Same issue with me

dasFelix, 2025-04-24:
Have the same Issue right now