Koenkk/zigbee2mqtt #17721: Devices suddenly go offline until reboot

11harveyj, 2023-05-18:
### What happened?

Trying to figure out whats going on, but two days in a row now half of my ZigBee devices go offline and the logs start filling up with "MAC channel access failure (255)".

Whats odd is that nothing has changed in my setup for a while, no new devices or anything either and it has been running rock solid until the past couple of days.

Not all devices go offline though, a lot of them still stay online.

When the devices are in this offline state, they still seem to respond to commands like turn on/off without problem. 

Restarting Z2M has brought the devices back, although a few still remain offline (but I can still control them and the state even updates on the Z2M dashboard for some).

I've enabled debug logging and will update with logs later on.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

Sonoff USB (CC2652P)

jensstadsgaard, 2023-05-21:
Seen something like this the last days. Using 1.30.4

It started after updating HA to 2023.5.3

11harveyj, 2023-05-21:
Happened again, seemingly completely randomly in the evening.

I've had to trim down the size of this log as it was too big for GH

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11525728/log.txt)

jensstadsgaard, 2023-05-22:
After downgrading HA to 2023.5.2 everything is working as it should.

closed 2023-06-25 by pantherale0