Koenkk/zigbee2mqtt #17726: TS011F_switch_5_gang is recognized as TS011F_plug_1

robsonwa, 2023-05-18:
### What happened?

The TS011F_switch_5_gang is recognized as TS011F_plug_1, but has in documentation this model.

https://www.zigbee2mqtt.io/devices/TS011F_switch_5_gang.html

i don't know what to do to fix this.

### What did you expect to happen?

An upgrade or Kb to change model

### Zigbee2MQTT version

1.30.4 commit: unknown

### Adapter firmware version

20210708

### Adapter

Sonoff 3.0

Koenkk, 2023-05-20:
Please provide the `data/database.db` entry of this device.

Koenkk, 2023-05-23:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-23 by Koenkk

robsonwa, 2023-05-23:
Wow, Thanks and congrats for your job.