Koenkk/zigbee2mqtt #17743: [Wrong device]: EGLO 12242 is detected as different devices

ibiBgOR, 2023-05-21:
### Link

https://www.zigbee2mqtt.io/devices/12242.html

### Model

ST64

### Description

Light Bulb

### Vendor

EGLO/AwoX

### Picture (link)

https://www.zigbee2mqtt.io/devices/12242.html

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0xa4c1384811a121fc","nwkAddr":7607,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"appVersion":2,"hwVersion":63}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":63,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1684670557980,"defaultSendRequestWhen":"immediate"}

### Notes

I'm in the process of transitioning my ZHA devices to Zigbee2MQTT. I now detect, that my lightbulbs are wrongly detected. All bulbs are the displayed bulb https://www.zigbee2mqtt.io/devices/12242.html of. But the first one pairs as https://www.zigbee2mqtt.io/devices/33957.html#awox-33957 and the second and third one pair as https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948
How can i fix this issue such that the devices are detected correctly?

ibiBgOR, 2023-05-21:
After restarting (recreating) the docker container, all devices are shown as https://www.zigbee2mqtt.io/devices/33957.html#awox-33957

Koenkk, 2023-05-21:
They use the same model ID so we cannot distinguish them from each other (`TLSR82xx`), are all the features working?

ibiBgOR, 2023-05-21:
Yes, after all of them getting detected as https://www.zigbee2mqtt.io/devices/33957.html#awox-33957, all features are working as expected. I was a little bit confused as pairing as https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948 would have excluded the white color adjustment.

On a sidenote: My bigger bulb by eglo (the globe model https://www.lampenwelt.de/eglo-connect-z-e27-g80-globe-5-5w-400lm-2200-6500k.html) yields the  https://www.zigbee2mqtt.io/devices/12242.html as model.

closed 2023-05-21 by ibiBgOR