Koenkk/zigbee2mqtt #17744: IKEA E1810 5-button remote fails interview with Z2M, works fine with ZHA.

wafliron, 2023-05-21:
### What happened?

In my Home Assistant instance, I've just finished migrating all my battery-powered ZigBee devices to their own dedicated mesh (about 80 devices). This new mesh runs through Zigbee2MQTT + a Sonoff ZBDongle-P (supported by six IKEA Tradfri repeaters). My mains-powered devices remain on a separate mesh (on a different ZB channel), using ZHA + a ConBee II.

Everything involved - HA, Z2M, controller firmwares, etc - is latest stable version. The channels used by each mesh are clear of interference from WiFi signals.

**By and large this migration has gone well, but I've run into a bizarre problem with a single IKEA E1810 5-button remote (https://www.zigbee2mqtt.io/devices/E1524_E1810.html) - it simply won't pair with Z2M (after probably 40 attempts).**

**The remote is detected quickly during the Z2M pairing process, but after ~15 seconds Z2M throws an error saying "Interview Failed". There is nothing particularly helpful in the logs (attached below), even at Debug level - no useful info (that I can see anyway) beyond non-specific messages about "Interview Failed".**

**What is bizarre is that the remote worked fine previously under ZHA - _and continues to work fine if I re-pair it through ZHA_.** Albeit with one small quirk I've noticed - it takes an unusually long time to pair (under ZHA), perhaps 30-60 seconds to get through the "Initialising" and "Configuring" steps. But after that, it's perfect.

I have five other of the same 5-button remotes - they all paired / work fine with Z2M. 

Feels like maybe there's a timeout on the interview process which is more generous under ZHA than Z2M, with the Z2M one being too short? I haven't been able to find a configurable option for this in Z2M though.

I've tried everything I can think of to solve / troubleshoot on my end - don't believe I've missed anything, but just in case, here's the things I've tried myself already:

- reviewed https://www.zigbee2mqtt.io/devices/E1524_E1810.html + applied any relevant advice
- using the "Configure" function on the remote in Z2M after the failed pairing / interview process (the remote still shows up in Z2M, but in an error-ed state)
- updating the remote to the latest firmware 24.4.5 (all of them are on this latest firmware)
- putting a new battery in the remote (and noting the remote pairs/works fine under ZHA with the old battery, which is still reporting near 100%)
- cleaning the battery contact pins + bending them inward to ensure they're making good contact with the battery
- continually pushing buttons on the remote during the pairing process to ensure it's not going to sleep (noting this also isn't necessary under ZHA, where it pairs fine)
- pairing directly via the nearest Tradfri repeater, with the remote centimetres away.
- pairing directly via the controller, with the remote centimetres away.
- pairing in "All" mode.
- rebooting the VM I have HA running in, and at the same time unplugging / re-plugging the Sonoff ZBDongle-P.
- power cycling all Tradfri repeaters in the network

Thanks.

### What did you expect to happen?

Device to be paired successfully with Zigbee2MQTT.

### How to reproduce it (minimal and precise)

Go through the normal pairing process (push button inside battery cover 4 times within 5 secs) for this device using a problematic one of my devices as per "What happened" notes.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

CC2652 (SONOFF ZBDongle-P)

### Debug log

info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/sensor/0x842e14fffe5f3e56/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x842e14fffe5f3e56 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/sensor/0x842e14fffe5f3e56/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x842e14fffe5f3e56 action","state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/sensor/0x842e14fffe5f3e56/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x842e14fffe5f3e56 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/sensor/0x842e14fffe5f3e56/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"0x842e14fffe5f3e56 last seen","state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/sensor/0x842e14fffe5f3e56/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x842e14fffe5f3e56 update state","state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info 2023-05-21 23:44:10MQTT publish: topic 'homeassistant/binary_sensor/0x842e14fffe5f3e56/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x842e14fffe5f3e56/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x842e14fffe5f3e56"],"manufacturer":"IKEA","model":"TRADFRI remote control (E1524/E1810)","name":"0x842e14fffe5f3e56","sw_version":"24.4.5"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"0x842e14fffe5f3e56 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x842e14fffe5f3e56","unique_id":"0x842e14fffe5f3e56_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
[... cut ...]
info 2023-05-21 23:44:24MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe5f3e56","ieee_address":"0x842e14fffe5f3e56"},"type":"device_announce"}'
debug 2023-05-21 23:44:26Retrieving state of '0x842e14fffe5f3e56' after reconnect
error 2023-05-21 23:44:30Failed to interview '0x842e14fffe5f3e56', device has not successfully been paired
info 2023-05-21 23:44:31MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffe5f3e56","ieee_address":"0x842e14fffe5f3e56","status":"failed"},"type":"device_interview"}'

MattWestb, 2023-05-21:
> Go through the normal pairing process (push button inside battery cover 4 times within 5 secs) for this device using a problematic one of my devices as per "What happened" notes.

This is not normal paring mode of IKEA controller is touch link paring that TF is using or direct to lights.
For getting it in paring mode press the paring button fast 4 times and the LED shall start flashing and then glowing its in paring mode. Its stop flashing then being OK paired and can needing reconfigured then normally the system is not setting all things OK.

wafliron, 2023-05-22:
> This is not normal paring mode of IKEA controller is touch link paring that TF is using or direct to lights. For getting it in paring mode press the paring button fast 4 times and the LED shall [...]

I think there’s some wires crossed here - it sounds like you’re describing the same pairing process that I am? 

Regardless, can confirm I am using the correct pairing process - the one described at https://www.zigbee2mqtt.io/devices/E1524_E1810.html, and the same one that works successfully for my other 5-button remotes.

closed 2023-06-29 by github-actions[bot]