Koenkk/zigbee2mqtt #17759: [Wrong device]: Tuya Generic Light Bulbs Model

cloudbr34k84, 2023-05-22:
### Link

https://www.aliexpress.com/item/1005002656711222.html?spm=a2g0o.order_list.order_list_main.5.3d391802q1rLXn

### Model

E14 Candle Bulb

### Description

Zigbee E14 Dimmable Smart Bulb RGB+CW+WW

### Vendor

Tuya

### Picture (link)

https://i.ebayimg.com/images/g/xvYAAOSwf8xiQvZ1/s-l1600.jpg

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0xa4c138cd7ccb25b4","nwkAddr":56699,"manufId":4417,"manufName":"_TZ3210_it1u8ahz","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"lightingColorCtrl":{"attributes":{"57344":1000,"57345":[0,65536113],"61453":1,"colorCapabilities":29,"tuyaRgbMode":0,"colorMode":2,"currentX":10040,"currentY":3116,"currentHue":80,"currentSaturation":254,"colorTemperature":500}},"genLevelCtrl":{"attributes":{"61440":0,"currentLevel":127}},"genOnOff":{"attributes":{"61440":0,"onOff":0}},"genBasic":{"attributes":{"zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1684784688800,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-05-23:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-23 by Koenkk

djcrawleravp, 2024-05-17:
How can I make my Z2M detect _TZ3210_it1u8ahz as TS0505B_1_2 again? 2_2 is not working for me