Koenkk/zigbee2mqtt #17762: [Wrong device]: Another Tuya Generic Water Leak Sensor

cloudbr34k84, 2023-05-22:
### Link

https://www.aliexpress.com/item/1005003923445751.html?spm=a2g0o.order_list.order_list_main.296.3d391802q1rLXn

### Model

Zigbee Water Leak Sensor

### Description

Zigbee Water Leak Sense + Alarm (40DB)

### Vendor

Tuya

### Picture (link)

https://ae01.alicdn.com/kf/Sffdd5ba0ab424b728bc1bfaeb1a2c39ea.jpg

### Database entry

{"id":58,"type":"EndDevice","ieeeAddr":"0xa4c1384d15ca09c4","nwkAddr":29963,"manufId":4417,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3000_upgcbody","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00256cd420","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":190}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00256cd420","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1684784687423,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-05-23:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-23 by Koenkk

kzajac83, 2023-06-02:
Well, not sure how to resolve problem because until implement this issue picture and description was correct, picture and description is 100% the same as  #17761 

On box there is 898ZT model
[image]
[image]

I have 2 devices with 1m cable long without alarm and database for both are:

{"id":29,"type":"EndDevice","ieeeAddr":"0xa4c1386c7cde5d88","nwkAddr":61585,"manufId":4417,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3000_upgcbody","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00259087e4","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087e4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1685697778236,"defaultSendRequestWhen":"immediate"}

{"id":35,"type":"EndDevice","ieeeAddr":"0xa4c138da916b7897","nwkAddr":13444,"manufId":4417,"manufName":"_TZ3000_upgcbody","powerSource":"Battery","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,3,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0207","manufacturerName":"_TZ3000_upgcbody","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00259087e4","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00259087e4","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1685690058150,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-06-03:
Please provide a picture, jpg, 150x150 with a white background.

kzajac83, 2023-06-03:
Picture is

https://ae01.alicdn.com/kf/Sd5376add013f4111b65020a3ec7ec7ed6.jpg

Koenkk, 2023-06-04:
fixed, thanks!