Koenkk/zigbee2mqtt #17780: Xiaomi Aqara D1 2 gang smart wall switch (with neutral wire) QBKG24LM failed to pair

Spica2000, 2023-05-23:
### What happened?

Xiaomi Aqara D1 2 gang smart wall switch (with neutral wire) QBKG24LM failed to pair with error "Device cannot be configured" it is listed in supported devices (https://www.zigbee2mqtt.io/devices/QBKG24LM.html). I tried to pair it numerous times and it definitely works with original Xiaomi controller. I have no problems with other Aqara devices including other D1 switches (but with no neutral wire) I have.

I use Home Assistant 2023.5.3 Supervisor 2023.04.1 with zigbee2mqtt 1.30.4-1

### What did you expect to happen?

Device is paired.

### How to reproduce it (minimal and precise)

1. Enable pairing in zigbee2mqtt
2. Enter pairing mode in the QBKG24LM
3. I see "Starting interview" message
4. After some time I see the message "Device cannot be configured"

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

20220524

### Adapter

Zigbee CC2652P1

Koenkk, 2023-05-24:
I guess this happens after you press the yellow configure button. This device does not require configuring so this is not necessary. If not, please provide the debug log.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Spica2000, 2023-06-05:
Here is the logs:
Info 2023-06-05 20:37:05Device '0x00158d00067a81cb' joined
Info 2023-06-05 20:37:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00067a81cb","ieee_address":"0x00158d00067a81cb"},"type":"device_joined"}'
Info 2023-06-05 20:37:05MQTT publish: topic 'zigbee2mqtt/0x00158d00067a81cb', payload '{"last_seen":"2023-06-05T17:36:58.262Z"}'
Info 2023-06-05 20:37:05Starting interview of '0x00158d00067a81cb'
Info 2023-06-05 20:37:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00067a81cb","ieee_address":"0x00158d00067a81cb","status":"started"},"type":"device_interview"}'
Info 2023-06-05 20:37:05MQTT publish: topic 'zigbee2mqtt/0x00158d00067a81cb', payload '{"last_seen":"2023-06-05T17:36:58.443Z"}'
Debug 2023-06-05 20:37:05Device '0x00158d00067a81cb' announced itself
Error 2023-06-05 20:39:23Failed to interview '0x00158d00067a81cb', device has not successfully been paired

Koenkk, 2023-06-06:
- can you try to pair the device close to the coordinator?
- if that fails, can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Spica2000, 2023-06-11:
The device is quite close to coordinator. l've tried to swap it on same Aqara D1 but without neutral wire and pairing is OK, signal level is high.

Debug:
2023-06-11T14:39:47.477Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":115,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":224}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":2760,"endpoint":1,"linkquality":105,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-06-11 17:39:47: Received Zigbee message from 'RozetkaTV', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":224}' from endpoint 1 with groupID 0
2023-06-11T14:39:47.499Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2023-06-11 17:39:47: MQTT publish: topic 'zigbee2mqtt/RozetkaTV', payload '{"child_lock":"UNLOCK","current":0,"energy":35.75,"indicator_mode":"off/on","last_seen":"2023-06-11T14:39:47.478Z","linkquality":105,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":224}'
2023-06-11T14:39:50.897Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,133,10,0]
2023-06-11T14:39:50.897Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,133,10,0]
2023-06-11T14:39:50.898Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [13]
2023-06-11T14:39:50.899Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,133,10,0,13]
2023-06-11T14:39:50.900Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [133,10,0] - 13
2023-06-11T14:39:50.901Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":2693,"relaycount":0,"relaylist":[]}
2023-06-11T14:39:50.901Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-06-11T14:39:50.946Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6]
2023-06-11T14:39:50.946Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6]
2023-06-11T14:39:50.948Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,133,10,1,1,0,87,0,19,249,72,0,0,7,8,5,10,0,0,40,0,133,10]
2023-06-11T14:39:50.948Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,133,10,1,1,0,87,0,19,249,72,0,0,7,8,5,10,0,0,40,0,133,10]
2023-06-11T14:39:50.950Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [29,24]
2023-06-11T14:39:50.950Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,133,10,1,1,0,87,0,19,249,72,0,0,7,8,5,10,0,0,40,0,133,10,29,24]
2023-06-11T14:39:50.951Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,133,10,1,1,0,87,0,19,249,72,0,0,7,8,5,10,0,0,40,0,133,10,29] - 24
2023-06-11T14:39:50.951Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":2693,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":4782355,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,5,10,0,0,40,0]}}
2023-06-11T14:39:50.955Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":40,"attrData":0}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":2693,"endpoint":1,"linkquality":87,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-06-11 17:39:50: Received Zigbee message from 'DoilvBoiler', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
2023-06-11T14:39:50.964Z zigbee-herdsman:controller:endpoint DefaultResponse 0xb4e3f9fffec64dd7/1 6(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-06-11T14:39:50.965Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb4e3f9fffec64dd7:2693/1 (0,0,2)
2023-06-11T14:39:50.967Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":2693,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":87,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,5,11,10,0]}}
2023-06-11T14:39:50.968Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,133,10,1,1,6,0,87,0,30,5,24,5,11,10,0,243]
2023-06-11T14:39:50.969Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2023-06-11 17:39:50: MQTT publish: topic 'zigbee2mqtt/DoilvBoiler', payload '{"last_seen":"2023-06-11T14:39:50.955Z","linkquality":87,"state":"OFF"}'
2023-06-11T14:39:50.980Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2023-06-11T14:39:50.981Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
[... cut ...]
2023-06-11T14:41:50.404Z zigbee-herdsman:controller:error Interview failed for '0x00158d00067a81cb with error 'Error: Interview failed because can not get node descriptor ('0x00158d00067a81cb')'
Zigbee2MQTT:error 2023-06-11 17:41:50: Failed to interview '0x00158d00067a81cb', device has not successfully been paired
Zigbee2MQTT:info  2023-06-11 17:41:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d00067a81cb","ieee_address":"0x00158d00067a81cb","status":"failed"},"type":"device_interview"}'

Koenkk, 2023-06-12:
> The device is quite close to coordinator.

Can you try to only permit joining via the coordinator and pair again? It seems requests to the device fail (this is something I cannot fix from z2m)

Spica2000, 2023-06-12:
I'm not sure that I understood you. If you mean regular pairing process like join permit and entering in pairing on the device, I tried to do it multiple times with the same result. Or do you mean some other process?

The case that device was OK on the same place with another coordinator (Xiaomi Aqura original coordinator with alternative firmware running z2m and Home Assistance).

jg-bb, 2023-07-15:
Unstale

Hey @Koenkk , thanks for a lot of great input on your side! Have followed couple of issues although this seems to be the newest one, so decided to write here.

Basically it seems that there is a big problem with Aqara D1 switches. Both single and double button ones (QBKG23LM and QBKG24LM). The triple one works though perfectly (QBKG26LM).

Mainly the devices are always not recognised. I was able to manually put all the needed info into the database.db and then the interview seems to be fine, although it's never updating + when trying to send a request, that should be handled, there is a 'no response received' error message.

Do you think this is something that you could take a look in near future? I can support you from testing side given I have hardware and solid experience in testing. Asking because I have whole house connected to Aqara and recently lost the Mi App configuration, which is a pita for me ;-)

Spica2000, 2023-07-17:
I've tried solutions such as "push many times the buttons after initiating the pairing" and restarting the coordinator.  I've also tried suggested "enable joining only via the coordinator". Nothing works unfortunately.

closed 2023-08-25 by github-actions[bot]

danielelbatal, 2023-10-28:
Tried everything listed here and also having exactly the same issue.