Koenkk/zigbee2mqtt #17781: "not connected to mqtt sever" after a day of normal behavior

cheetahunter, 2023-05-23:
### What happened?

I have the latest  version running on hassio. suddenly started to get the message in log: "not connected to mqtt sever" after a day of normal behavior  . 
after I restart the z2m its starts running ok again untill the next time.

### Zigbee2MQTT version

1.30.4-1 

### Adapter firmware version

20210120

### Adapter

zzh

lucas-apr, 2023-06-05:
I have exact the same issue. On top of that the Z2M front end becomes unreachable and the only way to fix it is to restart z2m until it breaks again which happens multiple times a day.

myrlund, 2023-06-06:
Experiencing exactly the same problem here.

I get nothing in the logs indicating anything but normal operation, then:

```
zigbee2mqtt  | Zigbee2MQTT:error 2023-06-04 13:58:03: Not connected to MQTT server!
```

FWIW, I have tried:
1. Reinstalling, downgrading and upgrading brokers and Home Assistant, even switching from Mosquitto to EMQX for a few days with no change.
2. Switching from self-managed container install to supervised install of Home Assistant, Mosquitto and Z2M. Setting up Z2M again I copied over my config folder, so it's natural to believe that there is something off with my state that is triggering this.

Still, the exact same problem reappears each time, only a few hours from each Z2M restart until it falls down again.

Please let me know if I can help out with logs, configs or anything else to help pin this down :)

cheetahunter, 2023-06-06:
I have also tried reinstalling, downgrading and upgrading brokers  z2m and Home Assistant and nothing worked.

as for now I have downgraded z2m to 1.18.1-1  and its working ok for more than a day.
I will update if there will be any changes.

lucas-apr, 2023-06-06:
I have also noticed that when z2m goes down zigbee2mqtt/bridge/state changes to offline. Restarting z2m brings zigbee2mqtt/bridge/state back to "online".

lucas-apr, 2023-06-07:
Just did a complete Proxmox backup restore of the hass os and the issue persist. I am running out of options. I am thinking about deploying Z2M in a separte continer.

> I have also tried reinstalling, downgrading and upgrading brokers z2m and Home Assistant and nothing worked.
> 
> as for now I have downgraded z2m to 1.18.1-1 and its working ok for more than a day. I will update if there will be any changes.

Are you experiencing the issue with 1.18.1-1? 

Just did a complete Proxmox backup restore of the hass os (along with Z2M to 1.30.4-1) and the issue persist. I am running out of options. I am thinking about deploying Z2M in a separate container to see whether we have the same issue with different versions of z2m.

lucas-apr, 2023-06-08:
So it happened today again multiple times. I noticed when it happens I get high CPU utilization (over 50%) for the Z2M addon under home assistant. Further investigation has shown that node index.js  was taking all the CPU in hassio. After Z2M was rebooted node index.js came back down. Not sure whether it is helpful but this is what I have found.

myrlund, 2023-06-09:
Just a shot in the dark: did anyone else trigger OTA updates shortly before the problems started?

lucas-apr, 2023-06-09:
Yes. I have a few ikea devices and I wanted to get them updated.

cheetahunter, 2023-06-09:
Hi
I didnt trigger any ota updates before.
Version 1.18.1-1 seems to work ok although I have some devices not recognized in this version.

lucas-apr, 2023-06-11:
hello,

I replaced the sonoff usb dongle P coordinator with a new sonoff usb P dongle and I am getting the same issue. So it is not the hardware so it seems.

lucas-apr, 2023-06-13:
> On my end I have been able to resolve the issue by flashing my very
> outdated zzh! coordinator firmware to the latest version. I have at least
> had 26 hours of uptime so far, where I previously saw only a few hours at a
> time between forced restarts.

I deleted the mosquito and z2m addons and deployed them separately on
another VM in docker. So far so good. It has been 48 hours without any
issues. I was thinking about reinstalling the add-ons but I kind of like it
having them separated from home assistant.

Definitely the hardware of the USB dongle was not the issue on my end. Even
after getting a brand new dongle, brand new usb extension cable and
flashing the newest firmware did not fix the issue. It must be something
with software (HA? MQTT broker? Z2M?) but I am not sure which part of it

myrlund, 2023-06-13:
Yeah, it definitely feels like this is merely a symptom of another set of root causes causing the MQTT connection to go stale in some way.

closed 2023-07-22 by github-actions[bot]