Koenkk/zigbee2mqtt #17782: Coordinator is constantly crashing after HUE Firmwareupdate to 1.104.2

MrLight, 2023-05-24:
### What happened?

Since I have updated multiple HUE Bulbs to 1.104.2 Firmware Version my Coordinator is constantly crashing.
The log is quite empty. 
The network is running for something around 10min to 4h. 
I have reflashed the coordinator without success. I have also repaired all updated HUE Bulbs without success.
To get the network up and running again I have to replug the coordinator and restart the zigbee2mqtt container.

Before the update the network was very stable with more or less no issues.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

### Debug log

```
2023-05-24T05:44:51.547Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2023-05-24T05:44:51.547Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7
```

rossowl, 2023-05-26:
I have updated hue to same version 1.104.2 without any problems.
Check the adapter extension cable, move the adapter to a different location, unplug nearby devices. I had the same error and it made an SSD disc placed 1 meter away from the adapter.

MrLight, 2023-05-27:
I agree that it is most probably not a major issue. I'm still in the phase of investigation.
Short summary: It is definitely liked to the HUE bulbs with the new firmware. The network is stable for at least a day when all updated bulbs are disconnected. When I powerup a random upgraded bulb the network will crash in the next hours. It will speed up as more updated bulbs are available. 
I have tried to downgrade the bulbs but this seems to be blocked...
I have moved the coordinator, turned off all of my legacy OSRAM devices, restartet my routers, unlinked the HUE bulbs from binding groups, checked all modemmanager / USB power suspend optimizations on the host system...

As it seems that it is specially linked to my case I plan to start with a second zigbee2mqtt network transfering the HUE bulbs first and than all of my plenty other devices.  Hopefully after recreation the issue will be gone.

MrLight, 2023-05-30:
short Update: I reconstructed my network and the issue was gone. It's stable now, even with the HUE bulbs causing trouble. Can't explain why and haven't made deep investigations. The straight forward approach to go over zero was sufficient...

closed 2023-05-30 by MrLight

rossowl, 2023-06-04:
I had a similar issue a few days ago. My neighbor started a new mesh wifi. I changed the zigbee channel from 11 to 24 and the "SREQ --> ZDO - mgmtPermitJoinReq" errors disappeared. I turned off eventually reset all router devices and used the same pan_id, ext_pan_id and network_key. I had to remove the antenna from the coordinator and cover coordinator with aluminum foil. After reboot Z2M, everything paired up on channel 24 nicely. So far 120 devices. Now 17 IKEA, 41 Legrand, 13 Lidl, 28 Philips Hue, 20 Xiaomi. About 50 more devices to go yesterday. I hope this helps.

rossowl, 2023-06-05:
And try more powerful stick. I have found that sonoff works up to about 150 devices with no problems. 117 are routers. Then every other pairing crash with an error and z2m reboots. Then you need to wait for all paired devices to connect themselves. Any activity during that time will result with an error and z2m restarts. I replaced it with Electrolama and it works with 172 devices without error and the network is very stable. If anything changes in a few days, I will post it here.