Koenkk/zigbee2mqtt #17785: K4027C/L4027C/N4027C/NT4027C not updating status

federicosantolini, 2023-05-24:
### What happened?

When my window (connected to the bticino device) changes its state, it is not reported to Z2M until i click the refresh button on the "Exposes" tab. this happens if i click on the physical button, by voice via Alexa\Google and if i open\close it from Z2M\HA.

### What did you expect to happen?

I'd like to have the state\position reported.

### How to reproduce it (minimal and precise)

open\close the window\cover

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

26720700

### Adapter

Conbee2

Koenkk, 2023-05-24:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/bcd1df5573e7504fe35dfe9fc237104f

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

federicosantolini, 2023-05-24:
You are always the best! The converter fixed it! thanks!

closed 2023-05-24 by federicosantolini

Koenkk, 2023-05-25:
Great, integrated the fix

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

developerGM, 2023-09-07:
@federicosantolini 
can you set the percentage to open/close the device using zigbee2mqtt?
because I can only open or close (0, 50, 100)

altrdev, 2023-09-07:
I have the same problem, if i click on the physical button the state is not updated until I click the refresh button on the "Exposes" tab. 
z2m version 1.33-0.1 .

developerGM, 2023-09-07:
My problem is different. 
I can't set a percentage to open/close.

@altrdev If you want open at 30%, your K4027C works?

federicosantolini, 2023-09-07:
@developerGM it should depend on the firmware version you have

developerGM, 2023-09-07:
Firmware 20211215 and version 0036
If I try to launch ota update I recive this message 

Firmware on xxxx is newer than last firmware online