Koenkk/zigbee2mqtt #17799: OTA updates: Disparity between version displayed in frontend and `state`

SHxKM, 2023-05-26:
### What happened?

For two different devices having the exact same model, where `update.installed_version` and `update. latest_version` values are `X` (the same), one of the devices in the frontend is showing one version, and the other is showing a different one.

I have several [Hue Econic outdoor wall lamp](https://www.zigbee2mqtt.io/devices/1743830P7.html#philips-1743830p7). These recently received firmware updates.

One (actually several) of those devices, `A`, is showing it has version `1.104.2` in the frontend. When I click on "check firmware version" it shows no updates available. This is its state tab contents (only relevant parts):

```
{
    "update": {
        "installed_version": 16785666,
        "latest_version": 16785666,
        "state": "idle"
    },
    "update_available": false,
}
```

Another device of the same type, `B`, is showing it has version `1.93.11` in the frontend. When I click on "check firmware version" it shows no updates available. This is its state tab contents (only relevant parts):

```
{
    "update": {
        "installed_version": 16785666,
        "latest_version": 16785666,
        "state": "idle"
    },
    "update_available": false,
}
```

As you can see these are exactly the same, so I don't understand where Z2M frontend is pulling the different versions from (1.93.11, 1.104.2).

Note: this has happened after an OTA was interrupted due to a Z2M restart.

### What did you expect to happen?

If two devices have the same version, I expect that to reflect in the frontend.

### How to reproduce it (minimal and precise)

Not sure.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

0x26580700

### Adapter

Conbee II

SHxKM, 2023-05-26:
Since posting this I discovered this is not exclusive to one model / one firmware version but has happened with other Philips Hue lights models and firmware versions randomly. For example I have 10 spotlights, 9 are on `1.104.2`, and one is at `1.101.10`. When I click on this one to check for updates it says there are none, and its version in the “State” is identical to the others.

Koenkk, 2023-05-26:
Can you provide the debug log when checking for fw updates for both?

SHxKM, 2023-05-27:
> Can you provide the debug log when checking for fw updates for both?

Hi sorry, I don’t recall the logs containing anything special. Anyway this seems to be a “known” problem and I used [this method](https://github.com/Koenkk/zigbee2mqtt/issues/17740#issuecomment-1556146513) to bust Z2M’s cache, or whatever was the cause for the confusion.

sjorge, 2023-05-28:
You can try reading seBuilfId and dateCode attributes from the genBasic cluster on endpoint 12 doe the hue bulbs showing the old version but no update available.

Sometimes after an OTA it seems that info is not properly refreshed.

closed 2023-07-06 by github-actions[bot]