Koenkk/zigbee2mqtt #17800: [New device support]:  Immax NEO 07505L

fireport, 2023-05-26:
### Link

https://www.luciamo.it/immax-neo-07505l-tastiera-intelligente-smart-zigbee-3-0-3xaaa-4-5v/

### Database entry

{"id":17,"type":"EndDevice","ieeeAddr":"0xa4c138625f46a98b","nwkAddr":29638,"manufId":4417,"manufName":"_TZE200_nyvavzbj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685094081269,"defaultSendRequestWhen":"immediate"}

### Comments

The package and the exterior aspect are pretty similar to immax  neo 07505l but zigbee2mqtt doesn't identify it because tuya fingerprint changes from _TZE200_n9clpsht to _TZE200_nyvavzbj. I created an external converter but I am unable to detect sos and disarm buttons.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {

        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_nyvavzbj']),
        model: '07505L',
        vendor: 'Immax',
        description: 'Neo smart keypad',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [],
        exposes: [e.action(['disarm', 'arm_home', 'arm_away', 'sos']), e.tamper()],
        meta: {
            tuyaDatapoints: [
                [24, 'tamper', tuya.valueConverter.trueFalse1],
                [26, 'action', tuya.valueConverter.static('disarm')],
                [27, 'action', tuya.valueConverter.static('arm_away')],
                [28, 'action', tuya.valueConverter.static('arm_home')],
                [29, 'action', tuya.valueConverter.static('sos')],
            ],

}
};

module.exports = definition;
```

Claudinetto2, 2023-06-04:
I have the same product and the same issue. Have you find a solution?

fireport, 2023-06-22:
I returned this keypad and I bought a develco keyzb-110 [https://www.zigbee2mqtt.io/devices/KEYZB-110.html#develco-keyzb-110](https://www.zigbee2mqtt.io/devices/KEYZB-110.html#develco-keyzb-110) that works like a charms.

closed 2023-07-30 by github-actions[bot]

gianlucasullazzo, 2024-05-21:
Hi, similar problem here, fingerprint is _TZE200_moycceze.
No way to make it work with custom converter. Any idea?

martin1997nm, 2024-06-03:
@gianlucasullazzo 
Please try my converter , i made it specificly to _TZE200_moycceze.
https://github.com/Koenkk/zigbee2mqtt/issues/20273#issuecomment-2145158257

gianlucasullazzo, 2024-06-03:
> No unfortunatelly , i tried it but right now im in contact with the producer to make this possible :)

Ok, great. 
Just FYI, I changed 2nd user code and now I see only the second one in the UI, but both codes are working fine.

martin1997nm, 2024-06-04:
Thx for the user manual, i had a different one, now i know what datapoint 112 i! :D Whit this modification you can get the user id from 0-9 as well, but the user code that you can get from the device is the last added user code. Im sad that the device not support sending this in one object to know the user id the user code and the action it self. Im updating the original post whit the code .

gianlucasullazzo, 2024-06-04:
great :) Do you know if user_id changes on action triggered? I'm not at home in this moment so I can't test.

martin1997nm, 2024-06-05:
Yes it changes! For example 2345 is usercode 2 so if you arm whit 2345 the message will be user_id 2 , action arm_away