Koenkk/zigbee2mqtt #17804: Unrecognized DP's for Woox R7060

Calimerorulez, 2023-05-26:
### What happened?

Hi,

I just added a Woox R7060 irrigation valve, and the following warnings appeared in the log:

```
2023-05-26 15:50:26zigbee-herdsman-converters:WooxR7060: Unrecognized DP #107 with data {"dp":107,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Info 2023-05-26 15:50:26MQTT publish: topic 'zigbee2mqtt/0x9035eafffexxxxxx', payload '{"battery":99,"last_seen":"2023-05-26T15:50:26+02:00","linkquality":29,"state":"OFF"}'
Warning 2023-05-26 15:50:26zigbee-herdsman-converters:WooxR7060: Unrecognized DP #106 with data {"dp":106,"datatype":3,"data":{"type":"Buffer","data":[48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48]}}
```

The valve does work though.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20211226

### Adapter

CC1352P

Koenkk, 2023-05-26:
Already fixed in #17615

closed 2023-05-26 by Koenkk

Calimerorulez, 2023-06-11:
> @Calimerorulez hey, can I ask where you purchased the Woox R7060 from? It seems to be a good model...

Hi @877dev,

I've bought mine at https://www.123led.nl/WOOX-R7060-Slimme-Zigbee-waterregelaar-Besproeiingscomputer-i7363-t5694.html.

I've also bought a RTX ZVG1, which tends to be a bit more stable, but longer in dimensions. The RTX had more Zigbee features than the Woox, if that's your thing ;)