Koenkk/zigbee2mqtt #17805: [New device support]: Smart Switch Model ZG-002 RF 1Ch AC 85-250v / USB 5v from Tuya

robsonwa, 2023-05-26:
### Link

https://pt.aliexpress.com/item/1005005088390765.html?pdp_npi=2%40dis%21BRL%21R%24121%2C23%21R%2472%2C74%21%21%21%21%21%402103250716851337090942361e4dbf%2112000031604593172%21btf&_t=pvid%3A6371c8a2-9f64-4ec9-9218-8caaf3b2eeed&afTraceInfo=1005005088390765__pc__pcBridgePPC__xxxxxx__1685133709&spm=a2g0o.ppclist.product.mainProduct&gatewayAdapt=glo2bra

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x00124b0024707016","nwkAddr":5145,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"ZB-SW01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"ZB-SW01","manufacturerName":"eWeLink","powerSource":1,"zclVersion":1,"appVersion":8,"hwVersion":1,"dateCode":"20210703","swBuildId":"1.0.7"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":8,"hwVersion":1,"dateCode":"20210703","swBuildId":"1.0.7","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1685133901457,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
I dont find anythings about this device.

Thanks a lot.

Koenkk, 2023-05-27:
This device is supported in the latest version of z2m: https://www.zigbee2mqtt.io/devices/ZB-SW01.html

closed 2023-05-27 by Koenkk

robsonwa, 2023-05-27:
Hi Koenkk, its not correct. Thats diferent, this have 5v input, and control If use from switch or 1 sec pulse for garage door for example

If you need any details tell me please.

thanks a lot

Koenkk, 2023-05-28:
I don't understand what you mean, can you explain your problem?

robsonwa, 2023-05-29:
I'm sorry Koenkk, you alright, just image from device is diffent but all function is OK.

Thanks a lot.

Koenkk, 2023-05-30:
Unfortunately we cannot distinguish this device as it uses the same model ID as other devices (`ZB-SW01`)