Koenkk/zigbee2mqtt #17806: Sensors stop communicating with Z2M after some time

beped, 2023-05-27:
### What happened?

My sensors stop communicating with the Z2M after some time. Sometimes that time is a few minutes, sometimes a few hours, sometimes it's 2 days. It's never accurate, but after a while they all need to be paired again, sometimes having to delete them from the device list and re-enter.

### What did you expect to happen?

What sensors i have:
8 x ZBSA-MS (inside Z2M it shows as SNZB-03) https://www.aliexpress.com/item/1005002468620848.html?spm=a2g0o.order_list.order_list_main.52.2f991802aZKwBg
2x SONOFF SNZB-01
2x SONOFF SNZB-02
6x SONOFF SNZB-03

Some of those sensors are 20cm from the coordinator, even so they eventually stop responding.

What has been tested so far:

- Changed the position of the zigbee dongle to see if some kind of interferense
- Changed the dongle firmware to a newer and even to a older version to see if theres any difference.
- Changed all sensors batteries
- Paired using the specific permit join (nearest router)

My map:
[image]

Devices:
Btw, Motion 2 is 30 cm from the coordinator
[image]

All sensors and routers are in a area of 15m x 10m well distributed with routers in every room.
I dont have any issues with the routers, only the sensors.

### Zigbee2MQTT version

1.30.4-1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee dongle P (with a extension cable)

beped, 2023-06-10:
in my case it made no difference to upgrade the routers. I currently downgraded and it is running on version 20221226. But the problem continues, eventually the sensors stop communicating with the HA and only work again after pairing again.

Raemsna, 2023-06-22:
Same issue here with SNZB-03. I have tried two different sensors and as OP sais here, they stop communicating after some time. 
Also, the time for the stop of communication can not be defined clearly. 
To make it work again, i have to unpair a SNZB-03 device and pair it again. After that it works for some period unless it stops again

Sonoff Dongle zStack3x0, coordinator version: 20221226, 1.31.2

Raemsna, 2023-06-26:
are there any logs we can provide to get a hint on this error?

marc-gist, 2023-07-12:
have you all checked that you do not have WiFi interference?  I used to have a very unstable network until I made sure that my WiFi routers all ran on channels far away from zigbee channel.

denis-obukhov, 2023-07-28:
@beped I even can't use my SNZB-03 motion sensor because of the [crash](https://github.com/Koenkk/zigbee2mqtt/issues/18400) on connect. What version of zigbee2mqtt are you using?

linuxstony, 2023-08-14:
I would like to bump this issue, because this is exactly my problem.

I have SNZB-01  SNZB-02  which stop communicating after some time. SNZB-03 however are used frequently (they are used on our main staircase) and work without problems.

The Button and Temperature-Sensors however stop working after some time. 

I am willing to provide you with all the details you need.

Aqara sensors work flawlesly - sadly they dont have the button type of device  :(

linuxstony, 2023-08-17:
I would like to add that my problem is only with Sonoff sensors - I use a LOT of devices and only the products from sonoff leave the "party" after some time. 

I am sorry to say that in this state you should remove the sonoff sensors from "supported devices"

Zigbee2MQTT Version 1.32.2 commit: 1ec1e57
Coordinator-Typ    ConBee2/RaspBee2
Coordinator-Version  0x26720700

[image]

giuliano108, 2023-09-04:
I have two SNZB-02 . They both got "stuck" at different times in their lifetime (a couple of weeks, a month). They re-pair just fine (use "force remove" in the zigbee2mqtt UI, pair again), but then go dormant and stop reporting data. I tried re-pairing a bunch of times, same behaviour.

Today I manage to unstuck a sensor (haven't tried with the second one yet):

* On the sensor, long press the reset button for 5s until the LED indicator flashes three times, which means the device has entered pairing mode. _No need_ to remove it from zigbee2mqtt first.
* When the sensor checks in, in zigbee2mqtt hit the "reconfigure" button (yellow button with two arrows)
* In the "reporting" tab for the device, hit "apply" for every single item (just use the default values)

This suggest the sensors lose the notion of how often they should be sending data, so they just stop trying. Reconfiguring the reporting intervals seems to get them back to doing their job, at least temporarily...
Should zigbee2mqtt be expected to refresh the config every now and then?

RonnyRusten, 2023-09-05:
For me the solution was to repair a couple of the routers that had stopped reporting (Availability: Offline). I didn't think this would affect the entire network, but it obviously did. This was devices I really don't use much, and located in a location we seldom use, so I just thought I'd do it "some other time". My advice is: "Fix your routers asap, or remove them from the network".

Palthe, 2023-09-17:
I have the same problem with the SNZB-03, tried it with the Sonoff dongle P and Sonoff dongle E version. The motion sensor works perfect for a couple of days. However, after a couple of days the sensor does not change the occupancy. The SNZB-04 in my network appear not to have this problem, but I still need to monitor this more closely.

rodders99, 2023-10-30:
I've recently started experiencing this exact problem. [SNZB-04] Device has been working fine for 2+years and recently while it shows as online, when I trigger it I can see the led on the sensor flash, but the binary sensor state doesn't change  and nothing is sent on MQTT topic.

chemuki205, 2023-12-11:
Same problem here, I have 5 SNZB-03 sensors and the same thing happens to all of them, they unpair every few days and you have to delete them and pair them again for them to work, with the addition that it is necessary to perform the operation several times because They frequently report configuration errors when pairing.

krisha, 2024-01-17:
I also have one SNZB-03 that "stops" communicating around ~8-20h after pairing. (distance to coordinator SLZB-06 is around 6m through two brick walls)

I sniffed the radio and actually it is still transmitting correct occupancy data but the coordinator sends a "leave and rejoin" request right after receiving. The transmitted occupancy data is somewhere lost/rejected on the way to GUI/MQTT.

The coordinator also ACKs the received data. But the "leave and rejoin" request seems to be completely ignored by SNZB-03 - even not an ACK.

Does anybody know what triggers the "leave and rejoin" request? Might this due to missing sequence numbers or any other timeouts (battery reporting)?

(Battery level does not seem to be related, additional capacitor on PIR seems to do nothing, no interference sources measured)

zigbee firmware: 20230507
SLZB-06 firmware: 2.0.13-dev 251223 (connected by ethernet)
Zigbee2MQTT version: 1.35.0, commit: b5a90e9

mharris1984, 2024-05-16:
I keep having this issue with all Aqara sensors. I purchased a bunch on sale and built in HA zigbee had known issues and many ppl recommend Zigbee2mqtt to replace and it is amazing. But I keep getting my Aqara sensors dropping after a few weeks/months. I can push the button on the. And they check in and update. But then no more. I have to delete the device and then rejoin it.

Tosko4, 2024-12-21:
I have the same with Xiaomi MCCGQ01LM door / window sensors. When pairing it works, and they continue to work for a while. But when inactive for a bit they all drop off and don’t report new state back to z2m. 

I’ve had these running on Phoscon/Deconz for years without problems. I guess I might have to migrate back T_T, as the sensors are essential in my case to trigger automations etc. 

Had anyone ever had a fix for this?

litecross91, 2025-01-02:
I have a same problem. System uptime 14 day. An Aqara motion sensor [RTCGQ11LM] today random stopped working/ not responding. I had to reset it and it has worked again since then. I have 25 end devices and randomly just dropped this one. I have another RTCGQ11LM device, but nothing wrong with it.

SLZB-06 -> 20240710 | v2.6.8.dev23
Proxmoy/HA - everything up to date

sstarcher, 2025-01-02:
@litecross91 `RTCGQ11LM` are sensors that are known to not function depending on what they are connected to.  I don't have the links to the docs handy, but you need to ensure your `RTCGQ11LM` is connected either directly to your controller or is connected to a router that will ping it every so often.  I know the IKEA wall repeaters work for this.  Your issue is likely not related to the issue in this thread.

litecross91, 2025-01-02:
> @litecross91 `RTCGQ11LM` are sensors that are known to not function depending on what they are connected to. I don't have the links to the docs handy, but you need to ensure your `RTCGQ11LM` is [...]

What I don't understand is why it has been working for two weeks without any errors. Then it stopped responding for no reason at all. I did nothing to the network or system.

sstarcher, 2025-01-02:
If you walk by the sensor a few times a day it will keep working.  If you don't walk by it for 24ish hours it likely breaks.  Walking by it triggers it to send data.  The issue with `RTCGQ11LM` is common and sometimes breaks when not being used depending on what it's connected to.  I pulled up the docs for you https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#troubleshooting-device-stops-sending-messages-disconnects-from-network

Tosko4, 2025-01-02:
I recently switched from Deconz (using their RaspBee module), which had been working flawlessly for years with zero latency and no dropped connections. However, I found that Zigbee2MQTT (Z2M) offers better support for the diverse range of devices I now use. Additionally, Z2M allows for more customization and advanced modifications that Deconz doesn’t support.

The switch was prompted by an accidental reset of my Deconz setup, which caused me to lose all paired devices. Unfortunately, I didn’t have a backup (stupid, I know…), so rather than re-pair everything in Deconz, I decided to take the opportunity to migrate to Z2M—a change I’d been considering for some time.

Now, I’d like to ensure that these door/window sensors are connected directly to the coordinator. While some of these sensors are within range, others are upstairs, and I’m concerned that forcing them to communicate directly with the coordinator might result in poor connectivity (if it’s even possible). How can I optimize this setup while maintaining reliable connections?

Should I purchase these IKEA signal repeaters? Do they require some kind of setup/pairing?

sstarcher, 2025-01-02:
@Tosko4 are you also having issues with `RTCGQ11LM` all of my comments are only directed at `RTCGQ11LM` devices.

litecross91, 2025-01-02:
> If you walk by the sensor a few times a day it will keep working. If you don't walk by it for 24ish hours it likely breaks. Walking by it triggers it to send data. The issue with `RTCGQ11LM` is [...]

Thank you very much! I read it, but I don't think that's the case. We were away from home 5 days out of 14 during the holidays but it worked when we got home. Today it worked fine from the morning and then suddenly stopped working in the afternoon.

Tosko4, 2025-01-02:
@sstarcher Oops, looks like I pulled a classic thread hijack move! My bad. I’m actually rocking the Xiaomi MCCGQ01LM, so I guess I just crashed the wrong party. But hey, maybe the solution for your RTCGQ11LM could work for me too? Fingers crossed for some crossover magic!

Masterz69, 2025-01-26:
Have similar, but not the same issue and only with Sonoff SNZB-02.

At some point in time no more reading appears ion Z2M and sensor become Offline.
It happens with any previouslt reported battery level, f.e. 68%.
I can re-pair device, it came to Online, but still no more measurements...
If change battery and re-pair - same stuff: re-pair happens but nothing more.

Already have 4 sensors, which can re-pair, but reading sent/received/appears.
As well pushin button does not awake device (loks like), as no reading when and not possible to drop device from network.

user514, 2025-03-12:
This happens to me too. A MCCGQ01LM that's paired and working perfectly, but some hours later stop responding when moving away from magnet.

I noticed that if you open it, remove the battery and put it in againg, it starts working for some time. Don't need to re-pair to make it work. Just removing setting the battery again. Of course, I can not do this avery two hours to keeps it working... 

  (After 10 years of working with zigbee devices... I'm mainly dissapointed with how they work. I'm sure we need devices that work with small batteries, but I don't like how low reliable they are).

joshfedo, 2025-03-13:
Same issue for me aquara devices. They were stable on ZHA for 2 years. I switched to z2m and they've dropped off several times in the last 8 months. All with the same mesh, routers, devices , ect. This is clearly a z2m specific issue.

user514, 2025-03-13:
> They were stable on ZHA for 2 years. I switched to z2m and they've dropped off several times

...well in defense of zigbee2mqtt, I was being negative with zigbee in general in my post, but I have to say that after testing ZHA two years ago, I went back to zigbee2mqtt because it was more stable than ZHA in general, also gives you more information about whats going on when things went wrong. So I stick with it.

  It's sad because If things were more stable, it will be great. Sad that I have termometers that lost connection and get lost until I re-pair, etc. Batteries that give wrong information, some brands make models not completely compatible with zigbee standards, etc.

  Wifi sucks, eats battery, short range, but is more stable and normally power off/on, resets things... here, you have to re-pair, with a tedious process of pressing buttons to keep devices alive etc.

  Well, I think I'm talking too much here about generalities. But the subject of this post goes in that direction too.

closed 2025-05-22 by github-actions[bot]

Ultra-bob, 2025-07-20:
I still experience this issue, is there an open issue still tracking it?

user514, 2025-07-20:
> I still experience this issue, is there an open issue still tracking it?

I take this oportinuty to say that in the past few months I have migrated almost all my zigbee devices to bluetooth. I can not be more than happy. In the past years, zigbee has been a fight. Don't know if with mother coordinators it would have work better, but y expirience with a CC2531 coordinator and 2 zigbee routers couldn't have been worst.

Many discconects, devices dissapearing. Once a year I have to reupload the firmware to the coordinator because zigbee2mqtt didn't wont to start up.

The only thing that I have with zigbee now is a gas detector, because I haven't replaced it yet with a bluetooth version.

All de devices that I hav reaplaced with bluetooth devices like, door sensors, door beel, buttons, etc,... never have worked so good. Really fast response, never fall down, extremelly stable, battery readings are not a lie, like with all zigbee sensors that say they has 65% of battery and stop working... etc.

  Really dissapointing technology. I will keep away from zigbee devices. Did't test 'matter' or other technologies, but the new bluetooth proxy's with esp32 cards to expand range, is for me the way to go. stability and good battery life.

  I have to say it...

user514, 2025-07-21:
> it isn't even like that that ALL devices of the same kind are unreliable. I use some Sonoff buttons that work rock solid. Other refuse to cooperate.

> Other than that Zigbee works like a charm here

...to complete what I say in the previous message, finally yesterday I re-upload the firmware to my coordinator CC2531_DEFAULT_20211115 (I know it's from 2021), and as soon as I plug it into the USB of the raspi4 and START the zigbee2mqtt addon... magically evertything went normal. No error. Loaded perfectly with the mentioned error. So, in my case what happens was some corruption in the coordinator (internal caches or something I suppose). Nothing related with the end devices what makes my addon not start.

  In any case, I have had other errors and problems in the past years. I will stay away from zigbee. Good luck to other people using it, hope they have better luck than me with it.