Koenkk/zigbee2mqtt #17807: My Aqara B1 curtain motor (ZNCLDJ12LM) have some issues

jcob9320, 2023-05-27:
### What happened?

I'm using two B1 motors (ZNCLDJ12LM). They each have one different problem.

1. First motor 
   When I open or close the curtain in my HA system or soft touch, motors does not fully close or open, leaving 1% remaining. So, it either shows ‘99% open’ or ‘1% open’.

[image]
[image]
("열림" =  means open)

2. Second motor
   The second motor opens and closes perfectly. But the “running”, "motor_state" of the motor is not displayed when it is in operation. 
When I run the motor is in the process of opening, but it shows "running=false" and "motor_state=stop” in the z2m web UI.
It is inconvenient because I cannot monitor the active state of the motor in real-time on the z2m web UI.

[image]

The first motor's firmware version is 2019.www / 20200803
The second motor's firmware version is 2019.www / 20190329
I try to OTA firmware update for first motor (second motor does not exist on OTA) but I failed. 
(error shows= Failed to check update available for 'first motor' (Failed to find endpoint which support OTA cluster)

I recalibration, and repaired several times. I don't know how to do. what should I do?

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

SONOFF DONGLE-P (cc2652)

closed 2023-07-05 by github-actions[bot]

JoooostB, 2023-10-30:
Did you manage to fix your issues @jcob9320?

jcob9320, 2024-01-05:
Hi @JoooostB. Apologize for my late reply. I checked your reply too late. In summary, I managed to update the firmware for both motors connected to the Aqara M2 Hub. One motor's 'motor state' issue has been resolved (although the 1%/99% problem remains unsolved). But the other motor still has unresolved issues with 'motor state' and 'state.' Hence, I decided to give up and created a template cover sensor to use instead.