Koenkk/zigbee2mqtt #17809: Zigbee2MQTT stops with 'Adapter disconnected' error and does not restart

gribouk, 2023-05-27:
### What happened?

Dear All,

I am running Zigbee2MQTT in a docker with Sonoff zStack3x0 Coordinator revision 20210708 radio.
Every now and then I discover Zigbee2MQTT exiting operation without restart with following error:

```
error 2023-05-27 10:59:52: Adapter disconnected, stopping
info  2023-05-27 10:59:52: MQTT publish: topic 'zigbee2mqtt_sonoff_p_coordinator/bridge/state', payload '{"state":"offline"}'
info  2023-05-27 10:59:52: Disconnecting from MQTT server
info  2023-05-27 10:59:52: Stopping zigbee-herdsman...
error 2023-05-27 10:59:52: Failed to stop Zigbee2MQTT
```

below is my docker starting command which instructs explicitely to restart the container:
```
:~# docker run -d --name zigbee2mqtt_3 --restart=unless-stopped --device=/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0ead2e5b5586ec11b18ee13719c2d21c-if00-port0:/dev/ttyUSB0 -p 8082:8082 -v /zigbee2mqtt/zigbee2mqtt-data-3:/app/data -v /run/udev:/run/udev:ro -e TZ=Europe/Kiev koenkk/zigbee2mqtt
```

Regards,

### What did you expect to happen?

If an error aqures, the container restarts....

### Zigbee2MQTT version

1.30.4 commit: b2dd21e

### Adapter firmware version

20210708 

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

```
error 2023-05-27 10:59:52: Adapter disconnected, stopping
info  2023-05-27 10:59:52: MQTT publish: topic 'zigbee2mqtt_sonoff_p_coordinator/bridge/state', payload '{"state":"offline"}'
info  2023-05-27 10:59:52: Disconnecting from MQTT server
info  2023-05-27 10:59:52: Stopping zigbee-herdsman...
error 2023-05-27 10:59:52: Failed to stop Zigbee2MQTT
```

schumi2004, 2023-05-31:
I'm experiencing the same thing but not in a docker environment.
At random moments the adapter disconnects.

Adapter: SONOFF Zigbee 3.0 USB Dongle Plus-P
Adapter firmware: 20221226
Z2M version: 1.30.4

error 2023-05-31 11:48:26: Adapter disconnected, stopping
info  2023-05-31 11:48:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2023-05-31 11:48:26: Disconnecting from MQTT server
info  2023-05-31 11:48:26: Stopping zigbee-herdsman...
error 2023-05-31 11:48:26: Failed to stop Zigbee2MQTT

Iroamstaff, 2023-06-04:
If it can help you I had a similar problem as soon as I touched the adapter, my raspberry or my SSD the adapter disconnected.  The problem came from the charger of my SSD which had an alternating current leak on the USB ground.  To solve this problem I grounded

EPicLURcher, 2023-06-07:
Is this similar to this bug: https://github.com/Koenkk/zigbee2mqtt/issues/17947

gribouk, 2023-06-07:
> Is this similar to this bug: #17947

No, that one is of different origin  - i do not have problems finding the radio, the problem is that z2m does not reboot properly on failure.

ashishpandey, 2023-12-10:
I am also seeing this frequently. I have zigbee2mqtt running in docker, connected to zigstar UZG-01 over ethernet. The docker container dies frequently with below error
```
Zigbee2MQTT:error 2023-12-10 13:35:43: Adapter disconnected, stopping
Zigbee2MQTT:info  2023-12-10 13:35:43: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Zigbee2MQTT:info  2023-12-10 13:35:43: Disconnecting from MQTT server
Zigbee2MQTT:info  2023-12-10 13:35:43: Stopping zigbee-herdsman...
Zigbee2MQTT:error 2023-12-10 13:35:43: Failed to stop Zigbee2MQTT
```
it would be preferable for zigbee2mqtt to be more resilient to connectivity errors

MostHated, 2023-12-10:
I, too, am using a UZG-01 over ethernet. I did a continuous port check every 500ms to the UZG-01 on the port Z2M is connecting to, as well as the web interface, my home assistant box, etc, and didn't see the zigbee port go down, nor any of the others I was checking, but the Z2M docker gave that same error after 1h 59m 55s and does this repeatedly once the connection time eaches that same point:

```sh
error 2023-12-05 15:16:52: Adapter disconnected, stopping
info  2023-12-05 15:16:52: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2023-12-05 15:16:52: Disconnecting from MQTT server
info  2023-12-05 15:16:52: Stopping zigbee-herdsman...
error 2023-12-05 15:16:52: Failed to stop Zigbee2MQTT
```

ashishpandey, 2023-12-12:
> Z2M docker gave that same error after 1h 59m 55s and does this repeatedly once the connection time eaches that same point

Good observation. I setup my docker with auto restart, and see the issue repeat every 2 hours. the adapter is up, but disconnects still happen at the 2 hour mark

I saw the issue you reported: https://github.com/Koenkk/zigbee2mqtt/issues/20148

dellorianes, 2023-12-19:
I have the same problem.
I use HA as a virtual machine. My coordinator is a tube-zb-gw-cc2652p2-v2 working by IP with Zigbee2mqtt. 
randomly, the zigbee2mqtt stop working with the error:
Zigbee2MQTT:error 2023-12-19 16:46:44: Adapter disconnected, stopping
Zigbee2MQTT:error 2023-12-19 16:46:44: Failed to stop Zigbee2MQTT

As soon as I realize the zigbee network ios down I check the coordinator, in its IP, and it looks like the coordinator is ok.

I have to re-start the addon in order to have it working again. 

Do you solve it?

MostHated, 2023-12-19:
So far, it looks like we might have. Check here: https://github.com/Koenkk/zigbee2mqtt/issues/20148

MostHated, 2023-12-19:
A few of us who had a disconnected adapter, showing that uptime of the connection reset at the same time as the addon restarting were able to fix it (so far, it seems, at least) my setting static IP on the coordinator. Not sure if yours is the same, though.

https://github.com/Koenkk/zigbee2mqtt/issues/20148

jdodgen, 2024-01-08:
I'm experiencing a similar problem. Except it USB connected. 
https://github.com/Koenkk/zigbee2mqtt/issues/20231

elordude, 2024-02-08:
Any resolution on this?  my system was working without issues until few weeks ago - i've resorted to add a cron job to restart the Zigbee2MQTT docker every two hours.   

0 */2 * * * /usr/bin/docker restart zigbee2mqtt >/dev/null 2>&1

Not the most elegant solution but it is working, I know that if the timing aligns with the reload my scripts will not trigger.

Kowi030, 2024-02-10:
Same problem here: Adapter disconnected, stopping

jdodgen, 2024-02-10:
I added code to reboot  whenever disconnect detected. I log the date and
time.
Here is the log.

Tue 09 Jan 2024 06:11:15 PM PST
Tue 09 Jan 2024 07:09:41 PM PST
Wed 10 Jan 2024 01:55:15 PM PST
Wed 10 Jan 2024 02:33:00 PM PST
Wed 10 Jan 2024 05:40:45 PM PST
Fri 12 Jan 2024 12:46:18 PM PST
Fri 12 Jan 2024 01:22:16 PM PST
Fri 12 Jan 2024 03:13:30 PM PST
Sat 13 Jan 2024 10:48:08 AM PST
Thu 25 Jan 2024 03:54:54 PM PST
Sat 27 Jan 2024 12:30:26 PM PST
Mon 29 Jan 2024 03:29:15 PM PST
Thu 01 Feb 2024 05:23:51 PM PST
Sun 04 Feb 2024 01:01:43 PM PST

Kowi030, 2024-02-11:
Can you post the code please?

DarkosGahan, 2024-02-16:
I am having the same problem except for me it gets disconnected whenever I change a device parameter i.e: when I set calibration to 0 on a zigbee plug. I 

Sonoff Dongle Plus E
zigbee: 1.35.3
FW: 6.10.3.0 build 297 ( tried with 7.1.3 ) same issue

I have read that the dongle E is experimental so I ordered a version P 
let's see

ThorstenDtk, 2024-02-16:
Does any of you know what 'adapter disconnected' means? I'm wondering if this is a problem with the hardware and if the adapter really isn't reachable. Or if this is an issue with Z2M. What would be the best way to figure out?

DarkosGahan, 2024-02-16:
> Does any of you know what 'adapter disconnected' means? I'm wondering if this is a problem with the hardware and if the adapter really isn't reachable. Or if this is an issue with Z2M. What would be the best way to figure out?

What I can tell is that it's a real disconnect. I am running Home Assistant with zigbee2mqtt as local addon and when I get the message I see the USB getting reset in que qemu logs.

FixussuxiF, 2024-02-17:
Same here but dongle directly into the machine.
The first time it happened about one week ago. 
While discovering new devices in z2m the "count down" stopped while interviewing it (so, it got the device in the list but not paired, z2m stopped in the menatime)
But just today it happened without me trying to pair new devices, (eventually z2m restarts and shades are going up and down, lights on and off..  

is the situation worsening!?

here some data 

Home assistant green
Zigbee2MQTT
 1.35.3
EZSP v8
6.10.3.0 build 297
frontend 0.6.156
zigbee_herdsman_converters_version 18.28.3
zigbee_herdsman_version 0.33.8

not always the log.txt has the last things happening (while is stopping)

log.txt:

info  2024-02-17 14:58:48: Zigbee2MQTT started!
info  2024-02-17 14:58:48: MQTT publish: topic 'zigbee2mqtt/Hue Luz lavanderia', payload '{"brightness":254,"linkquality":192,"power_on_behavior":"off","state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'
info  2024-02-17 14:58:48: MQTT publish: topic 'zigbee2mqtt/Pasillo Dimmer botton/type', payload '{"light":"led"}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":240,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":240,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":240,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Switch luz cocina gabinetes', payload '{"current":0,"energy":22.62,"linkquality":236,"power":0,"power_outage_memory":"restore","state":"OFF","switch_type":"toggle","voltage":241}'
info  2024-02-17 14:58:49: MQTT publish: topic 'zigbee2mqtt/Felice Baño Duscia RGB', payload '{"brightness":254,"color":{"h":248.08695652173915,"hue":248.08695652173915,"s":90.19607843137256,"saturation":90.19607843137256},"color_mode":"hs","color_temp":475,"do_not_disturb":null,"linkquality":184,"state":"OFF"}'
error 2024-02-17 14:58:49: Publish 'set' 'state' to 'Switch luz cocina gabinetes' failed: 'Error: Command 0xa4c138eee1f5ef9e/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
error 2024-02-17 14:58:49: Publish 'set' 'brightness' to 'Felice Luz Cama2 Ikea' failed: 'Error: Command 0xbc026efffef36a58/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
error 2024-02-17 14:58:49: Publish 'set' 'brightness' to 'Felice Luz Cama1 ikea' failed: 'Error: Command 0x4c5bb3fffee181b1/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
error 2024-02-17 14:58:49: Publish 'set' 'brightness' to 'Felice Armario Center Left' failed: 'Error: Command 0x30fb10fffe59894b/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
error 2024-02-17 14:58:49: Publish 'set' 'brightness' to 'Felice Armario Right Right ' failed: 'Error: Command 0x287681fffe6ad1d0/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reserv [... cut ...] ection":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
error 2024-02-17 20:09:45: Adapter disconnected, stopping
info  2024-02-17 20:09:45: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-02-17 20:09:45: Disconnecting from MQTT server
info  2024-02-17 20:09:45: Stopping zigbee-herdsman...
error 2024-02-17 20:09:45: Failed to stop Zigbee2MQTT

EPicLURcher, 2024-02-17:
are you using the serial port as per the instruction in the add on install guide: port: /dev/ttyUSB0

I was having this issue last year, and changed to the full path, and this fixed my issue. My config looks like

port: >-
  /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

DarkosGahan, 2024-02-17:
> are you using the serial port as per the instruction in the add on install guide: port: /dev/ttyUSB0
>
> I was having this issue last year, and changed to the full path, and this fixed my issue. My [...]

In my case, yes I was using the full path. meanwhile I read that the sonoff dongle-E was experimental and somehow buggy. I ordered a dongle-P, started from scratch. Since then everything is running fine again.

moreover, I was having issues with OTA where updated were running for days without finishing.

since the dongle-P FW updates run fine.

BTW not sure if DEVs are reading this thread but I am also on a HA forum where ppl seem to suffer the same kind of issues with conbee's.

ThorstenDtk, 2024-02-17:
> In my case, yes I was using the full path. meanwhile I read that the sonoff dongle-E was experimental and somehow buggy. I ordered a dongle-P, started from scratch. Since then everything is running fine again.

I just finished the migration of ~100 devices from DeConz, I'm not keen on re-pairing all of them again. In my case OTA updates are the only cause of these errors. Updates of non-battery devices sometimes works, sometimes it doesn't. But after editing state.json the updates resume and finally complete. I can live with that. But the tradfri switches take a long time to update (4-10 hours), and for those the update always starts from scratch.

elordude, 2024-02-17:
I'm not using full path but i'm using specific dev link defined using usb-serail.rules  it worked flawless ofr many months without issue and it recently started having problems.  My ZigBee adapter a nortek gocontrol husbzb-1 - I have OpenHab running natively and Home Assistant/ZigBee2MQTT running as docker containers.  Everytime there is a disconnect I do not see any usb errors on the host logs - even with the two hours Z2M docker container restart I still have few disconnects, I disable the OTA Automatic check to see if it helps.  I wonder if this is a docker bug and not an Z2M issue. 

`lrwxrwxrwx 1 root root          7 Feb 13 11:28 /dev/ttyCADDX -> ttyUSB2
lrwxrwxrwx 1 root root          7 Feb 13 11:28 /dev/ttyINSTEON -> ttyUSB0
lrwxrwxrwx 1 root root          7 Feb 16 05:04 /dev/ttyZIGB -> ttyUSB3
lrwxrwxrwx 1 root root          7 Feb 16 05:04 /dev/ttyZWAVE -> ttyUSB1

SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="FT9JX8IX", SYMLINK+="ttyCADDX"
SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="8a2a", ENV{ID_USB_INTERFACE_NUM}=="00", SYMLINK+="ttyZWAVE"
SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="8a2a", ENV{ID_USB_INTERFACE_NUM}=="01", SYMLINK+="ttyZIGB"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A906XNGT", SYMLINK+="ttyINSTEON"

Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Zigbee2MQTT:error 2024-02-13 16:00:12: Adapter disconnected, stopping
Zigbee2MQTT:error 2024-02-13 16:00:13: Failed to stop Zigbee2MQTT
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Zigbee2MQTT:error 2024-02-14 14:00:12: Adapter disconnected, stopping
Zigbee2MQTT:error 2024-02-14 14:00:12: Failed to stop Zigbee2MQTT
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Zigbee2MQTT:error 2024-02-14 18:00:12: Adapter disconnected, stopping
Zigbee2MQTT:error 2024-02-14 18:00:12: Failed to stop Zigbee2MQTT
Using '/app/data' as data directory
Zigbee2MQTT:error 2024-02-14 18:00:23: Adapter disconnected, stopping
Zigbee2MQTT:error 2024-02-14 18:00:23: Failed to stop Zigbee2MQTT
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Zigbee2MQTT:error 2024-02-16 05:04:54: Adapter disconnected, stopping
Zigbee2MQTT:error 2024-02-16 05:04:54: Failed to stop Zigbee2MQTT
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory

`

MFornander, 2024-03-04:
Same problem here.  Docker setup and using a SLZB-06 over Ethernet PoE, not USB.  Z2M quits with "adapter disconnected" and doesn't restart.

Kowi030, 2024-03-04:
For me the solution was to swap my Raspberry 3 for an Intel Nuc. With the new system everything is running smoothly again.

fumantsu, 2024-03-14:
I don't want to disappoint all you guys but you are commenting to a closed issue. Nobody will pay attention.

tackin, 2024-03-14:
> I don't want to disappoint all you guys but you are commenting to a closed issue. Nobody will pay attention.

.... except those with the same problem guided to this issue by a google-search. Just reading the postes helped me solving it. Thanks to everybody commenting on even closed issues. ;-)

fumantsu, 2024-03-14:
coincidence. Almost all closed threads are not getting attention from
anyone and definitely not from the devs which means that if you don't open
a new issue nothing ever will happen.

webbertakken, 2024-03-30:
Started seeing this behaviour after adding 2 additional devices. Has been running great for months before that.

Perhaps it errors out after a certain threshold?

stefan-schweiger, 2024-04-09:
@Koenkk can we get this issue reopend as it seems to still affect many people?

DennisDeKoster, 2024-04-15:
> @Koenkk can we get this issue reopend as it seems to still affect many people?

Yes please, I'm having the same issue since a while. Been running before without issues. Maybe something critical changed with updates over time?

stefan-schweiger, 2024-04-18:
For whatever reason it seems to happen more often currently. Before I had it about once a day, but for the last few this it keeps happening 3-4 times a day.

Koenkk, 2024-04-20:
For those using POE adapters, please check if the issue also happens when connecting over USB. If it doesn't the issue is likely in your network setup.

stefan-schweiger, 2024-04-21:
Is there any additional logging to activate for the USB problems which would help to track this down?

jdodgen, 2024-04-22:
> Can you post the code please?

Sorry to not get back. it runs as a script from the console when it fails I log date and reboot
z2m is part of a bigger system and runs as a separate process communicating through message queues

```
def task(watch_dog_queue):
    while True:
        # not sure why they need this but this will run it
        os.chdir("/opt/zigbee2mqtt")
        print(os.getcwd())
        # subprocess.run(["sudo", "systemctl", "stop", "zigbee2mqtt"])  # just incase 
        subprocess.run(["/usr/bin/npm", "start"]) # runs forever or until it crashes
        # previous plan was to attempt restart of z2m but it seems 
        # only a reboot fixes things, a very brut force solution
        # This will be changed when the z2m problem is figured out
        print("exited, attempting restart")
        watch_dog_queue.put(["shutdown", "zigbee2mqtt problems"])
        time.sleep(10) # just to keep it out of a hard loop    
```  
here are the important parts  of my "watchdog" it does a graceful shutdown of other processes, then calling os._exit(-1)

```
        item = watch_dog_queue.get(timeout=const.watch_dog_queue_timeout)
        ...
        (command, oprand) = item
        ...
        if command == "shutdown":
                ....
                os._exit(-1)
```
My code runs from .bashrc and I log date and reboot
```
sudo python3 main.py
sudo echo `date` >> bootlog
sudo reboot
```
current bootlog
```
Tue 09 Jan 2024 06:11:15 PM PST
Tue 09 Jan 2024 07:09:41 PM PST
Wed 10 Jan 2024 01:55:15 PM PST
Wed 10 Jan 2024 02:33:00 PM PST
Wed 10 Jan 2024 05:40:45 PM PST
Fri 12 Jan 2024 12:46:18 PM PST
Fri 12 Jan 2024 01:22:16 PM PST
Fri 12 Jan 2024 03:13:30 PM PST
Sat 13 Jan 2024 10:48:08 AM PST
Thu 25 Jan 2024 03:54:54 PM PST
Sat 27 Jan 2024 12:30:26 PM PST
Mon 29 Jan 2024 03:29:15 PM PST
Thu 01 Feb 2024 05:23:51 PM PST
Sun 04 Feb 2024 01:01:43 PM PST
Fri 01 Mar 2024 08:02:48 PM PST
Mon 04 Mar 2024 12:34:44 PM PST
```

filipefumaux, 2024-05-21:
Also same same here

verybadsoldier, 2024-05-26:
I am also experiencing the exact same problem as described in the issue.

I am not sure if zigbee2mqtt quitting due to "Adapter disconnected, stopping" is a bug, though. But the container not restarting even when configured to `restart=always` is not what I would have expected.

FixussuxiF, 2024-05-27:
I literally threw the Dongle-E in the trash after repairing over 100 devices with a new Dongle-P coordinator (it took me globally less time this than trying to find a solution). In the occasion I setup a HA vm on a proxmox which runs like charm (especially the reboot takes less than half the time then having HA running directly on the miniPC)

stefan-schweiger, 2024-05-27:
After I switched to use Ember all my issues are gone now for weeks with the dongle-e.

verybadsoldier, 2024-05-27:
This is the output of the docker daemon when I disconnect the stick (and docker is not restarting the container):
```
May 27 21:16:08 minion dockerd[873516]: time="2024-05-27T21:16:08.515959763+02:00" level=error msg="restartmanger wait error: error gathering device information while adding custom device \"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_25_9B_6C_0F-if00-port0\": no such file or directory"
```

I think this error is indicating that docker's "restartmanager" encountered a problem which led to not restarting the container: `restartmanger wait error`

The error is generated here in the source code:
https://github.com/moby/moby/blob/ceefb7d0b9c5b6fbd1ea7511592a4ddb28ec4821/daemon/monitor.go#L141

But I fail to understand what is happening there or why. Does anyone know what "restartmanger wait error" means or in which situation it happens?

verybadsoldier, 2024-05-27:
Hm, I think this is intended behavior of docker:
it actually tries to restart the container but the adapter/device which is supposed to get mapped into the container is not available, it gives up and does not try again.

As mentioned here:
https://forums.docker.com/t/restart-always-stops-trying-if-mounted-device-is-not-found/136296

So, is there a bug here involved at all?
* z2m exits purposelly (?) when adapter disappears
* docker stops trying to restart when mapped device is not found

Would you agree? If yes: what to do here to make the container restart without manual interaction? Any good ideas for a workaround?
How about a udev rule that starts the container upon connectiong the adapter?

mahonni, 2024-09-17:
> Hm, I think this is intended behavior of docker: it actually tries to restart the container but the adapter/device which is supposed to get mapped into the container is not available, it gives up [...]

I'm also interested in knowing how to restart without manual interaction. Any ideas?

schoren, 2024-09-17:
I was able to use udev to automatically restart on reconnect:

```
# /etc/udev/rules.d/99-zigbee.rules
SUBSYSTEM=="tty", ATTRS{idVendor}=="XXXX", ATTRS{idProduct}=="YYYY", SYMLINK+="ttyACM0", RUN+="/usr/bin/restartstart-z2m.sh"
```

```
#/usr/bin/restartstart-z2m.sh

docker restart zigbee2mqtt
```

Works like a charm

Jeppedy, 2024-10-25:
> I'm also interested in knowing how to restart without manual interaction. Any ideas?

I've made a fairly robust version of what others have posted/alluded to.  Youi'll want to change the alerts to your own alerting preferences/setup.
Set up a Ping/ICMP integration for the IP of your Zigbee gateway.
Whenever that gateway had been offline,  but comes online, and Zigbee2MQTT is not running (you may have to change a config to see the running addons), it restarts the addon.

`alias: "Watchdog: Z2M Addon"
description: Start Z2M add-on when down and network coord becomes available
triggers:
  - entity_id:
      - binary_sensor.zigbeegw_ping
    to: "on"
    for:
      hours: 0
      minutes: 0
      seconds: 15
    trigger: state
    id: GWNowSeen
    from: "off"
    alias: When Zigbee GW comes Online
  - entity_id:
      - binary_sensor.zigbeegw_ping
    to: "off"
    for:
      hours: 0
      minutes: 0
      seconds: 15
    trigger: state
    id: GWNoLongerSeen
    from: "on"
    alias: When Zigbee GW goes Offline
  - trigger: state
    entity_id:
      - binary_sensor.zigbee2mqtt_running
    from: "on"
    to: "off"
    id: stopped
    alias: When Z2M Addon stops running
  - trigger: state
    entity_id:
      - binary_sensor.zigbee2mqtt_running
    from: "off"
    to: "on"
    id: started
    alias: When Z2M Addon starts running
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - GWNowSeen
          - condition: state
            entity_id: binary_sensor.zigbee2mqtt_running
            state: "off"
        sequence:
          - data:
              announce_message: Zigbee Gateway on-line.  Attempting to restart Zigbee2MQTT
              send_tvs: "false"
              send_voice: "false"
              send_mobile: "true"
            action: script.notify_standard
          - action: hassio.addon_start
            metadata: {}
            data:
              addon: 45df7312_zigbee2mqtt
      - conditions:
          - condition: trigger
            id:
              - GWNoLongerSeen
        sequence:
          - data:
              announce_message: Zigbee Gateway no longer seen on the network
              send_tvs: "false"
              send_voice: "false"
              send_mobile: "true"
            action: script.notify_standard
      - conditions:
          - condition: trigger
            id:
              - stopped
        sequence:
          - data:
              announce_message: Zigbee2MQTT has stopped and is not currently running
              send_tvs: "false"
              send_voice: "false"
              send_mobile: "true"
            action: script.notify_standard
      - conditions:
          - condition: trigger
            id:
              - started
        sequence:
          - data:
              announce_message: Zigbee2MQTT is now running
              send_tvs: "false"
              send_voice: "false"
              send_mobile: "true"
            action: script.notify_standard
mode: single
`

biddie-fiddler, 2024-10-25:
> I've made a fairly robust version of what others have posted/alluded to.

Damn, you've gone and one upped me 😉

One thing that my automation doesn't do, that I think would be good, is to at a counter to track restart attempts, and to stop the automation after X amount of retries, send a notification or something to inform of the issue. 

I like your solution tho!

s1n7ax, 2024-11-30:
Same here with USB connector. Container is stopped but never restarted

```
[2024-11-30 09:22:16] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/p water pump', payload '{"child_lock":"UNLOCK","current":0.01,"energy":51.98,"indicator_mode":"off/on","linkquality":21,"power":0,"power_outage_memory":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":225}'
[2024-11-30 09:22:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/p refrigerator ', payload '{"child_lock":"UNLOCK","current":0.39,"energy":14.71,"indicator_mode":"off/on","linkquality":7,"power":79,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":225}'
[2024-11-30 09:22:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/p refrigerator ', payload '{"child_lock":"UNLOCK","current":0.39,"energy":14.71,"indicator_mode":"off/on","linkquality":10,"power":79,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":225}'
[2024-11-30 09:22:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/p ac', payload '{"child_lock":"UNLOCK","current":0.03,"energy":0.96,"indicator_mode":"off/on","linkquality":94,"power":1,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":215}'
[2024-11-30 09:22:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/p ac', payload '{"child_lock":"UNLOCK","current":0.03,"energy":0.96,"indicator_mode":"off/on","linkquality":102,"power":1,"power_outage_memory":"restore","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":219}'
[2024-11-30 09:22:40] info: 	zh:zstack:znp: Port closed
[2024-11-30 09:22:40] info: 	zh:zstack:znp: closing
[2024-11-30 09:22:40] error: 	z2m: Adapter disconnected, stopping
[2024-11-30 09:22:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-11-30 09:22:40] info: 	z2m: Disconnecting from MQTT server
[2024-11-30 09:22:40] info: 	z2m: Stopping zigbee-herdsman...
[2024-11-30 09:22:40] info: 	z2m: Stopped zigbee-herdsman
[2024-11-30 09:22:40] info: 	z2m: Stopped Zigbee2MQTT
```

chris669, 2024-12-11:
had the same problem 2 weeks ago with my slaesh usb adapter on my hassio and again today. I suspected the usb cable the first time and replugged it.  but the second time I didn't touch it. for me the only solution was a full reboot. so my pov is that this is a code issue, not hardware related. thanks. 

`[2024-12-11 12:10:21] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-11 12:10:21] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_90_DA_9F-if00-port0''
[2024-12-11 12:10:21] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_90_DA_9F-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-12-11 12:10:21] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-11 12:10:21] error: 	z2m: Failed to start zigbee `

chris669, 2024-12-13:
Hi @Koenkk, crashed today again... how can I help? 

`[2024-12-13 09:57:19] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-13 09:57:19] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_90_DA_9F-if00-port0''
[2024-12-13 09:57:19] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_90_DA_9F-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-12-13 09:57:19] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-13 09:57:19] error: 	z2m: Failed to start zigbee
[2024-12-13 09:57:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-13 09:57:19] error: 	z2m: Exiting...
[2024-12-13 09:57:19] error: 	z2m: Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_90_DA_9F-if00-port0`

verybadsoldier, 2024-12-13:
> Same here with USB connector. Container is stopped but never restarted

> Portainer shows this when inspected though I can see the device still there under the same path

I am thinking the container _is_ trying to restart and I think the USB _is_ not available for a short period in the moment of the restart:
https://github.com/Koenkk/zigbee2mqtt/issues/17809#issuecomment-2134012934

lscarneiro, 2024-12-17:
In my case, the issue is that `zigbee2mqtt` stops "cleanly" or in other words, with code 0, when any disturbance to my USB adapter happens.

I generally reconnect the USB adapter and expect that my `restart: always` will do the job, but unfortunately, not, because the image exited with code `0` (not a failure).

is there a way to force `zigbee2mqtt` to keep restarting or retrying or at least exit with a failure `1` code such that my docker daemon can keep restarting it?

I say that because my problem is hardware, not software. If I fix the "hardware" problem (which happens 2 times a month on avg.) I would love for it to just restart itself, instead of making me log into portainer to manually start a "stack" that has `restart: always` set already, which is sorta weird.

Jeppedy, 2024-12-17:
> In my case, the issue is that `zigbee2mqtt` stops "cleanly" or in other words, with code 0, when any disturbance to my USB adapter happens.

To me, everything is working as designed.  What folks are asking for is Z2M to add more resilience.  That's one approach to resilience, but another is externally applied resilience.  Should a component retry more often for long periods of time?  Maybe.  But until it does, there's an easy way for us to supply that same behavior.
I haven't had an outage since I've implemented this script, even through network reboots, wiring changes, HA restarts.  It's been rock solid.
See my watchdog script here:
https://github.com/Koenkk/zigbee2mqtt/issues/17809#issuecomment-2436887042

verybadsoldier, 2024-12-17:
It seems that script is only usable if you are running a Home Assistant server, right?

lscarneiro, 2024-12-17:
> It seems that script is only usable if you are running a Home Assistant server, right?

I would assume so...

In my case Home Assistant is just another container on my RPi server, managed by portainer, and zigbee2mqtt is yet another container.

I'm not sufe if it's useful for me either.

lscarneiro, 2024-12-17:
to be honest, external scrips are more of a workaround than actually a fix.

it's really weird for a piece of software to stop "successfully" when something WRONG happens, if the `zigbee2mqtt` lost communication with the adapter, it should fail instead of "well, guess I will just end my shift early, then... See ya!".

that's why it caught me off-guard when `restart: always` wasn't sufficient, there was clearly a problem (no adapter available) and the container decided that "the work is finished with success, no need to restart"... on a failure. So the watchdog (Docker) that watches for a failure looks into it and see "well, no failure here, I guess this is just expected and I won't restart it, then"

Jeppedy, 2024-12-17:
Totally agree.  From an IT mindset, that's not an "ended successfully"
situation.  Absolutely the return code should indicate an error.  Only a
successful requested shutdown should return a 0 return code.

verybadsoldier, 2024-12-17:
> that's why it caught me off-guard when restart: always wasn't sufficient, there was clearly a problem (no adapter available) and the container decided that "the work is finished with success, no need to restart"...

Sorry, I am a bit confused: does `restart: always` not actually mean to restart _even_ on exit code 0? I thought only `on-failure` does restart only on exit code >0.

henkisdabro, 2024-12-22:
> I've made a fairly robust version of what others have posted/alluded to. Youi'll want to change the alerts to your own alerting preferences/setup. Set up a Ping/ICMP integration for the IP of your [...]

This really helped me, thank you. As you mentioned it require to create a new integration (for the ICMP Ping of the device), adjust the entity id names in the code and also create the script that notifies my device. Appreciate you sharing this.

MichaelUray, 2024-12-25:
> Sorry, I am a bit confused: does restart: always not actually mean to restart even on exit code 0? I thought only on-failure does restart only on exit code >0.

Yes, this is at least what I understand by the documentation as well.

https://docs.docker.com/engine/containers/start-containers-automatically/

cherealnice, 2025-01-08:
For me, this error is happening because of power issues with the USB port to which I have my adapter connected. It would be nice to have a good solution for when something like this happens, to either retry connections to the adapter before crashing or to crash in such a way that the docker restart option can pick up on it. I tried `unless-stopped` and `always` and neither work.

webbertakken, 2025-01-08:
On Unraid `restart: always` works as expected. Have been running with it for at least 6 months without problems.

Another workaround could be to run a script on a cronjob on your server to restart the container twice a day. Afaik the USB only disconnects after a longer period of time (but I'm not sure).

In any case, the script could be as simple as this:

```bash
#!/bin/bash
echo "Restarting Zigbee2MQTT service at $(date)"
docker restart zigbee2mqtt
```

[13 more replies]