Koenkk/zigbee2mqtt #17833: All Tuya thermostats are dangerous for zigbee network.

NikitaKorneev, 2023-05-29:
### What happened?

_TZE200_ye5jkfsb TS0601, _TZE200_znzs7yaw TS0601, _TZE200_aoclfnxz TS0601 And ALL the other Tuya Zigbee thermostats have a bug in their firmware that flood the network with messages.

Having more than couple of device will shut down an entire network at one point.

debounce option will not help. 

### How to reproduce it (minimal and precise)

Add any Tuya - based zigbee thermostat and sniff the network/look at the info logs. You should see over 50 messages per second.

### Zigbee2MQTT version

latest

### Adapter firmware version

latest (id does not matter)

### Adapter

Egony stick v4

lux73, 2023-05-29:
all Tuya Devices are very noisy ;-)

NikitaKorneev, 2023-05-29:
> all Tuya Devices are very noisy ;-)

The reason I created this issue is because devices I am talking about paralyze network entirely!

gpshml, 2023-05-29:
> > all Tuya Devices are very noisy ;-)
> 
> The reason I created this issue is because devices I am talking about paralyze network entirely!

I have just tried adding a Moes based one and it is doing the same thing, even when un-joined.  Going to have to un-install it I think.

Koenkk, 2023-05-30:
@NikitaKorneev can you confirm this also happens with the TuYa gateway? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

NikitaKorneev, 2023-05-31:
> @NikitaKorneev can you confirm this also happens with the TuYa gateway? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

I have ordered everything I need for sniffing the network. BRB in about four days. Sorry, cannot do it earlier due to having no hardware

NikitaKorneev, 2023-07-05:
My apologies - had no time to do the sniffing. I will try to do it by the end of next week.

closed 2023-08-14 by github-actions[bot]

smarticz, 2023-12-10:
Yes, I have this problem too.

kokireg, 2024-02-12:
I have the same problem with TZE204_aoclfnxz

NikitaKorneev, 2024-02-12:
I was unable to sniff the network to get the logs. Does anyone have such ability?

0anton, 2024-02-20:
I can confirm it too.

Not all models are affected, but I met at least two already (one is sold as Beco Zigbee Thermostat, another MOES Smart Tuya).

Some models starts spamming the network almost immediately, other can work for months until they go crazy.

Besides significantly reducing the capacity of the Zigbee network by constantly transmitting void messages, devices also become unmanageable.

This is especially problematic, if those at the remote location. I often experienced that they stuck on the "switched on" mode, overheating the house left uninhabited over the winter. 

Right now thermostat went crazy. It is uncontrollable, spams at rate of multiple messages per second and stuck in the "heat" state.

Frequent status update (typical for all spammy Tuya thermostats):

```
Zigbee2MQTT:info  2024-02-21 00:59:48: MQTT publish: topic 'zigbee2mqtt/Теплый пол кухня 2', payload '{"child_lock":null,"current_heating_setpoint":25,"deadzone_temperature":null,"heat":"OFF","linkquality":21,"local_temperature":4,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}'
Zigbee2MQTT:info  2024-02-21 00:59:49: MQTT publish: topic 'zigbee2mqtt/Теплый пол кухня 2', payload '{"child_lock":null,"current_heating_setpoint":25,"deadzone_temperature":null,"heat":"OFF","linkquality":21,"local_temperature":4,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}'
Zigbee2MQTT:info  2024-02-21 00:59:49: MQTT publish: topic 'zigbee2mqtt/Теплый пол кухня 2', payload '{"child_lock":null,"current_heating_setpoint":25,"deadzone_temperature":null,"heat":"OFF","linkquality":21,"local_temperature":4,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}'
Zigbee2MQTT:info  2024-02-21 00:59:49: MQTT publish: topic 'zigbee2mqtt/Теплый пол кухня 2', payload '{"child_lock":null,"current_heating_setpoint":25,"deadzone_temperature":null,"heat":"OFF","linkquality":21,"local_temperature":4,"local_temperature_calibration":null,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":null,"running_state":"idle","sensor":null,"system_mode":"off"}'
```

Dropping duplicate (debouncing) messages Z2M helps only partially, at least reducing the CPU usage of the Home Assistant server.

I wonder if it is possible to remotely restart such thermostat...

[image]

In this case this is a `_TZE204_aoclfnxz` device.

smarticz, 2024-02-22:
I also checked the power consumption at iddle 

50-70W!

This device is the worst 😭😭😭

monqui-ulia, 2024-09-05:
Some problem with MOES BHT002/BHT006 device, _TZE204_aoclfnxz . Does anyone knows how to update firmware?

NikitaKorneev, 2024-09-11:
> Some problem with MOES BHT002/BHT006 device, _TZE204_aoclfnxz . Does anyone knows how to update firmware?

As I understand it - Tuya and only Tuya can.

ubombi, 2024-10-19:
That's sad. Also have this problem.  We can ask Beca (manufacturer) to fix their firmware, and share it. Via email, or whatsup/we-chat. [contact page](https://www.becaenergy.com/contact/)
```
sales@becaenergy.com
support@becaenergy.com
skype: Jingzi6789

0086-592-6151256
0086-17850149590
```

ubombi, 2024-10-23:
Upd: Reasonable response from Beca.
> Thank you for your feedback and for sharing the link, so that we can learn about the user experience first-hand.
>
> Regarding the problem of Zigbee sending data frequently, we have learned about it and upgraded the program for the BHT-> 006GBZB model.
> 
> This problem will not occur again with the new product.
> 
> For customers who have already purchased, if you want to upgrade, you need an update tool worth $50, which is more expensive than the thermostat. It is not recommended to buy it.

Also asked for the new firmware.  Or we can for someone to dump it from a newer device.
I assume it requires a UART adapter, but deeply hope OTA is possible _(by some default Tuya mechanism)_

0anton, 2024-10-23:
Got similar response from Chinese representative of the producer (via
Whatsapp, can share the number if needed). Same offer: to reprogram the
devicee using some tool. I believe, anyway that a usual RS232-TTL
programmer will do the job just fine. Only need to know the attachment
points on the board and have an image / programming software. OTA was also
told to me as not possible for this kind of devices. I remember this part
of the conversation (about OTA) very well, because it somehow surprised me
a lot, that they have nothing. He relied, its costs to order the feature in
the firmware (!) and costs to maintain the infrastructure (distribution
point). I was surprised by such cost awareness and also by seeming very
complex and developed ecosystem they live in: hardware manufacturer AND a
specialist firmware developer, which charges per each feature (like OTA)
and the manufacturer carefully choose the feature set…
Still it sucks a lot to have those thermostats with chatter bug, they spam
the network a lot, affecting I believe the responsiveness of other devices,
like light-button systems (those just hang, getting unresponsive then
thermostats fires their bursts of identical messages)…

jan0o0sch, 2024-11-07:
@0anton 
I would be very interested in a solution to flash a new firmware to these thermostats ... I have a lot of them installed and they are killing my network! I really need a solution .... thanks in advance!

jvdburgt, 2025-01-09:
I have the same problem (2 of these thermostats on my network).

Previously, I had the same problem with Tuya presence sensors, but managed to fix this by flashing them using an ST-Link and jumper cables. Would be great if that could be done here as well.

JagerSprinkles, 2025-01-09:
> I have the same problem (2 of these thermostats on my network).
> 
> Previously, I had the same problem with Tuya presence sensors, but managed to fix this by flashing them using an ST-Link and jumper cables. Would be great if that could be done here as well.

Can you provide a link to what you flashed them with? I have multiple of these presence sensors that constantly spam the network. I would like to try flashing a new firmware.

jvdburgt, 2025-01-10:
> > 
> Can you provide a link to what you flashed them with? I have multiple of these presence sensors that constantly spam the network. I would like to try flashing a new firmware.

Take a coffee, put on some background music, relax, and start reading this thread: https://github.com/Koenkk/zigbee2mqtt/issues/19045#issuecomment-2581286972

Read the first 100-150 posts or so, and you know what to do ;)

xBohlix, 2025-01-30:
> Got similar response from Chinese representative of the producer (via
> Whatsapp, can share the number if needed). Same offer: to reprogram the
> devicee using some tool. I believe, anyway that a usual [...]

Did you get any further with this? I have 7 of these Thermostats and they are constantly spamming my z2m network. I noticed this behaviour the first time when I opened the z2m map, where only these devices were blinking all the time, updating the LQI. Looking into my logs they are constantly sending their payload.

Onepamopa, 2025-04-04:
Does anyone have newer firmware for the Moes thermostats? 
Zigbee Model
    TS0601
Zigbee Manufacturer
    _TZE204_aoclfnxz
Description
    Smart heating thermostat
Manufacturer
    [Moes](https://www.zigbee2mqtt.io/supported-devices/#v=Moes)
Model
    [BHT-002/BHT-006](https://www.zigbee2mqtt.io/devices/BHT-002_BHT-006.html#moes-bht-002%252Fbht-006)

Got 5 of these in my network, it's barely working......

iJuiceman, 2025-04-28:
I have the same issue.  I have used the Throttle setting for 30.  This will only let it publish every 30 seconds.  This has made a bit of a difference to the chattyness of several of my Zigbee devices.

Onepamopa, 2025-04-28:
There is custom firmware -- https://github.com/slacky1965/tuya_thermostat_zrd
If your thermostat is compatible - flash it. The firmware improves things a lot.

xBohlix, 2025-04-30:
> There is custom firmware -- https://github.com/slacky1965/tuya_thermostat_zrd If your thermostat is compatible - flash it. The firmware improves things a lot.
> 
> Throttling does not help because the devices continue to spam the airwaves, the only difference is that z2m does not see the spam when you throttle it. Doesn't improve things one bit.

I cannot say how much I love you 😄!!
This is awesome! Everything worked as expected. No more chatty devices in my mqtt network. Now my other devices respond much better than before.

jvdburgt, 2025-04-30:
I followed all steps in the firmware how-to guide. However, after I added 
```
ota:
zigbee_ota_override_index_location: local_ota_index.json
```
in the Z2M configuration.yaml file and restart Z2M, the added lines are gone from the config file. Just to be sure, I renamed the old configuration.yaml file, and uploaded a fresh one with the added lines of code. Still, one restart later and the added lines are gone again. 

Please advice.

nocin, 2025-04-30:
@jvdburgt make sure to stop Z2M before changing the configuration.yaml, else it will be overwritten each time. At least this was my experience

Gerhard2025, 2025-09-15:
Thanks, but same after 1min powerless and restart.

BTW: I have 4 more identical devices that are shown in devicelist as:
(They have still the original firmware 4a)
But i cant do configuration here as well and only LQI and identify are shown

Zigbee-Modell
TS0601
Zigbee-Hersteller
_TZE204_aoclfnxz
Beschreibung
Original Tuya thermostat ready for custom Firmware update
Unterstützungsstatus
Unterstützt

IEEE-Addresse
0xa4c138dff26ff0f1
Netzwerk-Adresse
0xCABC / 51900
Hersteller
[Slacky-DIY](https://www.zigbee2mqtt.io/supported-devices/#v=Slacky-DIY)
Modell
[Original Tuya thermostat ready for update](https://www.zigbee2mqtt.io/devices/Original_Tuya_thermostat_ready_for_update.html#slacky-diy-original%2520tuya%2520thermostat%2520ready%2520for%2520update)

Onepamopa, 2025-09-15:
Force remove? Re-pair, then try interrogation again? Mine are these:

[image]

Onepamopa, 2025-09-15:
Just follow the instructions, that's how I did it. (note that I removed some stuff recently from there, so I can't say for sure what was what).

PaTara43, 2025-11-11:
Hi, guys

I faced the same proble while writing a custom converter for an engo thermostat (it's tuya, basically)

The culprit was `extend: [tuya.modernExtend.tuyaBase({dp: true, forceTimeUpdates: true, timeStart: "1970"})],`

When replaced with 

```
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    extend: [tuya.modernExtend.tuyaBase({respondToMcuVersionResponse: false})],
```

I was actually for another [reason ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/tuya.ts#L2164)but saved my network as well

p.s. The `tuya.modernExtend.tuyaBase` is necessary for time sync and schedule mode

PaTara43, 2025-11-11:
> Cool finding Pavel!
> 
> So you say, setting `respondToMcuVersionResponse: false` will prevent those thermostats to flood the network with their updates? This would be indeed a great fix!

You may want to try to make an external converter for your device copying the official one and adding the parameter