Koenkk/zigbee2mqtt #17835: SSM-U01 lumi.switch.n0agl1 Failed to find endpoint which support OTA cluster

aldarund, 2023-05-29:
### What happened?

When clicking check update it just show up error message Failed to find endpoint which support OTA cluster

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

 ZigStar Stick v4

### Debug log

Debug 2023-05-29 23:34:49Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Тест энергия кухея","transaction":"kmi5m-4"}'
Info 2023-05-29 23:34:49Checking if update available for 'Тест энергия кухея'
Debug 2023-05-29 23:34:49Check if update available for '0x54ef44100082ce2c' (lumi.switch.n0agl1)
Info 2023-05-29 23:34:49MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Тест энергия кухея"},"error":"Failed to check if update available for 'Тест энергия кухея' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"kmi5m-4"}'
Error 2023-05-29 23:34:49Failed to check if update available for 'Тест энергия кухея' (Failed to find endpoint which support OTA cluster)
Debug 2023-05-29 23:34:49AssertionError [ERR_ASSERTION]: Failed to find endpoint which support OTA cluster at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:243:9) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:144:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:73) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:126:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1184:20) at Receiver.emit (node:events:513:28)

Koenkk, 2023-05-30:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/717d5125f22ca77c0fd4a73d46fe268f

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

aldarund, 2023-05-30:
Yes, that fixes the issue

Koenkk, 2023-05-31:
Great, integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-31 by Koenkk