Koenkk/zigbee2mqtt #17840: [New device support]: Halemeier HA-ZGMW2-E

ronrem, 2023-05-30:
### Link

https://cloud.halemeier.de:444/index.php/s/HYSztHDjfb8WrHy?path=%2FEnglish#pdfviewer

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0x84b4dbfffeada8a9","nwkAddr":32635,"manufId":4723,"manufName":"Halemeier GmbH","powerSource":"Mains (single phase)","modelId":"HA-ZGMW2-E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"HA-ZGMW2-E","manufacturerName":"Halemeier GmbH","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r60"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r60","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1685459988015,"defaultSendRequestWhen":"immediate"}

### Comments

Analysed the logs:
Info 2023-05-30 17:19:46Starting interview of '0x84b4dbfffeada8a9'
Info 2023-05-30 17:19:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84b4dbfffeada8a9","ieee_address":"0x84b4dbfffeada8a9","status":"started"},"type":"device_interview"}'
Info 2023-05-30 17:19:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84b4dbfffeada8a9","ieee_address":"0x84b4dbfffeada8a9"},"type":"device_announce"}'
Info 2023-05-30 17:19:48Successfully interviewed '0x84b4dbfffeada8a9', device has successfully been paired
Warning 2023-05-30 17:19:48Device '0x84b4dbfffeada8a9' with Zigbee model 'HA-ZGMW2-E' and manufacturer name 'Halemeier GmbH' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-05-30 17:19:48MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x84b4dbfffeada8a9","ieee_address":"0x84b4dbfffeada8a9","status":"successful","supported":false},"type":"device_interview"}'
Warning 2023-05-30 17:21:21Received message from unsupported device with Zigbee model 'HA-ZGMW2-E' and manufacturer name 'Halemeier GmbH'
Warning 2023-05-30 17:21:21Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Supported color modes

Cool white and warm white

Koenkk, 2023-05-31:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/e38fdc7252de83d5c47b773d5e851141

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

also:

Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

ronrem, 2023-06-01:
Thanks, it works. I'll get back on the color temperature question.

ronrem, 2023-06-01:
The color temp range is:

Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":160,"colorTempPhysicalMax":450}