Koenkk/zigbee2mqtt #17841: [Wrong device]: Zigbee Fingerbot

cloudbr34k84, 2023-05-31:
### Link

https://www.aliexpress.com/item/1005004091078287.html

### Model

Fingbot Plus

### Description

White Zigbee Fingbot Plus

### Vendor

Adaprox

### Picture (link)

https://w24cdn.cz/www.maximin.cz/_/1200x800-0-0-0-0/product/product_130269/5e40a3a9f84278455adbe61c8591b605.png

### Database entry

{"id":80,"type":"EndDevice","ieeeAddr":"0xa4c13820005330ca","nwkAddr":43090,"manufId":4417,"manufName":"_TZ3210_dse8ogfy","powerSource":"Battery","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[6,61184,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3210_dse8ogfy","powerSource":3,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256cd420","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1685503993466,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-05-31:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-05-31 by Koenkk

kristofejro, 2023-06-03:
Hi @cloudbr34k84 Is it full zigbee device or it has bluetooth? I was looking for fingerbot based on zigbee but it was always based on BT. Thanks.

cloudbr34k84, 2023-06-03:
It's full zigbee.  It just came out.  Got it from aliexpress

kristofejro, 2023-06-03:
Thanks mate. I will order them also from aliexpress.
So far it is not listed here: https://www.zigbee2mqtt.io/supported-devices/

cloudbr34k84, 2023-06-03:
Yeah it needs some support as it only supports switch on,  switch off[image]

Gr8z, 2023-06-13:
How can we add support for more options such as:

- Mode (Click/Switch)
- Touch Control (On/Off)
- Down Movement (0-100%)
- Up Movement (0-100%)
- etc

cloudbr34k84, 2023-06-13:
Need to write a converter and sniff more data points out.

Gr8z, 2023-06-14:
I have sniffed out the codes but am stuck on getting these into Z2M. I tried adding them into an external convertor but don't know how.

code | Name | type | values
-- | -- | -- | --
101 | Mode | enum | Click/Switch/Program
107 | Touch Control | bool | on/off
104 | Reverse | enum | up_on/up_off
102 | Down Movement | number | 50-100
106 | Up Movement | number | 0-50
103 | Sustain Time | value | 0s-10s
1 | Switch | enum | On/Off
105 | Battery | number | 0-100

rluvaton, 2023-06-22:
> I have sniffed out the codes but am stuck on getting these into Z2M. I tried adding them into an external convertor but don't know how.
> 
> code	Name	type	values
> 101	Mode	enum	Click/Switch/Program
> 107	Touch Control	bool	on/off
> 104	Reverse	enum	up_on/up_off
> 102	Down Movement	number	50-100
> 106	Up Movement	number	0-50
> 103	Sustain Time	value	0s-10s
> 1	Switch	enum	On/Off
> 105	Battery	number	0-100
> Here is the code I wrote but I am sure its totally wrong:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee')
> const tz = require('zigbee-herdsman-converters/converters/toZigbee')
> const exposes = require('zigbee-herdsman-converters/lib/exposes')
> const reporting = require('zigbee-herdsman-converters/lib/reporting')
> const extend = require('zigbee-herdsman-converters/lib/extend')
> const e = exposes.presets
> const ea = exposes.access
> const tuya = require('zigbee-herdsman-converters/lib/tuya')
> 
> const definition = {
>   // Since a lot of TuYa devices use the same modelID, but use different datapoints
>   // it's necessary to provide a fingerprint instead of a zigbeeModel
>   fingerprint: [
>     {
>       // The model ID from: Device with modelID 'TS0601' is not supported
>       // You may need to add \u0000 at the end of the name in some cases
>       modelID: 'TS0001_1',
>       // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>       manufacturerName: '_TZ3210_dse8ogfy',
>     },
>   ],
>   zigbeeModel: ['TS0001'],
>   model: 'TS0001_1',
>   vendor: 'Adaprox',
>   description: 'White Zigbee FingerBot plus',
>   fromZigbee: [tuya.fz.datapoints],
>   toZigbee: [tuya.tz.datapoints],
>   configure: tuya.configureMagicPacket,
>   exposes: [
>     // Here you should put all functionality that your device exposes
>     e.battery(),
>     e.enum('101', ea.STATE_SET, ['Click', 'Switch', 'Program']),
>     e.boolean('107', ea.STATE_SET),
>     e.enum('104', ea.STATE_SET, ['up_on', 'up_off']),
>     e.percent('102', ea.STATE_SET),
>     e.percent('106', ea.STATE_SET),
>     e.value('103', ea.STATE_SET, {min: 0, max: 10, unit: 's'}),
>     e.enum('1', ea.STATE_SET, ['On', 'Off']),
>   ],
>   meta: {
>     // All datapoints go in here
>     tuyaDatapoints: [
>       [101, 'mode', tuya.converters.enum({0: 'Click', 1: 'Switch', 2: 'Program'})],
>       [107, 'touch_control', tuya.converters.boolean],
>       [104, 'reverse', tuya.converters.enum({0: 'up_on', 1: 'up_off'})],
>       [102, 'down_movement', tuya.converters.percent],
>       [106, 'up_movement', tuya.converters.percent],
>       [103, 'sustain_time', tuya.converters.value({min: 0, max: 10, unit: 's'})],
>       [1, 'switch', tuya.converters.enum({1: 'On', 0: 'Off'})],
>       [105, 'battery', tuya.converters.value({min: 0, max: 100})],
>     ],
>   },
> }
> 
> module.exports = definition
> ```

where you able to change the `sustain_time` in home assistant without zigbee2mqtt, it doesn't seem to work...

Hanzofm, 2024-09-08:
Hi, I have similar problem, recognises all options but it not execute

4rianton, 2024-10-22:
paired, options visible, but nothing works.