Koenkk/zigbee2mqtt #17843: Lot of traffic generated by IKEA E1757/E1926 / Smart Blinks FYRTUR/KADRILJ

tekman54190, 2023-05-31:
### What happened?

Hi team, 

I just upgraded the firmwares of all my IKEA smart blinds during night. This morning I can see a lot of traffic generated by all of the blinds for no apparent reason. This is overloading the global network and slowing down event detection of "real" events. 
I just upgraded the SONOFF Zigbee Dongle firmware to the latest one (20221226) without more luck. 

The smart blinds have been upgraded to firmware version 24.4.11, and zigbee2mqtt version is the latest one at the moment: 1.30.4
I found this article: https://github.com/Koenkk/zigbee2mqtt/issues/13681 but it does not seems to be exactly the same issue. But symptoms are the same: batteries are draining a lot. 

I attached the logs in the article to show how much event are sent to the coordinator (FYI I've 9 blinds). 

Can someone help me? or find a way to rollback to the old firmware? 

Thanks guys for your help. 

### How to reproduce it (minimal and precise)

Just upgrade the blinds FW through OTA

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

20221226

### Adapter

zStack3x0 / Sonoff Zigbee 3.0 dongle plus

### Debug log

May 31 10:22:09 assel-rpi-domotique npm[20335]: Zigbee2MQTT:debug 2023-05-31 10:22:09: Received Zigbee message from 'SDB - Store droit', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
May 31 10:22:09 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:09: MQTT publish: topic 'zigbee2mqtt/SDB - Store droit', payload '{"battery":9,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"SDB - Store droit","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffea393f2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1757","networkAddress":46974,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:09.909Z","linkquality":0,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:debug 2023-05-31 10:22:10: Received Zigbee message from 'Bibliotheque - Store droit', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Bibliotheque - Store droit', payload '{"battery":37,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"Bibliotheque - Store droit","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe1c467a","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1926","networkAddress":40946,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.149Z","linkquality":80,"position":10,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Salon - Store milieu', payload '{"battery":94,"device":{"applicationVersion":34,"dateCode":"20190311","friendlyName":"Salon - Store milieu","hardwareVersion":1,"ieeeAddr":"0x842e14fffe667b99","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1926","networkAddress":42036,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.275Z","linkquality":142,"position":40,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:debug 2023-05-31 10:22:10: Received Zigbee message from 'Chambre 2 - Store droit', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Chambre 2 - Store droit', payload '{"battery":35,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"Chambre 2 - Store droit","hardwareVersion":1,"ieeeAddr":"0x84fd27fffe9b34e5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1757","networkAddress":23376,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.283Z","linkquality":32,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Salon - Store droit', payload '{"battery":76,"device":{"applicationVersion":34,"dateCode":"20190311","friendlyName":"Salon - Store droit","hardwareVersion":1,"ieeeAddr":"0x842e14fffe79c42e","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1926","networkAddress":58271,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.299Z","linkquality":76,"position":40,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Bibliotheque - Store droit', payload '{"battery":37,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"Bibliotheque - Store droit","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe1c467a","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1926","networkAddress":40946,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.338Z","linkquality":83,"position":10,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:debug 2023-05-31 10:22:10: Received Zigbee message from 'Chambre 2 - Store gauche', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Chambre 2 - Store gauche', payload '{"battery":8,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"Chambre 2 - Store gauche","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe4045d6","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1757","networkAddress":42332,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.405Z","linkquality":32,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:debug 2023-05-31 10:22:10: Received Zigbee message from 'SAM - Store Gauche', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/SAM - Store Gauche', payload '{"battery":99,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"SAM - Store Gauche","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe12b7c7","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1926","networkAddress":61982,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:10.418Z","linkquality":80,"position":40,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'
May 31 10:22:10 assel-rpi-domotique npm[20335]: Zigbee2MQTT:info  2023-05-31 10:22:10: MQTT publish: topic 'zigbee2mqtt/Chambre 2 - Store gauche', payload '{"battery":8,"device":{"applicationVersion":33,"dateCode":"20190311","friendlyName":"Chambre 2 - Store gauche","hardwareVersion":1,"ieeeAddr":"0x680ae2fffe4045d6","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1757","networkAddress":42332,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3} [... cut ...] 20190311","friendlyName":"Chambre 2 - Store droit","hardwareVersion":1,"ieeeAddr":"0x84fd27fffe9b34e5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1757","networkAddress":23376,"powerSource":"Battery","softwareBuildID":"24.4.11","stackVersion":99,"type":"EndDevice","zclVersion":3},"last_seen":"2023-05-31T08:22:11.386Z","linkquality":29,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":false}'

kloodhu, 2023-05-31:
same problem after FW update:

2023-05-31 10:42:08MQTT publish: topic 'zigbee2mqtt/IkeaBlind01', payload '{"battery":168,"last_seen":"2023-05-31T10:42:08+02:00","linkquality":116,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
info 2023-05-31 10:42:08MQTT publish: topic 'zigbee2mqtt/IkeaBlind01', payload '{"battery":168,"last_seen":"2023-05-31T10:42:08+02:00","linkquality":116,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
debug 2023-05-31 10:42:09Received Zigbee message from 'IkeaBlind01', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0
info 2023-05-31 10:42:09MQTT publish: topic 'zigbee2mqtt/IkeaBlind01', payload '{"battery":168,"last_seen":"2023-05-31T10:42:09+02:00","linkquality":112,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'
info 2023-05-31 10:42:09MQTT publish: topic 'zigbee2mqtt/IkeaBlind01', payload '{"battery":168,"last_seen":"2023-05-31T10:42:09+02:00","linkquality":112,"position":100,"state":"OPEN","update":{"installed_version":604241937,"latest_version":604241937,"state":"idle"},"update_available":null}'

tekman54190, 2023-05-31:
Interesting thing... it seems firmware has been removed. I missed one blind yesterday evening, and now I cannot upgrade from 2.3.088 to 24.... as no new firmware is available... 
Strange

kloodhu, 2023-05-31:
the blind witlout fw update:
2023-05-31 10:50:33Read result of 'genPollCtrl': {"checkinInterval":172800,"longPollInterval":4,"shortPollInterval":4,"fastPollTimeout":1}

the blind after fw update:
2023-05-31 10:48:59Read result of 'genPollCtrl': {"checkinInterval":4,"longPollInterval":4,"shortPollInterval":1,"fastPollTimeout":1,"fastPollTimeoutMax":78}

tekman54190, 2023-05-31:
NIce findings... is there a way to change these values?

Snapperud, 2023-05-31:
Same problem for me

kloodhu, 2023-05-31:
in the device's page you can use the dev console to write the value to the "checkinInterval". it helped me.

popy2k14, 2023-05-31:
Guys, my Z2M also showed me the new firmware versions and i started the update on one fyrtur, which is currently running.
Because of the issues and ikea have obviously pulled the update (also nothing in the release notes), i want to stop the update.

There is no "stop" button. How can i stop the OTA?
Can i restart Z2M to stop it or is there another "nicer" way?

tekman54190, 2023-05-31:
> in the device's page you can use the dev console to write the value to the "checkinInterval". it helped me.

Thanks issue fixed on my side ;)

popy2k14, 2023-05-31:
One thing: mine is updating to 2.4.0.40 (604241937 dec = 24040011 hex = 2.4.040) , see log:

MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rollo Fenster klein', payload '{"battery":100,"last_seen":"2023-05-31T11:15:24+02:00","linkquality":134,"position":100,"state":"OPEN","update":{"installed_version":587761201,"latest_version":604241937,"progress":16.27,"remaining":7708,"state":"updating"},"update_available":null}'

Which version code is the problematic one?
Or did i converted/parsed the version number wrong?

tekman54190, 2023-05-31:
> 

> One thing: mine is updating to 2.4.0.40 (604241937 dec = 24040011 hex = 2.4.040) , see log:
>
> MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Rollo Fenster klein', payload [...]

I think the 24.x branch is faulty :D any way you can just adjust the checkinterval and no more issue after that

popy2k14, 2023-05-31:
> I think the 24.x branch is faulty :D any way you can just adjust the checkinterval and no more issue after that

thx for the hint with the checkinterval attribute.
Can you please post an screenshot how to send this to the device (sorry rather new to Z2M) and which value (172800 when i got your right).

tekman54190, 2023-05-31:
[image]
seems also that a factory reset of the blind fixes the issue

popy2k14, 2023-05-31:
thx, will try it after the one has finished the update

popy2k14, 2023-05-31:
> [image] seems also that a factory reset of the blind fixes the issue

Was there any other issue that you performed an factory reset? Or just as precautions :-)

Is there any changelog of the firmware? 
On the usual link https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html there is no evidence of the new firmware.

tekman54190, 2023-05-31:
> Was there any other issue that you performed an factory reset? Or just as precautions :-)
>
> Is there any changelog of the firmware? On the usual link [...]

This particular blinds was so busy, it was not possible to write the checkinterval value on it :)

popy2k14, 2023-05-31:
> This particular blinds was so busy, it was not possible to write the checkinterval value on it :)

ok, got you. So you have just changed the value on other blinds and they are running well.
So no need to factory reset after setting the value :-)

closed 2023-05-31 by tekman54190

Skeletorjus, 2023-05-31:
Thank you all for shedding some light on this. Had an interesting evening trying to find out why the Zigbee-network was extremely unreliable.

After a while I remembered that the Fyrturs got an update and found this thread...

Had no luck changing the setting because the network load was too high, but I pulled the batteries and everything is working great again. Will do a factory reset tomorrow.

Thanks again!

mherke, 2023-06-01:
Thanks for the hint to set the intervall back to 172800. That fixed my issues. The blinds are now responding as usual.

ur7x, 2023-06-01:
> seems also that a factory reset of the blind fixes the issue

In my case a power cycle of the binds did not fix the problem, but the Checkinterval patch worked great

Thanks!

popy2k14, 2023-06-02:
@ur7x i think he mean an factory reset by pressing both buttons for 5 seconds until the led begins to flash. This way, all settings in the blind will be reset to factory defaults.