Koenkk/zigbee2mqtt #17849: Eco-dim.05 reporting as not supported

joep1222, 2023-05-31:
### What happened?

What happened?
Tried adding a new EcoDim Eco-Dim.05. Interview = ok. Support status = unsupported.

### What did you expect to happen?

I expected that the eco-dim0.5 was supported:
https://www.zigbee2mqtt.io/devices/Eco-Dim.05.html

Here someone where it works
https://github.com/Koenkk/zigbee2mqtt/issues/13511

[image]

### How to reproduce it (minimal and precise)

Added external converter. 
Difference with issue: https://github.com/Koenkk/zigbee2mqtt/issues/13511
My database.db does have a ID
device: 0x84b4dbfffe08a0e5

[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/11617340/database.txt)

### Zigbee2MQTT version

 1.30.4 commit: unknown

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff | Zigbee 3.0 | Dongle Plus-E

### Debug log

[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/11617335/log1.txt)

Koenkk, 2023-06-01:
Can you provide the data/database.db entry of this device?

joep1222, 2023-06-03:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/11642078/database.zip)
Here is the database.db.

Koenkk, 2023-06-03:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/24abeafddd26a9b092abd07bad8def00

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

joep1222, 2023-06-03:
It almost works.  The dimmer is recognized and supported.  On and off and dim level from the dimmer works well and the status can be read directly in ha.  Control from ha not possible.  See the log below.

` 2023-06-03 18:05:49Publish 'set' 'state' to '0x84b4dbfffe08a0e5' failed: 'Error: Command 0x84b4dbfffe08a0e5/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56890 - 1 - 27 - 6 - 11 after 10000ms)'
Debug 2023-06-03 18:05:49Error: Command 0x84b4dbfffe08a0e5/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56890 - 1 - 27 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2023-06-03 18:05:49MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x84b4dbfffe08a0e5' failed: 'Error: Command 0x84b4dbfffe08a0e5/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 56890 - 1 - 27 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x84b4dbfffe08a0e5"},"type":"zigbee_publish_error"}'
Debug 2023-06-03 18:05:55Saving state to file /config/zigbee2mqtt/state.json
Debug 2023-06-03 18:06:11Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,29,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator`

Koenkk, 2023-06-03:
Looks like the device is not reachable, can you try to re-pair it close to the coordinator?

joep1222, 2023-06-03:
Thanks for the fast reply.
Try pairing again.  Distance +/-1.5 meters.  Tried with the meter cupboard door open and the antenna pointed at the dimmer.  Another dimmer, eco-dim.07 at +/- 2 meters distance, works without problem.

Koenkk, 2023-06-04:
Is this the case for both left and right?

Koenkk, 2023-06-04:
This looks like some connection problem, I don't think I can solve it from z2m.

joep1222, 2023-06-11:
I was testing last week.  With zha it works without delay or something like that.  However, I would still like to use zigbee2mqtt because of the compatibility of other devices.  I also exchanged the ecodim.05 dimmer with an ecodim.07 dimmer which does not cause any problems.  Given the short distance, the connection seems good to me.  Are there other things I can test?

Koenkk, 2023-06-12:
Could you provide the herdsman debug log when trying to change the state of this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

joep1222, 2023-06-17:
I fixed the issue. 
Set the transition time to 1 instead of "empty". 
Now it works perfect!
[image]

Koenkk, 2023-06-17:
Cool!

closed 2023-06-17 by Koenkk