Koenkk/zigbee2mqtt #17856: AM43-0.45/40-ES-EB [_TZE200_zah67ekd] Open/Close/Stop case independent.

SargonofAssyria, 2023-06-01:
### What happened?

Sending "Open', "Close" or "Stop" does not activate device.

### What did you expect to happen?

Should Open, Close or Stop.
Please make the action independent if the case sent.
Let "Open" and "OPEN" be the same, etc....

### Zigbee2MQTT version

1.31.0-dev commit: b97a09c7

### Adapter firmware version

20230507

### Adapter

TT CC1352P

Koenkk, 2023-06-01:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-01 by Koenkk

SargonofAssyria, 2023-06-01:
Perfect Tnx.