Koenkk/zigbee2mqtt #17857: [New device support]:

vpaulos, 2023-06-01:
### Link

https://aliexpress.ru/item/1005004988422423.html?spm=a2g2w.cart.cart_split.2.2e644aa6L6zF9y&sku_id=12000032999852523

### Database entry

'TS0601' '_TZE204_5toc8efa' 0xa4c138e467d721fb 0xA849

### Comments

I`ve got 2 same devices. One work fine, and the second is unsupported. Please how can I clone the config from one to second?
**unsupported:** 
Device type Router
Zigbee Model TS0601
Zigbee Manufacturer _TZE204_5toc8efa
Support status Not supported
IEEE Address 0xa4c138e467d721fb
Network address 0xA849 

**AND SUPPORTED IS:**
Device type Router
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_aoclfnxz
Description Moes BHT series Thermostat
Support status Supported

IEEE Address 0x9035eafffefd866e
Network address 0x720F
Manufacturer [Moes](https://www.zigbee2mqtt.io/supported-devices/#v=Moes)
Model [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb)

Koenkk, 2023-06-01:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-06-01 by Koenkk

vpaulos, 2023-06-01:
and in release it will be? when?